open62541/open62541 #3436: network problems，Linux device runs for three or four days

ruikai, 2020-02-14:
## Description
I porting the code(open62541.c, open62541.h) to the device ,it is run OK,but when it runs for three or four days,the The indicator light of network port flashes very fast，It feels like there's a lot of data in and out of the network,and last For more than 20 minutes,And then everything goes back to normal，Where should I start looking for problems
## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-02-14:
Can you increase the log level so that individual packages are logged?

ruikai, 2020-02-14:
I don't know where to add logs. Before that, did anyone encounter similar problems? Where can I start to find problems

ruikai, 2020-02-14:
the open62541.c and open62541.h that I'm using is the version  of year 2018 ,Is it possible that the version is too old?Where can I get the latest version？
tks

jpfr, 2020-02-14:
On our homepage.
Or in the release tab of the github page.

jpfr, 2020-02-14:
check the cmake options for the log level.

jpfr, 2020-10-03:
Please reopen when there is more log output or debug information available.

closed 2020-10-03 by jpfr