open62541/open62541 #3437: Data races in unit tests

andranik-s, 2020-02-14:
## Description

ThreadSanitizer (gcc 7.4.0) reports issues when running unit tests (UA_MULTITHREADING is NOT enabled). Most of them are not critical e.g. unsynchronized access to `bool running`, but there are some more important ones.

The most problematic test is `check_discovery` where ThreadSanitizer reports 61 warnings. Those warning do not seem to be false positives.

Let's take a look at one of them:
 
<details>
    <summary>ThreadSanitizer's report</summary>

```WARNING: ThreadSanitizer: data race (pid=25557)
  Write of size 8 at 0x7b1c00000a10 by main thread:
    #0 __UA_TimerZip_ZIP_INSERT open62541/src/ua_timer.c:44 (check_discovery+0x1b6bd)
    #1 UA_TimerZip_ZIP_INSERT open62541/src/ua_timer.c:44 (check_discovery+0x1b85c)
    #2 addCallback open62541/src/ua_timer.c:90 (check_discovery+0x1cd8a)
    #3 UA_Timer_addRepeatedCallback open62541/src/ua_timer.c:115 (check_discovery+0x1cf6b)
    #4 addRepeatedCallback open62541/src/server/ua_server.c:375 (check_discovery+0x2b5fe)
    #5 UA_Server_addPeriodicServerRegisterCallback open62541/src/server/ua_services_discovery.c:708 (check_discovery+0x51e79)
    #6 Server_register_periodic open62541/tests/server/check_discovery.c:221 (check_discovery+0xc3e3)
    #7 tcase_run_tfun_nofork.isra.9 <null> (check_discovery+0xd9fea)
    #8 __libc_start_main <null> (libc.so.6+0x21b96)

  Previous read of size 8 at 0x7b1c00000a10 by thread T2:
    #0 UA_TimerZip_ZIP_MIN open62541/src/ua_timer.c:44 (check_discovery+0x1be27)
    #1 UA_Timer_process open62541/src/ua_timer.c:155 (check_discovery+0x1d3d8)
    #2 UA_Server_run_iterate open62541/src/server/ua_server.c:624 (check_discovery+0x2c43b)
    #3 serverloop_register open62541/tests/server/check_discovery.c:95 (check_discovery+0xb778)
    #4 <null> <null> (libtsan.so.0+0x296ad)

  Location is heap block of size 104 at 0x7b1c00000a10 allocated by main thread:
    #0 malloc <null> (libtsan.so.0+0x2ae03)
    #1 addCallback open62541/src/ua_timer.c:73 (check_discovery+0x1cc10)
    #2 UA_Timer_addRepeatedCallback open62541/src/ua_timer.c:115 (check_discovery+0x1cf6b)
    #3 addRepeatedCallback open62541/src/server/ua_server.c:375 (check_discovery+0x2b5fe)
    #4 UA_Server_addRepeatedCallback open62541/src/server/ua_server.c:385 (check_discovery+0x2b68b)
    #5 UA_Server_init open62541/src/server/ua_server.c:312 (check_discovery+0x2b266)
    #6 UA_Server_newWithConfig open62541/src/server/ua_server.c:340 (check_discovery+0x2b36f)
    #7 UA_Server_new open62541/plugins/ua_config_default.c:48 (check_discovery+0xcd622)
    #8 setup_register open62541/tests/server/check_discovery.c:104 (check_discovery+0xb838)
    #9 srunner_run_setup <null> (check_discovery+0xd9de7)
    #10 __libc_start_main <null> (libc.so.6+0x21b96)

  Thread T2 (tid=25560, running) created by main thread at:
    #0 pthread_create <null> (libtsan.so.0+0x2bcee)
    #1 setup_register open62541/tests/server/check_discovery.c:119 (check_discovery+0xb942)
    #2 srunner_run_setup <null> (check_discovery+0xd9de7)
    #3 __libc_start_main <null> (libc.so.6+0x21b96)

SUMMARY: ThreadSanitizer: data race open62541/src/ua_timer.c:44 in __UA_TimerZip_ZIP_INSERT
```
</details>
as we can see `UA_Server *server_register` is accessed from two different threads without any synchronization:

https://github.com/open62541/open62541/blob/af9767240c10f86b29261d8cdfe3abfd26a63345/tests/server/check_discovery.c#L221-L222

https://github.com/open62541/open62541/blob/af9767240c10f86b29261d8cdfe3abfd26a63345/tests/server/check_discovery.c#L93-L97

Full reports:
[discovery.log](https://github.com/open62541/open62541/files/4203626/discovery.log)
[the_rest.log](https://github.com/open62541/open62541/files/4203627/the_rest.log)

## Background Information / Reproduction Steps

Change CMake file to enable `-fsanitize=thread` and run tests. Don't forget to redirect stderr somewhere you can see it.

Used CMake options:

```bash
cmake -DBUILD_SHARED_LIBS=1 CMAKE_BUILD_TYPE=Debug -DUA_BUILD_UNIT_TESTS=1 ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 18.04.4 LTS
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-02-14:
@Pro Can you have a look at the discovery part?

Pro, 2020-02-14:
Not sure if I can manage to fix the issues within the next few weeks.
@andranik-s if you would like to push some fixes for these issues I would be really glad.

I would also suggest that we enable the `-fsanitize=thread` flag by default in CI.

andranik-s, 2020-02-14:
@Pro I can take a closer look at them within the next week

closed 2025-02-04 by andreasebner