open62541/open62541 #3439: BadsecurityCeckFailed !

bnadem, 2020-02-14:
i just create certificat and private key with other tools, and with your server_encryption programm

my certification is look lik 

Basic Fields
  serial number: 0089839869a2daf1dd83eb084f1f481372
  signature algorithm: SHA256withRSA
  issuer: /CN=localhost/C=ES/ST=MAD/L=LLL/O=MyServer/OU=MyServer/serialNumber=000.00
  notBefore: 200214132503Z
  notAfter: 291223132503Z
  subject: /CN=localhost/C=ES/ST=MAD/L=LLL/O=MyServer/OU=MyServer/serialNumber=000.00
  subject public key info: 
    key algorithm: RSA
    n=00c1f9bc84794117...
    e=010001
X509v3 Extensions:
  subjectKeyIdentifier :
    21e17470b95a8a3103780930af63d1a0d42361858004ffd9
  keyUsage CRITICAL:
    digitalSignature,nonRepudiation
  subjectAltName :
    **DNS,localhost**,**URI,urn:open62541.server.application**,**IP,127.0.0.1**
  basicConstraints CRITICAL:
    cA=true
  authorityKeyIdentifier :
    kid=43330745eec02b934abd94aa2c77cab9e40c5c83d9745939
signature algorithm: SHA256withRSA
signature: 55cf0eb2a186e33d...

./server_encryption publickeyderFormat privateKeyderFormat
become error **BadsecurityCeckFailed

have you an Idea Please?
certificat extention hae change into txt format just only for Upload

[Cert_req1_Signed.txt](https://github.com/open62541/open62541/files/4204601/Cert_req1_Signed.txt)
[PrKey_req1.txt](https://github.com/open62541/open62541/files/4204603/PrKey_req1.txt)

**

jpfr, 2020-02-14:
This is not enough information to understand the problem.
When does BadSecurityCeckFailed happen? At startup? When a client connects?

Also, please turn up the logging and show outputs where possible.

bnadem, 2020-02-15:
When  i Start your exanple_encryption with public und privat key from
github path : example/encryption/server_encryption.exe , there no Client,
only server first

bnadem, 2020-02-26:
can you please , close this issues, i know who is error , thanks

closed 2020-02-26 by jpfr