open62541/open62541 #3441: During LDS-ME setup, Getting Error

jdhanasekar, 2020-02-17:
Hi,

I am working on to setup a LDS-ME server in Linux system. So, I have copied the content of server_lds.c to myServer.c. When I run ```./myServer```, I am getting below error,

```
root@quanta:/home/user/dhanasekar/BMS/OPC/open62541# gcc -std=c99 open62541.c myServer.c -o myServer
myServer.c: In function ‘main’:
myServer.c:38:22: error: ‘UA_ServerConfig_Discovery {aka struct <anonymous>}’ has no member named ‘mdns’
     config->discovery.mdns.mdnsServerName = UA_String_fromChars("LDS");
                      ^
myServer.c:48:22: error: ‘UA_ServerConfig_Discovery {aka struct <anonymous>}’ has no member named ‘mdns’
     config->discovery.mdns.serverCapabilitiesSize = 1;
                      ^
myServer.c:51:22: error: ‘UA_ServerConfig_Discovery {aka struct <anonymous>}’ has no member named ‘mdns’
     config->discovery.mdns.serverCapabilities = caps;

```
To fix the issue, I have commented out ```UA_ENABLE_DISCOVERY_MULTICAST``` and ```UA_ENABLE_DISCOVERY``` in ```open62541.h```. But, it is not working. 

Can you please help me to solve the issue?. 

FYR, I have shared the content of ```myServer.c```,

```
#include <open62541/plugin/log_stdout.h>
#include <open62541/server.h>
#include <open62541/server_config_default.h>

#include <signal.h>
#include <stdlib.h>

static volatile UA_Boolean running = true;
static void stopHandler(int sig) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "received ctrl-c");
    running = false;
}

int main(void) {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    UA_Server *server = UA_Server_new();
    UA_ServerConfig_setDefault(UA_Server_getConfig(server));

    UA_ServerConfig *config = UA_Server_getConfig(server); //dhanasekar
    UA_ServerConfig_setDefault(config); //dhanasekar

// dhanasekar - Start - LDS

// This is an LDS server only. Set the application type to DISCOVERYSERVER.
    // NOTE: This will cause UaExpert to not show this instance in the server list.
    // See also: https://forum.unified-automation.com/topic1987.html
    config->applicationDescription.applicationType = UA_APPLICATIONTYPE_DISCOVERYSERVER;
    UA_String_clear(&config->applicationDescription.applicationUri);
    config->applicationDescription.applicationUri =
            UA_String_fromChars("urn:open62541.example.local_discovery_server");

    // Enable the mDNS announce and response functionality
    config->discovery.mdnsEnable = true;

    config->discovery.mdns.mdnsServerName = UA_String_fromChars("LDS");

    // See http://www.opcfoundation.org/UA/schemas/1.03/ServerCapabilities.csv
    // For a LDS server, you should only indicate the LDS capability.
    // If this instance is an LDS and at the same time a normal OPC UA server, you also have to indicate
    // the additional capabilities.
    // NOTE: UaExpert does not show LDS-only servers in the list.
    // See also: https://forum.unified-automation.com/topic1987.html

    // E.g. here we only set LDS, and you will not see it in UaExpert
    config->discovery.mdns.serverCapabilitiesSize = 1;
    UA_String *caps = (UA_String *) UA_Array_new(1, &UA_TYPES[UA_TYPES_STRING]);
    caps[0] = UA_String_fromChars("LDS");
    config->discovery.mdns.serverCapabilities = caps;

    /* timeout in seconds when to automatically remove a registered server from
     * the list, if it doesn't re-register within the given time frame. A value
     * of 0 disables automatic removal. Default is 60 Minutes (60*60). Must be
     * bigger than 10 seconds, because cleanup is only triggered approximately
     * every 10 seconds. The server will still be removed depending on the
     * state of the semaphore file. */
    // config->discoveryCleanupTimeout = 60*60;

// dhanasekar - End -LDS

    UA_StatusCode retval = UA_Server_run(server, &running);

    UA_Server_delete(server);
    return retval == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}

```

jdhanasekar, 2020-02-18:
I have built example program by this command,

```
 cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL -DUA_BUILD_EXAMPLES=FULL -DUA_ENABLE_AMALGAMATION=OFF -DUA_ENABLE_DISCOVERY=ON -DUA_ENABLE_DISCOVERY_MULTICAST=ON ..
```
After running ```discovery_server_lds```, when I try to access the lds in Prosys client, I am getting below error,

```Connection 6 | SecureChannel 2 | The server sends no timestamp in the request header. See the 'verifyRequestTimestamp' setting.```.

To fix this error, I need to comment TimeStamp code in open62541.c..But it is not available inside build folder. 

Please help me to fix this issue. 

In ```UaExpert```, I am getting below empty screen,

[image]

Do I need to compile again using ```gcc```  ??

RakshanPremsagar, 2020-12-06:
I would also like to know about this. How did you figure out?

teccheck, 2025-04-29:
The CTT covers discovery, and we have examples for LDS, including LDS ME. If problems persist, feel to reopen this issue

closed 2025-04-29 by teccheck