open62541/open62541 #3442: Looking for single file release for 32bits arm architecture

kaeros, 2020-02-17:
- [ ] Operating system:

Good morning,

I created this issue regarding a simple problem that I am facing at the moment: I cannot manage to find the arm 32bits (Raspbian) version of the open62541 library.

To develop my issue: I am trying to use this library in order to dockerize a very small opcua C client which final running target is an arm 32 based architecture. Using the "single file releases" (the last one on this page: https://open62541.org/releases/), I have successfully ran my code on a Linux 64 bits architecture but, as I have discovered afterward, it won't run on a 32 bits arm arch since that is not the target of this open62541 library release. 

On the presentation page of the open62541 website (https://open62541.org/), there are clickable boxes that indicates that the library is working on several OS architectures including Raspbian. Unfortunately there is no downloadable link behind these options except for the Linux 64bits one. 

So this is my question: does the library is in fact 32bits arm compatible ? If so, could you please indicate me where I could find the "single file release" specific to Raspbian? If there is no "single file release", do you support cross compiling from Linux64bits to 32Bits arm (I haven't seen any mention of "32bits arm" in the /arch repository of the git project)?

Best regards

jh-isw, 2020-02-17:
Hello,
[raspberrypi.tar.gz](https://github.com/open62541/open62541/releases/download/v1.0/open62541-raspberrypi.tar.gz)
is this of use for you?
You can find the releases here:
[https://github.com/open62541/open62541/releases](https://github.com/open62541/open62541/releases)

Best regards

jpfr, 2020-02-17:
This PR adds a CMake flag to force 32bit: #3443

It is a simple matter of adding the `-m32` compiler directive.
Our code compiles and works on any architecture I know.
32bit/64bit. Big/Little/Mixed Endian, and so on.

jpfr, 2020-02-17:
Merged. But this does not actually change the content of the single file release.

Just use the -m32 Compiler option .

Reopen if required.

closed 2020-02-17 by jpfr

kaeros, 2020-02-18:
Thank you for your quick answer.

I will test it on my end as soon as possible and let you know if that worked.