open62541/open62541 #3447: ERROR: Could not open a Session with StatusCode BadServiceUnsupported

jadhavshital, 2020-02-18:
Hi,
I have built open62541 encrypted library on windows 10. I tried with examples given [here](https://github.com/open62541/open62541/tree/master/examples/encryption). For this I had created certificates using https://github.com/open62541/open62541/tree/master/tools/certs/create_self-signed.py script. And those examples worked fine.

But when I try to connect with Prosys OpcUa Simulation Server, I am getting below error - 
[image]

My sample code is -
``	/* Load certificate and private key */
    UA_ByteString certificate = loadFile(cert_file_path.c_str());
    UA_ByteString privateKey = loadFile(key_file_path.c_str());

    /* Load the trustList. Load revocationList is not supported now */
    size_t trustListSize = 0;
    if (argc > MIN_ARGS)
        trustListSize = (size_t)argc - MIN_ARGS;
    UA_STACKARRAY(UA_ByteString, trustList, trustListSize);
    for (size_t trustListCount = 0; trustListCount < trustListSize; trustListCount++){
        trustList[trustListCount] = loadFile(argv[trustListCount + 4]);
        std::cout << " Trustlist [trustListCount] " << (trustList[trustListCount]).data << ", i = " << trustListCount << std::endl;
    }

    UA_ByteString* revocationList = NULL;
    size_t revocationListSize = 0;

    UA_Client* client = UA_Client_new();
    UA_ClientConfig* cc = UA_Client_getConfig(client);
    cc->securityMode = UA_MESSAGESECURITYMODE_SIGNANDENCRYPT;
    cc->securityPolicyUri = UA_STRING_ALLOC("http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256");
    UA_ClientConfig_setDefaultEncryption(cc, certificate, privateKey,
        trustList, trustListSize,
        revocationList, revocationListSize);

    UA_ByteString_clear(&certificate);
    UA_ByteString_clear(&privateKey);
    for (size_t deleteCount = 0; deleteCount < trustListSize; deleteCount++) {
        UA_ByteString_clear(&trustList[deleteCount]);
    }
   
    UA_StatusCode retval = UA_Client_connect(client, endpointUrl.c_str());``

Prosys simulation server - 
[image]
[image]
Certificate has been added to server -
[image]

Kindly help!

luiztsmendes, 2020-02-21:
I have a similar problem on Windows 10 platform when connecting to the Prosys OPC UA Simulation server (status code BadServiceUnsupported). However, the client example code I used was the simpler one provided in the open62514 tutorial "Building a Simple Client", which just connects to a server and reads the server's current time:

[image]

However, when using the same client example from the 0.3 branch it worked fine:

[image]

Could it be a problem with the default client configuration? I noticed that the client configuration for the 0,3 branch is much simpler than the one from the master branch.

Anyway, thanks for this outstanding job of developing an open OPC UA library!

luiztsmendes, 2020-02-21:
Just to avoid a misinterpretation, in my previous comment I meant the client example from the 0.3 branch compiled with the 0.3 library.

jadhavshital, 2020-02-24:
@luiztsmendes Thanks for the help. Now I am able to connect to Prosys OPC UA Simulation server using 0.3 branch code.

closed 2020-02-24 by jadhavshital