open62541/open62541 #3448: Subsribe to published NodeId

ljocha85, 2020-02-18:
I am trying to subscribe to published variables. The publisher works successfully by modifying the snipped code in the publisher example and adding an information model to the server.
```
/**
 * **DataSetField handling**
 *
 * The DataSetField (DSF) is part of the PDS and describes exactly one published
 * field. */
static void
addCustomDataSetField(UA_Server *server, UA_Variant var, const UA_DataType *type, const UA_NodeId nodeId, char* fieldNameAlias) {
     /* Add a field to the previous created PublishedDataSet */
    UA_DataSetFieldConfig dataSetFieldConfig;
    memset(&dataSetFieldConfig, 0, sizeof(UA_DataSetFieldConfig));
    dataSetFieldConfig.dataSetFieldType = UA_PUBSUB_DATASETFIELD_VARIABLE;
    dataSetFieldConfig.field.variable.fieldNameAlias = UA_STRING(fieldNameAlias);
    dataSetFieldConfig.field.variable.promotedField = UA_FALSE;
    dataSetFieldConfig.field.variable.publishParameters.publishedVariable = nodeId;
    dataSetFieldConfig.field.variable.publishParameters.attributeId = UA_ATTRIBUTEID_VALUE;
    UA_Server_addDataSetField(server, publishedDataSetIdent, &dataSetFieldConfig, &dataSetFieldIdent);  
}
```
```
static void 
addInformationModel(UA_Server *server) {
    /* create publish nodes from umati nodeset */
	// Data set field PubSubConnection
	UA_NodeId nodeId;
	UA_Variant var;
	UA_Variant_init(&var);
	
	UA_Double value = 0.0;
     
	 // Double
	 nodeId = UA_NODEID_NUMERIC(3, 1031);
	 UA_Variant_setScalar(&var, &value, &UA_TYPES[UA_TYPES_DOUBLE]);
	 addCustomDataSetField(server, var, &UA_TYPES[UA_TYPES_DOUBLE], nodeId, "FeedOverride");

	// Double
	nodeId = UA_NODEID_NUMERIC(3, 1037);
	UA_Variant_setScalar(&var, &value, &UA_TYPES[UA_TYPES_DOUBLE]);
	addCustomDataSetField(server, var, &UA_TYPES[UA_TYPES_DOUBLE], nodeId, "FeedOverride");
}
```
From the example pubsub_subscribe_standalone.c I only can get my published variables by comparing their data types.

```
        /* Loop over the fields and print well-known content types */
        for(int i = 0; i < dsm->data.keyFrameData.fieldCount; i++) {
            const UA_DataType *currentType = dsm->data.keyFrameData.dataSetFields[i].value.type;
            if (currentType == &UA_TYPES[UA_TYPES_DOUBLE]) {
                UA_Double value = *(UA_Double *)dsm->data.keyFrameData.dataSetFields[i].value.data;
                UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
                            "Message content: [Double] \tReceived data: %f", value);
            }
        }
```
But how can I get the published variables through their NodeIds? Because the UA_DataSetMessage only provides the NodeId of the builtin types.

```
		UA_NodeId nodeId_1031 = UA_NODEID_NUMERIC(3, 1031);
		UA_NodeId nodeId_1037 = UA_NODEID_NUMERIC(3, 1037);

        /* Loop over the fields and print well-known content types */
        for(int i = 0; i < dsm->data.keyFrameData.fieldCount; i++) {
            if (UA_NodeId_equal(&currentType->typeId, &nodeId_1031)) {
                UA_Double value = *(UA_Double *)dsm->data.keyFrameData.dataSetFields[i].value.data;
                UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
                            "Message content: [3, 1031] \tReceived data: %f", value);
            } else if (UA_NodeId_equal(&currentType->typeId, &nodeId_1037)) {
                UA_Double value = *(UA_Double *)dsm->data.keyFrameData.dataSetFields[i].value.data;
                UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
                            "Message content: [3, 1037] \tReceived data: %f", value);
            }
        }
```
Or is there another way how to subsribe published variable by their NodeId?

--------------------------------------------------------------------------------

## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=FULL -DUA_ARCHITECTURE=POSIX -DUA_ENABLE_AMALGAMATION=ON ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master branch
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: POSIX
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

ljocha85, 2020-03-05:
Is my understanding correct, that the OPC UA Subcriber is still WIP and that only Meta Data Subcriber is possible.

As you can see in the picture.

[image]

chuongvhn, 2020-03-18:
what you need can be found in the tutorial_[pubsub_subscribe.c](https://github.com/open62541/open62541/blob/master/examples/pubsub/tutorial_pubsub_subscribe.c) example.  
You need more than Node ID to subscribe : PublisherID, wirterGroup, DatasetWriter ...
https://github.com/open62541/open62541/blob/6c0c79c2bd2d0faa7d243a3098531ca728a7df9c/examples/pubsub/tutorial_pubsub_subscribe.c#L95

chuongvhn, 2020-03-18:
Currently you can subscribe to node has built-in datatype only, custom datatype is not supported yet.
https://github.com/open62541/open62541/issues/3337

chuongvhn, 2020-03-23:
> 
> 
> Is my understanding correct, that the OPC UA Subcriber is still WIP and that only Meta Data Subcriber is possible.
> 
> As you can see in the picture.
> 
> [image]

Could you please provide me link to this image ? 
Thanks

Aljoschka, 2020-03-23:
https://github.com/open62541/open62541/blob/master/FEATURES.md

Aljoschka, 2020-03-30:
@chuongvhn Do you have the same impression that the Subcricer is still under WIP? Because if I get two UA_Int32 in the Subcriber for example, I can't tell which integer value is from which NodeId in the Information Model.

chuongvhn, 2020-04-30:
@Aljoschka : according to OPC-UA spec, we can subscribe to a node by specifying PublisherID, WriterGroupID and DataSetID but I'm skeptical whether open6251 has implemented that. Yes, you're right PubSub is still WIP, especially Subscriber : subscribed for custom data type, subscribe by a set of PublisherId, WriterGroupId, DataSetId....

NoelGraf, 2021-05-28:
@ljocha85 is something still open?

tejashreeKate, 2021-08-18:
Is this still open? Can we subscribe  to custom datatype?

teccheck, 2025-04-29:
Closing as there's been massive overhauls (see [here](https://github.com/open62541/open62541/blob/master/FEATURES.md#opc-ua-pubsub)) and this is covered by CI now. Feel free to reopen if needed

closed 2025-04-29 by teccheck