open62541/open62541 #3451: client :  acknowledge or disable an alarm & condition

kikia1b2, 2020-02-20:
Is there  an example for client to acknowledge or disable an alarm? 
Use function UA_Client_writeEventNotifierAttribute(UA_Client *client, const UA_NodeId nodeId,
                                      const UA_Byte *newEventNotifier)?
 But I have no idea about how to set these paras to acknowledge or disable an alarm?   
Thank you .

kikia1b2, 2020-02-20:
@ccvca

schroeder-, 2020-02-20:
Call the "Acknowledge""(NodeId: i=9111) or "Disable"(NodeId: i=9028) method of the Datatype AcknowledgeableConditionType (NodeId: i=2881)  with the EventNotifier of the event.

kikia1b2, 2020-02-22:
@schroeder- in tutorial_client_event.c,  alarm attributes (like eventId ) can be got in handler_events(), but how to get the alarm's objectId or conditionId which will be used in the following function to acknowledge the alarm??
	UA_StatusCode retval = UA_Client_call(client, objectId, methodId, 2, &input,  &outputSize,  &output);
Thank you so much.

schroeder-, 2020-02-24:
For the ConditionId add to your filter: 
```
  selectClauses[x].typeDefinitionId = UA_NODEID_NUMERIC(0, UA_NS0ID_BASEEVENTTYPE);
    selectClauses[x].browsePathSize = 1;
    selectClauses[x].browsePath = (UA_QualifiedName*)
        UA_Array_new(selectClauses[x].browsePathSize, &UA_TYPES[UA_TYPES_QUALIFIEDNAME]);
    if(!selectClauses[x].browsePath) {
        UA_SimpleAttributeOperand_delete(selectClauses);
        return NULL;
    }
    selectClauses[x].attributeId = UA_ATTRIBUTEID_VALUE;
    selectClauses[x].browsePath[0] = UA_QUALIFIEDNAME_ALLOC(0, "ConditionId");
```
Now you will get the ConditionId as NodeId in **handler_events**.

kikia1b2, 2020-02-24:
@schroeder- ConditionId seems not to be a filed atrribute. Nothing is returned in eventFileds[x]. 
Attached is the modified  tutorial_client_event.c as you said.
Thank you so much.
[tutorial_client_events.zip](https://github.com/open62541/open62541/files/4243608/tutorial_client_events.zip)

schroeder-, 2020-02-24:
Sry my bad, ConditionId is a special case you get it with this filter:

```
	selectClauses[i].typeDefinitionId = UA_NODEID_NUMERIC(0, UA_NS0ID_CONDITIONTYPE);
	selectClauses[i].browsePathSize = 0;
	selectClauses[i].browsePath = NULL;
	selectClauses[i].attributeId = UA_ATTRIBUTEID_NODEID;
```

kikia1b2, 2020-02-26:
[test_alarm.zip](https://github.com/open62541/open62541/files/4254812/test_alarm.zip)
@schroeder- Sry to bother you again. In server,  I created two unexposed conditions with UA_Server_createCondition() with different ConditionId, but in client (uaExpert), two conditions have the same ConditionId which result in that only one line of alarm is shown.  

I downloaded open62541 on Feb 7, 2020.  The ua_subscription_alarms_conditions.c source file is the newest.
Attached is the used easy server file. It's modified from tutorial_server_alarms_conditions.c
Thank you so much,

sqadasi, 2020-03-03:
@kikia1b2 it is a bug in the implementation of A&C. Conditions and RefreshEvents end up having the same ConditionId (although RefreshEvents shouldn't have any ConditionId).
Please see: https://github.com/open62541/open62541/pull/3415#issuecomment-582816205

NoelGraf, 2021-05-13:
The pull request #3540 was merged and should have fixed the problem.

closed 2021-05-13 by NoelGraf