open62541/open62541 #3452: sendSymmetricChunk() fails assertion

luibass92, 2020-02-20:
## Description

Hello everyone,

I am using open62541 on VxWorks (real-time OS) and I am also using the historical data plugin.

I have an historized node of about 1000 samples all in memory.
Each sample is about 250 Bytes that means approximately 250 KB in total.

From a client I try to read that node and, since the max sendBufferSize size is 64 KB (default), the 250 KB must be fragmented in 4 chunks.

Unfortunately something bad occurs that causes an assertion fail. In particular I can read on the shell of my OS the following message:

`Assertion failed: total_length <= connection->config.sendBufferSize, file open62541.c, line 20483
`

This assertion is inside the `sendSymmetricChunk()` function and I cannot figure out how to solve the problem.

The error disappear if I reduce my buffer from 1000 to 220 samples, this because the total weight decreases to approximately 55 KB and so it does not need to be split in chunks.

The same code, even with 1000 samples, works perfectly on more performing hardwares (with same OS).

I use the library to perform other jobs like "normal" reading of nodes through UA_ReadRequest and UA_ReadResponse and I did not faced any problem before.

I tried to look into the code by myself but this part is deep in the lower levels of the library and I am having serious troubles, so I hope someone who worked on the snedSymmetricChunk function can help me :)

Find attached a wireshark screenshot taken while the problem occurs.

[image]

## Background Information / Reproduction Steps

VxWorks OS works with "Priority First" scheduling and the server side of the library run
 on a single thread so I am not expecting to face preemption problems caused by other threads with a higher priority.

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 1.1
 - [ ] Other OPC UA SDKs used (client or server): historical data plugin
 - [ ] Operating system: VxWorks (real-time OS)
 - [ ] Wireshark network dump attached
 - [ ] Critical issue

jpfr, 2020-02-21:
Ouch. We have to fix that.

Looking at the code snippet, you are on v1.0.1 and not v1.1 (current master). Please confirm.

Do you have encryption enabled?

luibass92, 2020-02-21:
Sorry I did a mistake, I am using version 0.4 of the library, I have just updated my previous post.
Yes, encryption is enabled.