open62541/open62541 #3453: GetEnpoints localeIds and profileUris encoding

HenningNT, 2020-02-20:
I have a question about the encoding of the GetEndpointRequest.

I'm using code from the master branch, commit c178eb4361e922b52c40cb137871b9a6d398584a, Mon Feb 17 23:39:20 2020 +0100, compiled with the amalgamation option.
I use this code, from the examples folder:

    UA_Client *client = UA_Client_new();
    UA_ClientConfig_setDefault(UA_Client_getConfig(client));

    /* Listing endpoints */
    UA_EndpointDescription *endpointArray = NULL;
    size_t endpointArraySize = 0;
    UA_StatusCode retval = UA_Client_getEndpoints(client, "opc.tcp://localhost:4840",
                                                  &endpointArraySize, &endpointArray);
    if(retval != UA_STATUSCODE_GOOD) {
        UA_Array_delete(endpointArray, endpointArraySize,
                        &UA_TYPES[UA_TYPES_ENDPOINTDESCRIPTION]);
        UA_Client_delete(client);
        return EXIT_FAILURE;
    }
    printf("%i endpoints found\n", (int)endpointArraySize);

I run the example server in the examples folder.
When the code completes it prints `1 endpoints found`.

When I run it against the server that I am going to use it prints out `0 endpoints found`. The UAExpert software do find the endpoints on this server.

I have used UAExpert and WireShark to capture and compare the requests:
[image]
The difference is how the fields localIds and profileUris are encoded. UAExpert uses zero, but open62541 encodes these fields as -1.

I am not sure what is correct, as the spec seems a bit vague about these fields.
Please advise on what to do about it. I'd write a fix for it, but I'm not sure what needs fixing, if anything.

jpfr, 2020-02-22:
Both are valid.
Length -1 means undefined.
Length 0 means defined with length zero.

Who is the vendor of the Server?

NoelGraf, 2021-05-13:
@HenningNT what is the status of the issue?

teccheck, 2025-04-29:
It seems like there hasn't been any recent activity or any further response, so I'll close this

closed 2025-04-29 by teccheck