open62541/open62541 #3454: Server freezes and continues only after Ctrl + C

maximilianpreisinger, 2020-02-20:
## Description

First of all: Many thank for this great and open OPC UA Library!

Now to the problem: On Windows 10 (x64) when I run a OPC UA Server (with V. 1.0.1 of this library), I have a separate update-thread which uses the callbacks `UA_Variant_setArray` and `UA_Server_writeValue` to update the values of my nodes. It runs in a loop, which repeats every 2s. I also write a `printf` statement to see weather this loop actually is running.
I also connected the signal Ctrl c to a stop handler, which stops the Server.

However, when I start the Server, at some point the update-thread seems to freeze, because there are no more printf statements coming. To me it seems, that `UA_Variant_setArray` or `UA_Server_writeValue` is stuck somewhere. When I then press Ctrl c, the server continues (printf statements are coming again). This is a bit counter-intuitive, since I thought Ctrl c would call the stop handler, and then stops the Server.
Instead, it seems that the Server gets stuck somewhere and waits for a Ctrl c (overwriting my stop handler signal), and then continues.

I have no clue, why the Server gets stuck in the first place.  I would be very grateful if someone could point me out how to further debug this, or giving me some hints.

## Background Information / Reproduction Steps

Unfortunately, this behaviour seems to be random and not reproduce-able. When I start the server.exe 10 times and let it run for a minute, it occurs maybe once.

It seems that this issue only occurs in Windows. I also compiled the Library and the Server on Ubuntu and I never saw this behavior. Furthermore, I tested both the v1.0.0 and v1.0.1 version of the library, and both are affected. It also seems that this issue is unrelated to an actual connection to the server. It also occurs, when no client is connected.

Used CMake options:

I am using cmake with the default options, and UA_ENABLE_AMALGAMATION (I include the library as open62541.c and open62541.h in my project).

This is a typical output log from the startup to the end of the program. However, when this Freeze occurs, there is no additional logging information (even on Level 100):

```
[2020-02-20 14:41:07.467 (UTC+0100)] warn/server        Username/Password configured, but no encrypting SecurityPolicy. This can leak credentials on the network.
[2020-02-20 14:41:07.467 (UTC+0100)] debug/session      Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: No value given; Copy the value from the TypeDefinition
[2020-02-20 14:41:07.468 (UTC+0100)] debug/session      Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: No value given; Copy the value from the TypeDefinition
[2020-02-20 14:41:07.468 (UTC+0100)] debug/session      Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: No value given; Copy the value from the TypeDefinition
[2020-02-20 14:41:07.470 (UTC+0100)] debug/session      Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: No value given; Copy the value from the TypeDefinition
[2020-02-20 14:41:07.481 (UTC+0100)] info/network       TCP network layer listening on opc.tcp://COMPUTER:4840/
[2020-02-20 14:41:10.560 (UTC+0100)] info/userland      received ctrl-c
[2020-02-20 14:41:10.563 (UTC+0100)] info/network       Shutting down the TCP network layer
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.0.1 and v1.0.0 (occurs with both versions)
 - [x] Other OPC UA SDKs used (client or server): None, only using open62541 Server
 - [x] Operating system: Windows x64 (on Linux, this issue seems not to occur)
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached (Level 100)
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

## Code Example
In the following my minimal code example. I add a new array type to the server, and then create 4 nodes of this type, which I update in a separate thread in a loop. If there are suggestions about my code, and how to speak to this library more properly, they are very welcome. However, first of all I really need to get rid of this freezing behaviour... 

```
#include <thread>
#include <csignal>
#include <ctime>
using namespace std;

extern "C" {
    #include "windows/open62541.h"
}

#define SLEEP_TIME 2s
#define NODE_ID_OFFSET 0
int arr_size = 10;
static UA_NodeId nodeIds[4];
static volatile UA_Boolean running = true;
static UA_NodeId arrayTypeId;

static void stopHandler(int sig) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "received ctrl-c");
    running = false;
}

void thread_function(UA_Server *ptr) {
    UA_Server *server = ptr;
    while (running==1) {
        printf("\nExecuting Thread Loop\n");
        for(int i=1; i<=4; i++) {
                UA_Variant array_value;
                UA_Double *array = new UA_Double[arr_size];
                UA_Variant_setArray(&array_value, array, arr_size, &UA_TYPES[UA_TYPES_DOUBLE]);
                UA_Server_writeValue(server, UA_NODEID_NUMERIC(1, NODE_ID_OFFSET+i), array_value);
                delete[] array;
        }
        this_thread::sleep_for(SLEEP_TIME);
    }
}

static void addArrayType(UA_Server *server) {
    char *qaname = strdup("ArrayType");
    UA_QualifiedName qaname_arrayType = UA_QUALIFIEDNAME(1, qaname);
    UA_VariableTypeAttributes vtAttr = UA_VariableTypeAttributes_default;
    vtAttr.dataType = UA_TYPES[UA_TYPES_DOUBLE].typeId;
    vtAttr.valueRank = UA_VALUERANK_ONE_DIMENSION;
    UA_UInt32 arrayDims[1] = {(UA_UInt32)arr_size};
    vtAttr.arrayDimensions = arrayDims;
    vtAttr.arrayDimensionsSize = 1;
    char *lang = strdup("en_US");
    char *displName = strdup("Array Variable Type");
    char *descr = strdup("This is a variable type which represents a one dimensional array");
    vtAttr.displayName = UA_LOCALIZEDTEXT(lang, displName);
    vtAttr.description = UA_LOCALIZEDTEXT(lang, descr);

    UA_Double *default_value = new UA_Double[arr_size];
    for (int i=0; i<arr_size; i++) default_value[i]=0;
    UA_Variant_setArray(&vtAttr.value, default_value, arr_size, &UA_TYPES[UA_TYPES_DOUBLE]);

    UA_Server_addVariableTypeNode(server, UA_NODEID_NULL, UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE), 
                                    UA_NODEID_NUMERIC(0,UA_NS0ID_HASSUBTYPE), qaname_arrayType, UA_NODEID_NULL, vtAttr, NULL, &arrayTypeId);
    delete[] default_value;
}

static void addArray(UA_Server *server, int nr) {
    char str_qaname[20];
    sprintf(str_qaname, "QA Name");
    UA_QualifiedName qaname = UA_QUALIFIEDNAME(1, str_qaname);
    UA_NodeId nodeId = UA_NODEID_NUMERIC(1, NODE_ID_OFFSET + nr);
    UA_VariableAttributes vAttr = UA_VariableAttributes_default;
    vAttr.dataType = UA_TYPES[UA_TYPES_DOUBLE].typeId;
    vAttr.valueRank = UA_VALUERANK_ONE_DIMENSION;
    UA_UInt32 arrayDims[1] = {(UA_UInt32)arr_size};
    vAttr.arrayDimensions = arrayDims;
    vAttr.arrayDimensionsSize = 1;
    char display[20];
    char description[20];
    sprintf(display, "Display Name");
    sprintf(description, "Description");
    char *lang = strdup("en_US");
    vAttr.displayName = UA_LOCALIZEDTEXT(lang, display);
    vAttr.description = UA_LOCALIZEDTEXT(lang, description);
    vAttr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;

    UA_Server_addVariableNode(server, nodeId, 
                                UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                                UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
                                qaname, arrayTypeId, vAttr, NULL, &nodeIds[nr-1]);
}

int main(int argc, char* argv[]) {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    UA_Server *server = UA_Server_new();
    UA_ServerConfig_setDefault(UA_Server_getConfig(server));

    addArrayType(server);
    for (int i=1; i<=4; i++) {
        addArray(server, i);
    }

    thread t(thread_function, server);
    UA_StatusCode retval = UA_Server_run(server, &running);
    t.join();

    UA_Server_delete(server);
    return retval == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}
```

basyskom-meerkoetter, 2020-02-21:
please attach a debugger, wait for the "freeze" and then collect backtraces for all threads. please post the result.

jpfr, 2020-02-21:
Please make sure to set `UA_MULTITHREADING` to a value bigger than 100.
Otherwise you can run into race conditions when the parallel thread writes into the information model.

maximilianpreisinger, 2020-02-27:
Sorry for my late reply, and thanks for your suggestions! I will try both and let you know about the outcome.

maximilianpreisinger, 2020-03-05:
On Linux, adding UA_ENABLE_MULTITHREADING=ON works well, but if I try to compile my project on Windows, I get now:

`open62541.h(929,12): error C3861: '_InterlockedExchangeSub': identifier not found`

Do I need to link to a windows library? I could not find an answer in the internet so far...

maximilianpreisinger, 2021-02-19:
Short update on this one. As I worked on this project last year, I used the versions 1.0.1 and 1.1-dev respectively.
Now a year later I upgraded this library to 1.1.5 and with the CMake options
```
UA_ENABLE_AMALGAMATION=ON
UA_LOGLEVEL=100
UA_MULTITHREADING=199
```
it compiles well within my project.

I did not test it enough, though, as to tell if the original bug is still there. I will observe it and give feedback here.

(I used the Amalgamation option against the recommendation from the docs, because I was not able to get the library running on Windows from the downloaded binary package (DLL) (don't know how to include the binary in my CMake project / if I try I get several errors...). If someone could point me to a guidance on how to use the binaries under Windows that would be great. But for now, even the Amalgamation option seems to work fine)

maximilianpreisinger, 2021-03-23:
I let run the code during the weekend, no problems occurred. As soon as I messed with the connected OPCUA client (UxExpert), I observed the exact same behaviour: Server freezes, pressing CTRL+C does not quit it, but makes the server to resume the operation. A second CTRL+C quits the server...

However, I was not able to reproduce this behaviour ever since. The issue seems to be far less likely than before the library upgrade.

The next step would be to utilize a debugger and a backtrace, as suggested by basyskom, however currently I don't have the resources to do that. Especially since this issue seems to be less likely and therefore is hard to reproduce.

alexandra42333, 2022-09-07:
Hi Maximilian, 

I am having the exact same issue you are describing above. I have no clue where this is coming from. Did you ever find the issue? If so, can you describe why this was happening and how you came up with a solution? 

Thanks a lot, Alexandra

teccheck, 2025-04-29:
The whole API is thread safe now if enabled ([see build options](https://www.open62541.org/doc/master/building.html#main-build-options)). Should issues reappear, feel free to open another issue or reopen this one if the same issue persists

closed 2025-04-29 by teccheck