open62541/open62541 #3461: Example client cannot connect to the Prosys OPC UA Simulation Server

luiztsmendes, 2020-02-24:
The example client from the open62541 tutorial "Building a simple client" (tutorial_client_firststeps.c in the examples folder) cannot connect to the Prosys OPC UA Simulation Server: it sends a request of type 459 and gets a "ServiceFault response with StatusCode BadServiceUnsupported". This problem does not happen in the stable 1.0.1 branch nor its 1.0 and 0.3 predecessors.

This issue is similar to the #3447 but @jadhavshital closed it just by knowing that the 0.3 branch solves his problems. Obviously, it needs to be fixed in the current master branch.

@luiztsmendes Thanks for the help. Now I am able to connect to Prosys OPC UA Simulation server using 0.3 branch code.

_Originally posted by @jadhavshital in https://github.com/open62541/open62541/issues/3447#issuecomment-590168948_

jpfr, 2020-02-25:
I don't have the Simulation Server installed atm.
Can you provide a log output (with an increased log fidelity) or even a Wireshark dump?

luiztsmendes, 2020-02-26:
@jpfr Thanks for looking at this issue. Here follows the client's output log and the zipped wireshark capture file as requested.

[image]

[TutorialClientFirstSteps_log.txt](https://github.com/open62541/open62541/files/4258205/TutorialClientFirstSteps_log.txt)
[TutorialClientFirstSteps2ProsysOPCUASim.pcapng.zip](https://github.com/open62541/open62541/files/4258207/TutorialClientFirstSteps2ProsysOPCUASim.pcapng.zip)

luiztsmendes, 2020-02-26:
Sorry, my mistake at pressing the wrong button

tumdash, 2020-08-06:
To whom it may concern:
Here is description of the problem on forum dedicated to simulator SW:
https://forum.prosysopc.com/forum/opc-ua-simulation-server/issue-while-connecting-new-version-of-prosys-5-0-0-217-due-to-latest-versions-of-bcp-jars/
Root cause of the issue is lack of ApplicationName in default client settings. 
So simple change of client as
```
    UA_Client *client = UA_Client_new();
    UA_ClientConfig *conf = UA_Client_getConfig(client);
    UA_ClientConfig_setDefault(conf);
    conf->clientDescription.applicationName = UA_LOCALIZEDTEXT_ALLOC("en", "TestClient");
    UA_StatusCode retval = UA_Client_connect(client, "opc.tcp://<OPCUA-Server>");
```
resolved the problem for me.

waltronix, 2020-08-07:
Thanks @tumdash!
 I found that out just now and wanted to post back :-D

@jpfr Does it make sense to always initialize the applicationName as it is done in the server settings?
Or do you know if this is no required by the standard?

o6-nbeier, 2026-10-21:
Closing for no activity.

closed 2026-10-21 by o6-nbeier