open62541/open62541 #3462: Creating Euromap 83 namespace fails:

mrspikeroot, 2020-02-25:
## Description

Hello.  I'm trying to create a server based on the Euromap 83 model, and getting the following errors when trying to create the namespace:
```
[2020-02-24 18:45:20.969 (UTC-0500)] info/session	Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: Parent node ns=2;i=1002 not found
[2020-02-24 18:45:20.969 (UTC-0500)] info/session	Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: The parent reference for ns=3;i=1009 is invalid with status code BadParentNodeIdInvalid
```

## Background Information / Reproduction Steps

This is how I created my generated code:

```bash
python /usr/share/open62541/tools/generate_datatypes.py --namespace=1 --type-csv=./UA-Nodeset/DI/OpcUaDiModel.csv --type-bsd=./UA-Nodeset/DI/Opc.Ua.Di.Types.bsd --no-builtin types_di

python /usr/share/open62541/tools/generate_datatypes.py --namespace=2 --type-csv=./euromap/Opc_Ua.EUROMAP83.RC1_01.NodeId.csv --type-bsd=./euromap/generated/Opc_Ua.EUROMAP83.1_01.DataTypes.bsd --no-builtin types_eu83

python /usr/share/open62541/tools/nodeset_compiler/nodeset_compiler.py --internal-headers --types-array=UA_TYPES --types-array=UA_TYPES_DI --existing ./UA-Nodeset/Schema/Opc.Ua.NodeSet2.xml --xml ./UA-Nodeset/DI/Opc.Ua.Di.NodeSet2.xml namespace_di

python /usr/share/open62541/tools/nodeset_compiler/nodeset_compiler.py --internal-headers --types-array=UA_TYPES --types-array=UA_TYPES_DI --types-array=UA_TYPES_EU83 --existing ./UA-Nodeset/Schema/Opc.Ua.NodeSet2.xml --existing ./UA-Nodeset/DI/Opc.Ua.Di.NodeSet2.xml --xml ./euromap/Opc_Ua.EUROMAP83.1_01.NodeSet2.xml namespace_eu83
```

And my main code:

```
int main(void) {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    UA_Server *server = UA_Server_new();
    UA_ServerConfig_setDefault(UA_Server_getConfig(server));

    /* create nodes from nodeset */
    UA_StatusCode retval = namespace_eu83(server);
    if (retval != UA_STATUSCODE_GOOD) {
        UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "Could not add the example nodeset. "
                                                           "Check previous output for any error.");
        retval = UA_STATUSCODE_BADUNEXPECTEDERROR;
    } else {
        retval = UA_Server_run(server, &running);
    }
    UA_Server_delete(server);
    return (int) retval;
}
```

These errors occur when I call namespace_eu83(), which returns UA_STATUSCODE_BADPARENTNODEIDINVALID.
The function starts like this, in case it sheds any light on the issue:
```
UA_StatusCode namespace_eu83(UA_Server *server) {
UA_StatusCode retVal = UA_STATUSCODE_GOOD;
/* Use namespace ids generated by the server */
UA_UInt16 ns[3];
ns[0] = UA_Server_addNamespace(server, "http://opcfoundation.org/UA/");
ns[1] = UA_Server_addNamespace(server, "http://opcfoundation.org/UA/DI/");
ns[2] = UA_Server_addNamespace(server, "http://www.euromap.org/euromap83/");
```

The node in Opc_Ua.EUROMAP83.1_01.NodeSet2.xml that it seems to be choking on is
```
    <UAObjectType NodeId="ns=1;i=1009" BrowseName="1:MachineInformationType">
        <DisplayName>MachineInformationType</DisplayName>
        <Description>General description of the machine</Description>
        <References>
            <Reference ReferenceType="HasProperty">ns=1;i=6052</Reference>
            <Reference ReferenceType="HasProperty">ns=1;i=6310</Reference>
            <Reference ReferenceType="HasProperty">ns=1;i=6053</Reference>
            <Reference ReferenceType="HasSubtype" IsForward="false">ns=2;i=1002</Reference>
            <Reference ReferenceType="HasProperty">ns=1;i=6152</Reference>
        </References>
    </UAObjectType>
```

I have a hunch this is related to namespace indexes, but I don't entirely understand how this is supposed to work.  I'm probably just doing something wrong, but any help would be greatly appreciated.  
Thanks! 

Used CMake options:

```bash
BUILD_SHARED_LIBS:BOOL=ON
CLANG_FORMAT_EXE:FILEPATH=CLANG_FORMAT_EXE-NOTFOUND
CMAKE_BUILD_TYPE:STRING=RelWithDebInfo
CMAKE_INSTALL_PREFIX:PATH=/usr
UA_ARCHITECTURE:STRING=posix
UA_BUILD_EXAMPLES:BOOL=OFF
UA_BUILD_TOOLS:BOOL=OFF
UA_BUILD_UNIT_TESTS:BOOL=OFF
UA_ENABLE_AMALGAMATION:BOOL=OFF
UA_ENABLE_DA:BOOL=ON
UA_ENABLE_DISCOVERY:BOOL=ON
UA_ENABLE_DISCOVERY_MULTICAST:BOOL=OFF
UA_ENABLE_ENCRYPTION:BOOL=OFF
UA_ENABLE_HISTORIZING:BOOL=OFF
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_MICRO_EMB_DEV_PROFILE:BOOL=OFF
UA_ENABLE_NODEMANAGEMENT:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_ALARMS_CONDITIONS:BOOL=OFF
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=OFF
UA_ENABLE_WEBSOCKET_SERVER:BOOL=OFF
UA_LOGLEVEL:STRING=300
UA_MULTITHREADING:STRING=0
UA_NAMESPACE_ZERO:STRING=FULL
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): master branch, a97cdb31b8a7c5ca429bf90b55db9360e8b56c9f
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: Fedora 31 and Ubuntu 19.10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [

 ] Self-contained code example attached
 - [ ] Critical issue

mrspikeroot, 2020-02-25:
Possibly related issues that might be clues:

During compilation, 
```bash
Building C object CMakeFiles/open62541-object.dir/src_generated/open62541/namespace0_generated.c.o
```
takes a very long time, and 
```bash
Linking C shared library bin/libopen62541.so
```
takes even longer and gives me the following info:
```bash
open62541/build/src_generated/open62541/namespace0_generated.c: In function ‘namespace0_generated’:
open62541/build/src_generated/open62541/namespace0_generated.c:113178:15: note: variable tracking size limit exceeded with ‘-fvar-tracking-assignments’, retrying without
113178 | UA_StatusCode namespace0_generated(UA_Server *server) {
       |               ^
[100%] Built target open62541
```

mrspikeroot, 2020-02-25:
Another possibly interesting piece of info:

The last step of generating the nodeset gives me warnings:

```bash
INFO:__main__:Preprocessing (existing) ./UA-Nodeset/Schema/Opc.Ua.NodeSet2.xml
INFO:__main__:Preprocessing (existing) ./UA-Nodeset/DI/Opc.Ua.Di.NodeSet2.xml
INFO:__main__:Preprocessing ./euromap/Opc_Ua.EUROMAP83.1_01.NodeSet2.xml
WARNING:nodes:Unknown Field Attribute ArrayDimensions
WARNING:nodes:Unknown Field Attribute ArrayDimensions
WARNING:nodes:Unknown Field Attribute ArrayDimensions
WARNING:nodes:Unknown Field Attribute ArrayDimensions
WARNING:nodes:Unknown Field Attribute ArrayDimensions
WARNING:nodes:Unknown Field Attribute ArrayDimensions
INFO:__main__:Generating Code for Backend: open62541
INFO:__main__:NodeSet generation code successfully printed
```

jpfr, 2020-02-25:
It could be that the reference to the parent
```
            <Reference ReferenceType="HasSubtype" IsForward="false">ns=2;i=1002</Reference>
```
is not recognized as such.
Can you show the generated code for `ns=1;i=1009`?

mrspikeroot, 2020-02-25:
Here it is:

```
/* MachineInformationType - ns=2;i=1009 */

static UA_StatusCode function_namespace_eu83_87_begin(UA_Server *server, UA_UInt16* ns) {
UA_StatusCode retVal = UA_STATUSCODE_GOOD;
UA_ObjectTypeAttributes attr = UA_ObjectTypeAttributes_default;
attr.displayName = UA_LOCALIZEDTEXT("", "MachineInformationType");
#ifdef UA_ENABLE_NODESET_COMPILER_DESCRIPTIONS
attr.description = UA_LOCALIZEDTEXT("", "General description of the machine");
#endif
retVal |= UA_Server_addNode_begin(server, UA_NODECLASS_OBJECTTYPE,
UA_NODEID_NUMERIC(ns[2], 1009),
UA_NODEID_NUMERIC(ns[1], 1002),
UA_NODEID_NUMERIC(ns[0], 45),
UA_QUALIFIEDNAME(ns[2], "MachineInformationType"),
 UA_NODEID_NULL,
(const UA_NodeAttributes*)&attr, &UA_TYPES[UA_TYPES_OBJECTTYPEATTRIBUTES],NULL, NULL);
return retVal;
}

static UA_StatusCode function_namespace_eu83_87_finish(UA_Server *server, UA_UInt16* ns) {
return UA_Server_addNode_finish(server, 
UA_NODEID_NUMERIC(ns[2], 1009)
);
}

```

zx-zx-zx, 2021-01-04:
> Another possibly interesting piece of info:
> 
> The last step of generating the nodeset gives me warnings:
> 
> ```shell
> INFO:__main__:Preprocessing (existing) ./UA-Nodeset/Schema/Opc.Ua.NodeSet2.xml
> INFO:__main__:Preprocessing (existing) ./UA-Nodeset/DI/Opc.Ua.Di.NodeSet2.xml
> INFO:__main__:Preprocessing ./euromap/Opc_Ua.EUROMAP83.1_01.NodeSet2.xml
> WARNING:nodes:Unknown Field Attribute ArrayDimensions
> WARNING:nodes:Unknown Field Attribute ArrayDimensions
> WARNING:nodes:Unknown Field Attribute ArrayDimensions
> WARNING:nodes:Unknown Field Attribute ArrayDimensions
> WARNING:nodes:Unknown Field Attribute ArrayDimensions
> WARNING:nodes:Unknown Field Attribute ArrayDimensions
> INFO:__main__:Generating Code for Backend: open62541
> INFO:__main__:NodeSet generation code successfully printed
> ```

Hello, I also encounter this problem. Have you solved it

C:\Users\ZXT\Desktop\OPC UA\open62541\open62541\tools\nodeset_compiler>python ./nodeset_compiler.py --types-array=UA_TYPES --e ../../deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml -e ../../deps/ua-nodeset/DI/Opc.Ua.Di.NodeSet2.xml --xml ../../deps/ua-nodeset/PlasticsRubber/GeneralTypes/1.02/Opc.Ua.PlasticsRubber.GeneralTypes.NodeSet2.xml namespace_plasticsrubber
INFO:__main__:Preprocessing (existing) ../../deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml
INFO:__main__:Preprocessing (existing) ../../deps/ua-nodeset/DI/Opc.Ua.Di.NodeSet2.xml
INFO:__main__:Preprocessing ../../deps/ua-nodeset/PlasticsRubber/GeneralTypes/1.02/Opc.Ua.PlasticsRubber.GeneralTypes.NodeSet2.xml
WARNING:nodes:Unknown Field Attribute ArrayDimensions
WARNING:nodes:Unknown Field Attribute ArrayDimensions
WARNING:nodes:Unknown Field Attribute ArrayDimensions
WARNING:nodes:Unknown Field Attribute ArrayDimensions
WARNING:nodes:Unknown Field Attribute ArrayDimensions
WARNING:nodes:Unknown Field Attribute ArrayDimensions
INFO:__main__:Generating Code for Backend: open62541
INFO:__main__:NodeSet generation code successfully printed

GuptaKishan, 2021-01-14:
Hi,
I am facing the same issue even for this simple example. I have tried several ways to solve the problem but I am getting the same error (AddNode (ns=2;i=1): The parent reference for is invalid). Even for this simple case...

<UAObjectType NodeId="ns=1;i=0" BrowseName="1:TestServer">
    <DisplayName>TestServer</DisplayName>
    <References>
      <Reference ReferenceType="HasSubtype" IsForward="false">i=58</Reference>
      <Reference ReferenceType="HasComponent">ns=1;i=1</Reference>
    </References>
  </UAObjectType>
  <UAVariable NodeId="ns=1;i=1" BrowseName="1:Device" ParentNodeId="ns=1;i=0" DataType="String">
    <DisplayName>Device</DisplayName>
    <References>
      <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=0</Reference>
      <Reference ReferenceType="HasTypeDefinition">i=63</Reference>
      <Reference ReferenceType="HasModellingRule">i=78</Reference>
    </References>

Is there any problem with the assigned node number?

GuptaKishan, 2021-01-14:
Hi,
I am facing the same issue even for this simple example. I have tried several ways to solve the problem but I am getting the same error (AddNode (ns=2;i=1): The parent reference for is invalid). Even for this simple case...

///  <UAObjectType NodeId="ns=1;i=0" BrowseName="1:TestServer">
///    <DisplayName>TestServer</DisplayName>
///    <References>
///      <Reference ReferenceType="HasSubtype" IsForward="false">i=58</Reference>
///      <Reference ReferenceType="HasComponent">ns=1;i=1</Reference>
///    </References>
///  </UAObjectType>
///  <UAVariable NodeId="ns=1;i=1" BrowseName="1:Device" ParentNodeId="ns=1;i=0" DataType="String">
///    <DisplayName>Device</DisplayName>
///    <References>
///      <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=0</Reference>
///      <Reference ReferenceType="HasTypeDefinition">i=63</Reference>
///      <Reference ReferenceType="HasModellingRule">i=78</Reference>
///    </References>
///  </UAVariable>

Is there any problem with the assigned node number?

GuptaKishan, 2021-01-14:
///  '''<UAObjectType NodeId="ns=1;i=0" BrowseName="1:TestServer">'''
///    '''<DisplayName>TestServer</DisplayName>'''
///    <References>
///      <Reference ReferenceType="HasSubtype" IsForward="false">i=58</Reference>
///      <Reference ReferenceType="HasComponent">ns=1;i=1</Reference>
///    </References>
///  </UAObjectType>
///  <UAVariable NodeId="ns=1;i=1" BrowseName="1:Device" ParentNodeId="ns=1;i=0" DataType="String">
///    <DisplayName>Device</DisplayName>
///    <References>
///      <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=0</Reference>
///      <Reference ReferenceType="HasTypeDefinition">i=63</Reference>
///      <Reference ReferenceType="HasModellingRule">i=78</Reference>
///    </References>
///  </UAVariable>

GuptaKishan, 2021-01-14:
'''<UAObjectType NodeId="ns=1;i=0" BrowseName="1:TestServer">'''

GuptaKishan, 2021-01-14:
(UAObjectType NodeId="ns=1;i=0" BrowseName="1:TestServer")

GuptaKishan, 2021-01-14:
&lt;UAObjectType NodeId="ns=1;i=0" BrowseName="1:TestServer"&lt;

GuptaKishan, 2021-01-14:
&lt;UAObjectType NodeId="ns=1;i=0" BrowseName="1:TestServer"&gt;

GuptaKishan, 2021-01-14:
Hi,
I am facing the same issue even for this simple example. I have tried several ways to solve the problem but I am getting the same error (AddNode (ns=2;i=1): The parent reference for is invalid). Even for this simple case...
  &lt;UAObjectType NodeId="ns=1;i=0" BrowseName="1:TestServer"&gt;
    &lt;DisplayName&gt;TestServer&lt;/DisplayName&gt;
    &lt;References&gt;
       &lt;Reference ReferenceType="HasSubtype" IsForward="false"&gt;i=58&lt;/Reference&gt;
      &lt;Reference ReferenceType="HasComponent"&gt;ns=1;i=1&lt;/Reference&gt;
    &lt;/References&gt;
  &lt;/UAObjectType&gt;
  &lt;UAVariable NodeId="ns=1;i=1" BrowseName="1:Device" ParentNodeId="ns=1;i=0" DataType="String"&gt;
    &lt;DisplayName&gt;Device&lt;/DisplayName&gt;
    &lt;References&gt;
      &lt;Reference ReferenceType="HasComponent" IsForward="false"&gt;ns=1;i=0&lt;/Reference&gt;
      &lt;Reference ReferenceType="HasTypeDefinition"&gt;i=63&lt;/Reference&gt;
      &lt;Reference ReferenceType="HasModellingRule"&gt;i=78&lt;/Reference&gt;
    &lt;/References&gt;
  &lt;/UAVariable&gt;

Is there any problem with the assigned node number?

Psoke, 2021-04-29:
Have you found a solution to the long compilation time and the error message concerning the namespace0_generated?

rfri4s, 2021-11-10:
I've got same problem. 

Make stills in this step after 12 hours:
[ 67%] Building C object CMakeFiles/open62541-object.dir/src_generated/open62541/namespace0_generated.c.o

I'm trying to compile in a Raspberry PI 2 B with Raspbian buster

closed 2026-10-20 by o6-nbeier