open62541/open62541 #3480: Getting error while running mDNS service

jdhanasekar, 2020-03-02:
When I run ```./discovery_server_multicast```, I am unable to see servers instead I am getting error 
```	Multicast DNS: failed setting IP_MULTICAST_IF to 42.42.42.42: Cannot assign requested address ```.  Please help me to solve this error.

[image]

cmake details:

```
cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_AMALGAMATION=ON -DUA_ENABLE_DISCOVERY=ON -DUA_ENABLE_DISCOVERY_MULTICAST=ON 
```

Operating System : Ubuntu 18.04 Server

RakshanPremsagar, 2020-12-06:
Hello, did you manage to understand this? I would like to know as well

teccheck, 2025-04-29:
This is covered by CTT now, so I expect this to be solved. Also, there's been lots of changes on this topic. Feel free to reopen if necessary

closed 2025-04-29 by teccheck