open62541/open62541 #3481: Connection errors between the client and the server in a virtualiyed environement

u1234587, 2020-03-02:
hello all, 
i am using the last version of Open62541 and the xen hypervisor 4.6. i am running my server in vm-server which runs on a Ubuntu 16.04.6 LTS, the client is running on another virtual machine called vm-client and running on Ubuntu 16.04.6 LTS as well. I can ping to internet and between both of the virtual machines but i cannot connect OPC UA client to the OPC UA Sever, I get this error in the client side:
root@vm-client:~/open62541# ./myClient opc.tcp://vm-server:4840/
[2020-03-02 12:44:09.402 (UTC+0000)] warn/userland	AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2020-03-02 12:44:09.402 (UTC+0000)] info/client	Connecting to endpoint opc.tcp://vm-server:4840/
[2020-03-02 12:44:09.402 (UTC+0000)] info/client	SecurityPolicy not specified -> use default #None
[2020-03-02 12:44:09.405 (UTC+0000)] warn/network	DNS lookup of vm-server failed with error Unknown error
[2020-03-02 12:44:09.405 (UTC+0000)] error/client	Opening the TCP socket failed
[2020-03-02 12:44:09.405 (UTC+0000)] error/client	Couldn't connect the client to a TCP secure channel
- I have opened the port 4840, but still the same problem 
- I have tried a solution i saw here, which is to execute this ./myClient "opc.tcp://vm-server:4840/" but still not working. 
Can anyone help?
Thanks in advance

jpfr, 2020-03-02:
Can you „telnet vm-server 4840“?

u1234587, 2020-03-06:
Hello,
Yes, and i have deactivated the firewall, but when i do nmap i still see
that it is closed! When i try telnet on the opc ua port it’s working. Any
suggestions ?
Thanks in advance

DelphiUser, 2020-03-19:
I had a similar issue (DNS lookup failed) which was not related to a certain environment. It seems that within the function UA_ClientConnectionTCP_init a call to UA_initialize_architecture_network() is missing.

u1234587, 2020-03-19:
Hello,
Thanks for the answer, but how can i solve that?
Thanks in advance

DelphiUser, 2020-03-19:
I just changed that function at the open62541.c file and recompiled it

u1234587, 2020-03-19:
> AO344DCJRTB2E5X3KK37OXDRIHUTDANCNFSM4K7TTUZQ> . <https://github.com/
> notifications/beacon/AO344DEYNXN6JNUPBBYIWPTRIHUTDA
> 5CNFSM4K7TTUZ2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5

Hello,
Is it possible to include your new source code?which includes that function
?
Thank you

NoelGraf, 2021-05-13:
The issue was fixed by the following pull request #3610.

closed 2021-05-13 by NoelGraf