open62541/open62541 #3486: undefined ref to `__signbitf' error (freeRTOS-LWIP, porting to NXP freescale)

andyrose-kr, 2020-03-04:
## Description
Hi,
I am trying to port open62541 to NXP freescale devkit. There were some errors in the process, but most of them were solved simply, and I'll post it later. I have another errors to solve, and want to get help.

When I build my project for the first time, some messages can be shown:
```bash
#pragma message: Integer endianness could not be detected to be little endian. Use slow generic encoding.
#pragma message: No native IEEE 754 format detected. Use slow generic encoding.
```

When I use 'UA_ServerConfig_setMinimalCustomBuffer( )' in my source code, then there are 2 errors.
```bash
In function 'Double_encodeBinary'
open62541.c:9087: undefined reference to `__signbitd'

In function 'Float_encodeBinary'
open62541.c:9055: undefined reference to `__signbitf'
```

I still can not find a solution, how can I fix it?
## Background Information / Reproduction Steps
At first, I build open62541.h/c in ubuntu 18.04. I followed the instructions in the following link. 
https://open62541.org/doc/current/building.html#freertos-lwip
https://github.com/open62541/open62541/pull/2511

Then, I added open62541.h/c to my NXP project and build it(S32DS C compiler: powerpc-eabivle-gcc). I create a task for opc ua server as follows:
```bash
void task_opcua_server( void *pv ) {

	UA_Server *server = UA_Server_new();
	UA_ServerConfig *config = UA_Server_getConfig(server);
	
	UA_UInt32 sendBufSize = 16000;
	UA_UInt32 recvBufSize = 16000;
	UA_ServerConfig_setMinimalCustomBuffer(config, 4840, NULL, sendBufSize, recvBufSize); <===
	.....
}
```

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [v ] open62541 Version (1.1):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [v ] Operating system: freeRTOSLWIP
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-03-07:
`signbit` is part of `<math.h>`.
See here: https://pubs.opengroup.org/onlinepubs/9699919799/basedefs/math.h.html

You might have to add math support in your compiler settings.

Or, if support is incomplete, you can add this yourself.
Assuming that floats are in the IEEE754 representation, you can access the bit for 32-bit float directly:
```c
int signbit = ((uint32_t) x & 0x80000000) >> 31;
```

NoelGraf, 2021-05-13:
@andyrose-kr what is the status of the issue?

teccheck, 2025-04-29:
Porting concerns are overhauled due to the event loop, and we are working on FreeRTOS support again. As there's been no recent activity, I'll close this issue

closed 2025-04-29 by teccheck