open62541/open62541 #3487: Run docker container with customized open62541 server on Raspberry Pi

tlichten-95, 2020-03-04:
## Description
I am connected to my raspberry pi 1 ( via putty and I have already followed the steps as described [here](https://open62541.org/doc/current/building.html) at "Building Debian Packages inside Docker Container with CMake on Ubuntu or Debian"

I have cloned the docker-deb-builder repo and build both the dockerfile-ubuntu-18.04 and
dockerfile-ubuntu-17.04.
Then, I have cloned the open62541 repo and checked out pack/1.0.
Afterwards, I have created the output folder in the docker-deb-builder folder and tried to run the 
following command in the development folder:
`./build -i docker-deb-builder:18.04 -o output ~/ma/development/open62541`

But it fails with following error message:
```
pi@raspberrypi:~/ma/dockerVersion/docker-deb-builder $ sudo ./build -i docker-deb-builder:18.04 -o output ~/ma/dockerVersion/open62541
Running docker:
docker run -it -it -v /home/pi/ma/development/open62541:/source-ro:ro -v /home/pi/ma/development/docker-deb-builder/output:/output -v /home/pi/ma/development/docker-deb-builder/build-helper.sh:/build-helper.sh:ro -e USER=0 -e GROUP=0 --rm  docker-deb-builder:18.04 /build-helper.sh
Unable to find image 'docker-deb-builder:18.04' locally
docker: Error response from daemon: pull access denied for docker-deb-builder, repository does not exist or may require 'docker login': denied: requested access to the resource is denied.
See 'docker run --help'.
```

What I am trying to accomplish is to run a docker container (with an open62541 server running in it) on my raspberry pi. 
Questions:
- How can I solve the problem described above?
- How or where can I integrate my customized open62541 server code?

## Background Information / Reproduction Steps
Running the command cat /etc/os-release gives me the following information:

```
pi@raspberrypi:~/ma/dockerVersion/docker-deb-builder $ cat /etc/os-release                                PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-03-07:
It looks docker tries to pull in external dependencies but cannot connect to its server.
Is the Raspi connected to the internet?

tlichten-95, 2020-03-09:
@jpfr hey julius, the raspberry pi is connected to the internet. Meanwhile, I have also tried to clone the open62541 repo and then run the dockerfile. But this also does not work. I have then saved and copied the image from my windows pc (where building and running the open62541 image worked) to the raspberry pi1. There, I have loaded the image using the docker load command. Running the image however causes the following error:

`standard_init_linux.go:211: exec user process caused "exec format error"`

Any suggestions on how to solve this? My guess is that the architectures for which the image is built do not match. If this is the case, how can I build the image for the arm architecture?

closed 2025-02-04 by andreasebner