open62541/open62541 #3493: Compiling with Multithreading on Windows

maximilianpreisinger, 2020-03-06:
## Description

When I compile my project using the open62541 library (amalgamation version) with UA_ENABLE_MULTITHREADING=ON (for v1.0.1) or with UA_MULTITHREADING=199 (for v1.1-dev, as of 6.3.2020) on Windows, I get the following error:

`open62541.h(929,12): error C3861: '_InterlockedExchangeSub': identifier not found`

I think I have to link my project to some (Windows) library, which defines this identifier. But to which one?

I didn't find any information to this issue in the documentation or on the internet.

## Background Information / Reproduction Steps

Used CMake options:

UA_ENABLE_AMALGAMATION=ON
UA_MULTITHREADING=199 or UA_ENABLE_MULTITHREADING=ON
UA_LOGLEVEL=100

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.0.1 and v1.1-dev (as of 6.3.2020)
 - [X] Other OPC UA SDKs used (client or server): doesn't compile in the first place
 - [X] Operating system: Windows
 - [X] Logs (with `UA_LOGLEVEL` set as low as necessary) attached (see error message)
 - [X] Wireshark network dump attached (not applicable)
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-03-07:
Which compiler do you use?

maximilianpreisinger, 2020-03-09:
I am using "Visual Studio 16 2019" and cmake for Windows.

Actually (as of reading my description now again, I am not sure if my description is clear enough): Compiling the open62541 library on Windows with cmake and Visual Studio works well and generates the open62541.h and *.c file.
However, the error occurs when compiling my project, where I include those files.

jpfr, 2020-03-09:
These are compiler intrinsics of VS2019 for atomic operations.
https://docs.microsoft.com/de-de/cpp/intrinsics/interlockedexchangeadd-intrinsic-functions?view=vs-2019

There definition in open62541 is defined here:
https://github.com/open62541/open62541/blob/eba3a3881626f2b0af0c78729c1995e5a971b3f4/include/open62541/architecture_definitions.h#L390

I am surprised this happens at `UA_MULTITHREADING=199`.
As seen in the code it should not happen before 200.

Or maybe `_InterlockedExchangeSub` is used somewhere in the internal mutex implementation of VS2019.

maximilianpreisinger, 2020-03-10:
I spent quite some time for debugging and get it working, however without success.

It seems that the Multithreading feature is designed exclusively for Linux, not for Windows (can you confirm this??) :
From the link you posted it seems that VS knows only _InterlockedExchangeAdd, but not _InterlockedExchangeSub. Therefore I tried to trick it with
``` #define _InterlockedExchangeSub(a,b) _InterlockedExchangeAdd(a,-b)```
in the header file of the library.

This seems to work at the first glance, but then the compiler complains about a missing `pthread.h`
I found a library of pthread for windows (https://sourceware.org/pthreads-win32/) to make this header file and its functions available. However, now the compiler complains about a duplicate definition:
```
2>C:\path\to\project\libs\pthread\pthread.h(320,17): error C2011: 'timespec': 'struct' type redefinition
2>C:\Program Files (x86)\Windows Kits\10\Include\10.0.18362.0\ucrt\time.h(39): message : see declaration of 'timespec'
2>Done building project "project.vcxproj" -- FAILED.
```

My conclusion is, that the library is not properly configured for windows, and I cannot fix that easily.  I really would appreciate if someone could help me making Multithreading available for Windows, since I really need this feature for our project, unless I find another solution without an additional thread that interacts with the opcua server.

jixian79, 2020-04-01:
@maximilianpreisinger  I found a function in <winbase.h>,  which defined the `InterlockedExchangeSubtract` function。The definetion is 
```
unsigned long
InterlockedExchangeSubtract(_Inout_ _Interlocked_operand_ unsigned long volatile *Addend,
                            _In_ unsigned long Value) {
    return (unsigned long)_InterlockedExchangeAdd((volatile long *)Addend, -(long)Value);
}
```
And I added it to `architecture_definitions.h`， and replaced the `_InterlockedExchangeSub` with it , it seemed  ok.
But I can't resolve the `pthread` library relationship. I  use cmake-gui to generate the open62541 build project and  install pthread by vcpkg. but it doesn't work.
Could you please tell me how to configure the  pthread library ?

jixian79, 2020-04-03:
@maximilianpreisinger 
finally  I found a solution , in  cmake-gui, set UA_COMPILE_AS_CXX, then change the `_InterlockedExchangeSub` with `InterlockedExchangeSubtract`.  In this way, can successfully compile the multi-thread library in Windows .

jixian79, 2020-04-03:
@jpfr

jpfr, 2020-06-25:
Great it worked out.

closed 2020-06-25 by jpfr

vladp, 2021-05-05:
I am also running into this problem on the same platform (windows 10), even with most recent VS compiler tools.

The above suggested fixes are not working for me.
Compiler cannot find neither `_InterlockedExchangeSub` nor `InterlockedExchangeSubtract`

Using master branch:

>git rev-parse --short HEAD
cf0d5ef9

>git log -1
```
commit cf0d5ef962498a007b3548314cabe4b6710e17e7 (HEAD -> master, origin/master, origin/HEAD)
Author: Julius Pfrommer <julius.pfrommer@web.de>
Date:   Fri Apr 30 08:26:57 2021 +0200

    refactor(core): Simplify the padding logic for asymmetric chunks

```
cmake command:

` cmake  "Visual Studio 16 2019" -A x64 -T v142 -DUA_COMPILE_AS_CXX=ON -DUA_MULTITHREADING=100 -DUA_ENABLE_SUBSCRIPTIONS=ON -DUA_ENABLE_METHODCALLS=ON  -DUA_ENABLE_NODEMANAGEMENT=ON   -DUA_ENABLE_PUBSUB=ON -DUA_NAMESPACE_ZERO=FULL ..\ `

official-minister, 2021-09-08:
Oh, no. I get same error too. Is the problem solved？

vladp, 2021-09-12:
@official-minister no.  This is marked as 'help-wanted'. See  https://github.com/open62541/open62541/issues/4504