open62541/open62541 #3495: Procedure Wiki for installing open62541 with prebuild binaries is wrong

luigicapogrosso, 2020-03-09:
## Description
Hello all,
I think that there is a problem in the Wiki explanation for the library installation on Ubuntu 18.04.1 LTS.

## Reproduction Steps
I have done these steps, in according with the Wiki page:
```
sudo add-apt-repository ppa:open62541-team/ppa
sudo apt-get update
sudo apt-get install libopen62541-dev
```
but, I got the following error:
```
E: Unable to locate package libopen62541-dev
```
So, I checked the last modified version (2020-01-19 19:37) from the PPA:
```
libopen62541-1-dev_20200119T184421~1.0~33~g54e2b367~ubuntu18.04.1_amd64.deb  
```
Then, running the command:
```
sudo apt-get install libopen62541-1-dev
``` 
everything worked properly. I think this thing needs to be changed in the Wiki.

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [X] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-03-09:
Looks good. Can you add the changes?
Just say so if we need to give more access rights for the wiki.
Should work out of the box though.

NoelGraf, 2021-05-13:
I close the issue as it has been included in the documentation.

closed 2021-05-13 by NoelGraf