open62541/open62541 #3496: Interoperability: UA_VariableAttributes incomplete, missing role and access level attributes

TeunScheffel, 2020-03-09:
## Description

When trying to connect from a B&R PLC to an example-based Open62541 Server on Linux (using Anonymous login for now), the PLC fails to connect and reports **BadAttributeInvalid (0x80350000)**.

UA_Expert output snapshot:

[image]
In UA_expert the next four attributes are indicating the exact same error as the PLC reports:
- RolePermissions
- UserRolePermissions
- AccessRestrictions
- AccessLevelEx

The current v1.0.1. UA_VariableAttributes structure does not (yet) hold a definition for these.

Is anyone already working on this maybe? ( And can I help testing? )

## Background Information / Reproduction Steps
V1.0.1. Open62541 in amalgamated form, compiled on/for Linux
UA-Expert application or B&R PLC that is expecting these to e correctly filled

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): v1.0.1.
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-03-10:
These are new attributes that have been added in v1.04 of the spec.

Maybe it helps if we return BADNOTIMPLEMENTED for now.

mlang-de, 2020-03-10:
Or Bad_NotSupported.

OPCUAUser, 2020-05-18:
Hi Community,

Is there a support for role based authorization in open62541. If its not supported, what is the roadmap for the same? I need to implement a user based authorization for the nodes.

For example, I have User/Roles A, B and C whereas the nodes are 1001, 1002.
User/Role A: can read/write the node 1001 & no permission for 1002
User/Role B: can read/write the node 1002 & no permission for 1001
User/Role A: can read/write the node 1001 & 1002 (all permissions)

Your support would be really helpful for me.

Thanks in advance!

piperoc, 2023-07-21:
It looks like no one answered in 3+ years...
I'm doing some digging, and the information is scarce (it's ok, it's an open source initiative -- not complaining). 

They created an _AccessControl_ plugin, you can see it in the example but there is not a lot of information. I was able to kind of make it work based on the examples: UserA can read and UserB cane read and write.

Not sure I would call this Role-Based authorization like the Unified SDK's (as an example) do. And it definitely does not set the those optional attributes (`AccessLevelEx`, `RolePermssions`, `UserRolePermissions`, `AccessRestrictions`).

It a stack-level API (like the history API) not really a toolkit. You'll have to pull in a lot of your own kung-fu...

Hopefully someone with inside knowledge can shed more light on this since I still get a few errors that I can't shake otherwise I'd be using this.

jpfr, 2023-07-27:
@mlang-de , good point.
We now return BadUnsupported as per https://github.com/open62541/open62541/pull/5904.

@piperoc , we can implemented role based access control via the current plugin interfaces.

The AccessControl plugin evaluates the login credentials and sets a context for the session.
You can define a custom struct for the void-ptr of the session context.
By defining roles in the context, this becomes available then whenever access rights for the individual nodes are evaluated.

You know a few OPC UA SDK. Do you have a favorite way of handling access control?
We can then think of making a more fleshed out sample implementation of the AccessControl plugin.

teccheck, 2025-04-29:
It seems like this is solved, as there's the access control plugin and `BadUnsupported` is returned for the mentioned attributes. Feel free to reopen, if needed

closed 2025-04-29 by teccheck