open62541/open62541 #3504: Custom information modeling - 'has an unknown target error'

jongunee, 2020-03-12:
## Description
I designed address space using UaModeler and tired to apply.
so I exported address space file as a xml file, changed name to server_nodeset.xml file and overwritted.
But It has an error.
I'll attach the server_nodeset.xml file source code below.

```
<?xml version="1.0" encoding="utf-8"?>
<UANodeSet xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd" xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd" xmlns:s1="http://yourorganisation.org/test/Types.xsd" xmlns:ua="http://unifiedautomation.com/Configuration/NodeSet.xsd" xmlns:xsd="http://www.w3.org/2001/XMLSchema">
    <NamespaceUris>
        <Uri>http://yourorganisation.org/test/</Uri>
    </NamespaceUris>
    <Aliases>
        <Alias Alias="Int32">i=6</Alias>
        <Alias Alias="Double">i=11</Alias>
        <Alias Alias="Organizes">i=35</Alias>
        <Alias Alias="HasModellingRule">i=37</Alias>
        <Alias Alias="HasTypeDefinition">i=40</Alias>
        <Alias Alias="HasSubtype">i=45</Alias>
        <Alias Alias="HasProperty">i=46</Alias>
        <Alias Alias="HasComponent">i=47</Alias>
        <Alias Alias="Range">i=884</Alias>
    </Aliases>
    <Extensions>
        <Extension>
            <ua:ModelInfo Tool="UaModeler" Hash="TAikd/eBPfGdGCgNNcfR6g==" Version="1.6.3"/>
        </Extension>
    </Extensions>
    <UAObjectType NodeId="ns=1;i=1006" BrowseName="1:ControllerEventType">
        <DisplayName>ControllerEventType</DisplayName>
        <References>
            **<Reference ReferenceType="HasSubtype" IsForward="false">i=2041</Reference> ------>error**
            <Reference ReferenceType="HasProperty">ns=1;i=6012</Reference>
            <Reference ReferenceType="HasProperty">ns=1;i=6013</Reference>
        </References>
    </UAObjectType>
    <UAVariable DataType="Int32" ParentNodeId="ns=1;i=1006" NodeId="ns=1;i=6012" BrowseName="1:state" AccessLevel="3">
        <DisplayName>state</DisplayName>
        <References>
            <Reference ReferenceType="HasModellingRule">i=78</Reference>
            <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
            <Reference ReferenceType="HasProperty" IsForward="false">ns=1;i=1006</Reference>
        </References>
    </UAVariable>
    <UAVariable DataType="Double" ParentNodeId="ns=1;i=1006" NodeId="ns=1;i=6013" BrowseName="1:Temperature" AccessLevel="3">
        <DisplayName>Temperature</DisplayName>
        <References>
            <Reference ReferenceType="HasModellingRule">i=78</Reference>
            <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
            <Reference ReferenceType="HasProperty" IsForward="false">ns=1;i=1006</Reference>
        </References>
    </UAVariable>
    <UAObjectType NodeId="ns=1;i=1003" BrowseName="1:ControllerType">
        <DisplayName>ControllerType</DisplayName>
        <References>
            <Reference ReferenceType="HasSubtype" IsForward="false">i=58</Reference>
            <Reference ReferenceType="HasComponent">ns=1;i=6003</Reference>
            <Reference ReferenceType="HasComponent">ns=1;i=7001</Reference>
            <Reference ReferenceType="HasComponent">ns=1;i=6004</Reference>
            <Reference ReferenceType="HasComponent">ns=1;i=7002</Reference>
            <Reference ReferenceType="HasComponent">ns=1;i=6005</Reference>
            <Reference ReferenceType="HasComponent">ns=1;i=6007</Reference>
        </References>
    </UAObjectType>
    <UAVariable DataType="Double" ParentNodeId="ns=1;i=1003" NodeId="ns=1;i=6003" BrowseName="1:PowerConsumption" AccessLevel="3">
        <DisplayName>PowerConsumption</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=2365</Reference>
            <Reference ReferenceType="HasModellingRule">i=78</Reference>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=1003</Reference>
[... 157 lines cut ...]
**Error**
Raise Exception("Reference " + str(ref) + " has an unknown target")
16>Exception: Reference ns=1;i=1006<--[ns=0;i=45]--ns=0;i=2041 has an unknown target

Any help would be appreciated?

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag):latest master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system:windows10
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x] Critical issue

opcua-tsn-team-kalycito, 2020-07-17:
Hi @jongunee ,

> Raise Exception("Reference " + str(ref) + " has an unknown target")
> 16>Exception: Reference ns=1;i=1006<--[ns=0;i=45]--ns=0;i=2041 has an unknown target

In open62541 stack, the sample server_nodeset.xml has nodeIds that depends on Opc.Ua.NodeSet2.Reduced.xml.

The above exception happens because the custom information model created by you using UaModeler (server_nodeset.xml) has the reference with the nodeId `ns=0;i=2041` but Opc.Ua.NodeSet2.Reduced.xml does not contain the nodeId `ns=0;i=2041`.

The following should solve:
**Step 1:** Replace the line ` DEPENDS "${CMAKE_SOURCE_DIR}/deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.Reduced.xml"`
 with line `DEPENDS "${CMAKE_SOURCE_DIR}/deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml"`

**Step 2:** Compile the CMake with -DUA_NAMESPACE_ZERO=FULL -DUA_BUILD_EXAMPLES=ON

Now the server will run successfully.

**For more information:**

- Free documentation: You can look at the open source documentation page: https://open62541.org/doc/current/
- Commercial Support: https://www.kalycito.com/development-support-packages/ 
- Paid online course: Practical introduction to OPC UA – code walk-through and examples in this course use the open62541 stack: https://opcfoundation.org/products/view/practical-introduction-to-opc-ua-part-i

NoelGraf, 2021-05-13:
@jongunee Could your problem be solved?

teccheck, 2025-05-20:
As there's been no further answer, I'll close this as completed

closed 2025-05-20 by teccheck