open62541/open62541 #3513: Array in method result structure serialized/padded wrongly

rhaberkorn, 2020-03-17:
## Description

Implementing an AutoID server using open62541, I came across problems with arrays as part of structures returned by the Scan() method. They are apparently wrongly serialized in the binary protocol.

The Scan() method returns RfidScanResult (cf. OPC UA for AutoID, p.24), which in turn may contain several RfidSighting (p.45) structures in an array. Or as expressed in the BSD file:

```xml
 <opc:StructuredType BaseType="tns:ScanResult" Name="RfidScanResult">
  <!-- NOTE: This is a workaround for the missing opc:Bit type - probably irrelevant here -->
  <!--opc:Field TypeName="opc:Bit" Name="LocationSpecified"/-->
  <!--opc:Field Length="31" TypeName="opc:Bit" Name="Reserved1"/-->
  <opc:Field TypeName="opc:UInt32" Name="Bits"/>
  <opc:Field SourceType="tns:ScanResult" TypeName="opc:CharArray" Name="CodeType"/>
  <opc:Field SourceType="tns:ScanResult" TypeName="tns:ScanData" Name="ScanData"/>
  <opc:Field SourceType="tns:ScanResult" TypeName="opc:DateTime" Name="Timestamp"/>
  <opc:Field SwitchField="LocationSpecified" SourceType="tns:ScanResult" TypeName="tns:Location" Name="Location"/>
  <opc:Field TypeName="opc:Int32" Name="NoOfSighting"/>
  <opc:Field LengthField="NoOfSighting" TypeName="tns:RfidSighting" Name="Sighting"/>
 </opc:StructuredType>
```

The schema generator generates an UA_RfidScanResult accordingly:

```c
typedef struct {
    UA_UInt32 bits;
    UA_String codeType;
    UA_ScanData scanData;
    UA_DateTime timestamp;
    UA_Location location;
    size_t sightingSize;
    UA_RfidSighting *sighting;
} UA_RfidScanResult;
```

The bug manifests whenever I add more than one RfidSighting to the RfidScanResult. The result you can see in this UaExpert screenshot:
[image]
The second sighting is apparently bogus. I've debugged the structure's contents of course - it's still perfectly fine when I store it into the output Variant. Neither does Valgrind show any kind memory corruption or the like.

I rather suspect a kind of padding issue after analyzing the network communication. You can refer to the [attached Wireshark log](https://github.com/open62541/open62541/files/4340889/array.pcapng.gz) in packet 13. The first RfidSighting ends in byte 00e7. But the second RfidSighting only starts at byte 00ec. There seem to be 4 padding bytes that are not accounted for and that are not expected by UaExpert's OPC UA stack. These 4 bytes are also present after the second RfidSighting instance.

This of course immediately raises the possibility of a 64-bit-only bug.
I haven't yet tried to recompile everything with `-m32`.

## Background Information / Reproduction Steps

Used CMake options:

```
BUILD_SHARED_LIBS:BOOL=ON
CLANG_FORMAT_EXE:FILEPATH=CLANG_FORMAT_EXE-NOTFOUND
CMAKE_BUILD_TYPE:STRING=Debug
CMAKE_INSTALL_PREFIX:PATH=/usr/local
UA_ARCHITECTURE:STRING=posix
UA_BUILD_EXAMPLES:BOOL=OFF
UA_BUILD_TOOLS:BOOL=ON
UA_BUILD_UNIT_TESTS:BOOL=OFF
UA_ENABLE_AMALGAMATION:BOOL=OFF
UA_ENABLE_DISCOVERY:BOOL=ON
UA_ENABLE_DISCOVERY_MULTICAST:BOOL=OFF
UA_ENABLE_ENCRYPTION:BOOL=ON
UA_ENABLE_HISTORIZING:BOOL=OFF
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_MICRO_EMB_DEV_PROFILE:BOOL=OFF
UA_ENABLE_NODEMANAGEMENT:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_ALARMS_CONDITIONS:BOOL=OFF
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=ON
UA_ENABLE_WEBSOCKET_SERVER:BOOL=OFF
UA_LOGLEVEL:STRING=100
UA_MULTITHREADING:STRING=0
UA_NAMESPACE_ZERO:STRING=FULL
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.0.1
 - [x] Other OPC UA SDKs used (client or server): UaExpert 1.5.1 331
 - [x] Operating system: Linux 4.15.0-88-generic #88~16.04.1-Ubuntu SMP Wed Feb 12 04:19:15 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [x] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-03-17:
I just added a test to verify chunking on v1.0.1. See #3515.
It appears to be fine for arrays of integer type.

Can you make a source dump available on a private channel for easier debugging?
julus.pfrommer@iosb.fraunhofer.de

jpfr, 2020-03-17:
Ah right. This is not a chunking issue, as the messages are rather small.

In addition to the `UA_RfidScanResult` structure and its members, there are UA_DataType structures generated by the NodeSet compiler. Can you just share these?

rhaberkorn, 2020-03-17:
Of course, I can!

I think it's easiest to upload the entire folder of files generated from the standard namespaces: [src_generated.tar.gz](https://github.com/open62541/open62541/files/4345587/src_generated.tar.gz)

I've looked into `types_autoid_generated.c` before but couldn't figure out what's wrong.

jpfr, 2020-03-17:
Found it. The Companion Spec makes us of structures with optional fields.

This feature has just been completed.
Will be merged in #3503 in time for the v1.1.

Let's get back to this case when the infrastructure is in place.

jpfr, 2020-03-17:
General note: The nodeset compiler should abort in such cases instead of generating incomplete representations.

rhaberkorn, 2020-03-19:
> General note: The nodeset compiler should abort in such cases instead of generating incomplete representations.

I don't quite understand. RfidSighting doesn't contain any optional fields and I removed all of them from RfidScanResult making all fields mandatory instead... or so I thought. Although I haven't touched `Opc.Ua.AutoID.NodeSet2.xml`. However, I wouldn't be surprised as I haven't studied these beasts in depth.
(And I don't want to - it's a horribly over-engineered and redundant modelling language. The BSD/CSV/NodeSet2.xml are not even meant to be manually edited AFAIK, being intermediate languages instead...)

If you can give me a quick pointer on what to change in the NodeSet2, that would be appreciated. Otherwise I'll just wait for v1.1 and hope that the generator will process it without modifications. Finding a workaround is not super urgent for me.

jpfr, 2020-05-08:
A few days ago, support for optional fields in structures has landed.
Can you try with the current master?

rhaberkorn, 2020-05-12:
PS: In both v1.0 and v1.1-rc1, the autoid_nodeids.h are unusable since they contain duplicate and conflicting macro definitions. I worked around this by manually picking the right the defintions into my source file.

rhaberkorn, 2020-05-13:
I suspect that lines like that in namespace_autoid_generated.c are responsible:
```c
variablenode_ns_2_i_6052_variant_DataContents[0].dataType = UA_NODEID_NUMERIC(ns[1], 3010);
```
It should read - still did in files generated by v1.0 - `ns[2]`. In fact, when I manually tweak these autogenerated files, the namespaces are correct and the method InputArguments point to the right namespace.

Unfortunately, I cannot isolate the code in nodeset_compiler responsible for generating these faulty lines, nor do I see any _obvious_ change in nodeset_compiler since v1.0 that could explain the difference.

jpfr, 2020-05-21:
@ichrispa would you have a look at the namespace compiler?

apisapiaDL, 2020-07-02:
Hello,
I had the same problem with the wrong namespace in the input and output arguments. It is described in #3533 

Thank you

jpfr, 2020-07-02:
Okay. Giving this a high priority.
@ichrispa are you aware of this one in the nodeset compiler?