open62541/open62541 #3516: Difference in addPubSubConnection() in pubsub example

Wooodys, 2020-03-17:
## Description
Hi, I want to program a server, which will publish sensor data und subscribe to incoming information at the same time. But I face some issues. By using the examples i was able to create a publish and subscribe server. But if i want to build both features into one server, I get the  error " Dataset reader not found. Check PublisherID, WriterGroupID and DatasetWriterID". I guess the problem is the UA_PubSubConnection_regist in addPubSubConnection. Its used in the subscribe example but not in the publish example. I dont understand what this call does. Or should I open another PubSubConnection?

My second issue is that I am not understanding how I could redirect the incoming Messages into a specific node. I know that UA_Server_DataSetReader_addTargetVariables gernerates the nodes, but is it possible to decide, in which nodeID the data is stored.

Thanks for any answer or help.

Wooodys

## Background Information / Reproduction Steps
Checkout master

Used CMake options:
Commandline options:
-DBIBER_COMPILER:FILEPATH="BIBER_COMPILER-NOTFOUND" -DUA_MULTITHREADING:STRING="0" -DUA_NAMESPACE_ZERO:STRING="FULL" -DUA_VALGRIND_INTERACTIVE_INTERVAL:STRING="1000" -DXELATEX_COMPILER:FILEPATH="XELATEX_COMPILER-NOTFOUND" -DXINDY_COMPILER:FILEPATH="XINDY_COMPILER-NOTFOUND" -DCMAKE_BUILD_TYPE:STRING="Debug" 

Cache file:
BIBER_COMPILER:FILEPATH=BIBER_COMPILER-NOTFOUND
UA_MULTITHREADING:STRING=0
UA_NAMESPACE_ZERO:STRING=FULL
UA_VALGRIND_INTERACTIVE_INTERVAL:STRING=1000
XELATEX_COMPILER:FILEPATH=XELATEX_COMPILER-NOTFOUND
XINDY_COMPILER:FILEPATH=XINDY_COMPILER-NOTFOUND
CMAKE_BUILD_TYPE:STRING=Debug

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): master
 - [ ] Other OPC UA SDKs used (client or server): UAExpert as Client 
 - [ ] Operating system: Windows 10 Visual Studio 2017 express
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

waheedejaz, 2020-04-07:
> I guess the problem is the UA_PubSubConnection_regist in addPubSubConnection.

UA_PubSubConnection_regist() is not required for the publisher. 

> I get the error " Dataset reader not found.

Check the return value when adding the dataSetReader and confirm if it is successfully created.

> Or should I open another PubSubConnection?

Not required, a publisher and a subscriber can work on the same connection.

I can help more if you share the code.

ZekeLandon, 2020-05-04:
Hi @Wooodys,

Subscriber does not have enough information to parse the incoming message. You are missing the `networkMessageContentMask` in Publisher application. I think you need to add the [these lines](https://github.com/open62541/open62541/blob/a77b20ff940115266200d31d30d3290d6f2d57bd/examples/pubsub/server_pubsub_publisher_rt_level.c#L127-L130) in your writerGroup configuration.

Let me know if this solves your problem.

Regards,
Zeke

Wooodys, 2020-07-13:
Hi,
thanks @ZekeLandon. Your hint was right,

Regards,
Wooodys

ZekeLandon, 2020-07-13:
That's great! Happy coding, @Wooodys 

--Zeke

o6-nbeier, 2026-10-20:
Thank you for taking the time to open this issue.

Since it seems that it has been resolved, we are closing this for now.

closed 2026-10-20 by o6-nbeier