open62541/open62541 #3519: HistoryData plugin does not call backend if server not running

juangburgos, 2020-03-19:
## Description

I configure the `historizingUpdateStrategy` with `UA_HISTORIZINGUPDATESTRATEGY_VALUESET` and while the server is running, if a variable's value is changed either programatically or by a client through the network, then the backend callbacks are called properly and history data is stored.

Though I have the issue that I setup my address space before starting the server, and during that setup I change some values to some variables (after they are created) for which I would like to have historical data. 

Sadly, the backend callbacks are not called during this setup stage (before the server is running). I don't know if I am doing something wrong.

If the node store is available to modify the address space "offline", being able to change variables' values. Why not call the history backend during those changes?

Most probably this is by design, but would like to understand why, it could be useful in some circumstances.

Thanks,

## Background Information / Reproduction Steps

Setup some variables with historizing and set their values before starting the server.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.0
 - [x] Other OPC UA SDKs used (client or server): UaExpert
 - [x] Operating system: Any

juangburgos, 2020-03-20:
@peter-rustler-basyskom you helped me once before. Please any light on this? :pray: I also have another doubt. Does the HistoryData pluginworks for events, A&E, etc?

MPJansen, 2020-03-23:
@juangburgos 

I just ran into the same issue (when trying to test locally monitored value callbacks) and found the underlying issue.
The server will handle callbacks in the server mainloop (aside from the first cb on creation apparently), when the server is stopped, obviously the mainloop of the server is not run.

You can call `UA_Server_run_iterate` to trigger a single server loop. Be mindful that most monitoring has a samplingInterval so my solution:

```
writeVar
sleep_for(50ms)
UA_Server_run_iterate()
receive callback
```

Hope this helps

KR.

MPJansen, 2020-03-23:
See for [example](https://github.com/open62541/open62541/blob/6aa4d8d9d2f76b0ba95acba06345ae38c3e4c068/tests/server/check_server_jobs.c#L41) the check_server_jobs test.

juangburgos, 2020-03-23:
Hey @MPJansen ! Your solution does makes sense, I will play with it. Although it is a *hack*, I would expect it would work just as it is. Thanks again.

andranik-s, 2020-03-24:
@juangburgos values should be historized even if the server is "offline", if `UA_HISTORIZINGUPDATESTRATEGY_VALUESET` is used. Just make sure to set up your historizing backend BEFORE writing any values. 
<details><summary>Some relevant code so you can check it for yourself</summary>

https://github.com/open62541/open62541/blob/0cd099611ed236bedc8572c1af45a43ee10b23e7/src/server/ua_services_attribute.c#L1194-L1205

https://github.com/open62541/open62541/blob/0cd099611ed236bedc8572c1af45a43ee10b23e7/plugins/historydata/ua_history_database_default.c#L594-L601

https://github.com/open62541/open62541/blob/0cd099611ed236bedc8572c1af45a43ee10b23e7/plugins/historydata/ua_history_data_gathering_default.c#L200-L207

</details>

As for your second question, history plugin supports events, but no default implementation is provided by `UA_HistoryDatabase_default`. Take a look at `UA_HistoryDatabase::setEvent` and `UA_HistoryDatabase::readEvent`.

juangburgos, 2020-03-28:
Thank you @andranik-s , I found out it was my mistake and resolved an issue at [QUaServer](https://github.com/juangburgos/QUaServer).

closed 2020-03-28 by juangburgos