open62541/open62541 #3527: history registerNodeId function have memory leak

jungyoungsoo, 2020-03-30:
## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [o] open62541 Version (release number or git tag): master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

sorry i don't speak english well. 
history have moemory leaks bug?
if call function update_data_test() increase memory.
never decrese memoy..
if many UA_Server_addVariableNode tag and update tag value from UA_Server_writeValue function.
continue increase memory... 

void update_data_test(UA_NodeId id, void* value, VlaueType type) {
    UA_VariableAttributes nAttr = UA_VariableAttributes_default;
    UA_UInt32 myUint32 = 11;
    UA_Variant_setScalar(&nAttr.value, &myUint32, &UA_TYPES[UA_TYPES_UINT32]);
    UA_StatusCode res = UA_Server_writeValue(g_server, id, nAttr.value);
    if (res != UA_STATUSCODE_GOOD) {
        UA_LOG_WARNING(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "writeValue failed. StatusCode %s", UA_StatusCode_name(res));
    }
}

void InitHistory() {
    UA_HistoryDataGathering g_gathering = UA_HistoryDataGathering_Default(1);
    g_config->historyDatabase = UA_HistoryDatabase_default(g_gathering);
    /* Define the attribute of the uint32 variable node */
    UA_VariableAttributes attr = UA_VariableAttributes_default;
    UA_UInt32 myUint32 = 40;
    UA_Variant_setScalar(&attr.value, &myUint32, &UA_TYPES[UA_TYPES_UINT32]);
    attr.description = UA_LOCALIZEDTEXT((char*)"en-US", (char*)"myUintValue");
    attr.displayName = UA_LOCALIZEDTEXT((char*)"en-US", (char*)"myUintValue");
    attr.dataType = UA_TYPES[UA_TYPES_UINT32].typeId;
    /* We set the access level to also support history read
     * This is what will be reported to clients */
    attr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE | UA_ACCESSLEVELMASK_HISTORYREAD;
    /* We also set this node to historizing, so the server internals also know from it. */
    attr.historizing = true;

    UA_NodeId uint32NodeId = UA_NODEID_STRING(1, (char*)"myUintValue");
    UA_QualifiedName uint32Name = UA_QUALIFIEDNAME(1, (char*)"myUintValue");
    UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
    UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
    UA_NodeId_init(&g_testHistory);
    UA_StatusCode retval = UA_Server_addVariableNode(g_server,
        uint32NodeId,
        parentNodeId,
        parentReferenceNodeId,
        uint32Name,
        UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE),
        attr,
        NULL,
        &g_testHistory);

    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER,
        "UA_Server_addVariableNode %s", UA_StatusCode_name(retval));

    UA_HistorizingNodeIdSettings setting;
    setting.historizingBackend = UA_HistoryDataBackend_Memory(1, 1);
    setting.maxHistoryDataResponseSize = 1000;
    setting.historizingUpdateStrategy = UA_HISTORIZINGUPDATESTRATEGY_VALUESET;
    /* At the end we register the node for gathering data in the database. */
    retval = g_gathering.registerNodeId(g_server, g_gathering.context, &g_testHistory, setting);
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "registerNodeId %s", UA_StatusCode_name(retval));
}

luibass92, 2020-05-29:
Hi,
the defaul history implementation is intended to be an "inifinte" 2D matrix, first dimension the values of each node that you want to historize, second dimension represents the nodes that you want to historize (so the number of registered nodes).

Let's suppose you just registered 1 Node to be historized (in this case you will have a Nx1 matrix). Every time the server writes a value to that particular node, this value must be stored somewhere in memory in order to be available if a client request the historical data of that particular node.

So it is normal that the memory grows at each write (if you use the default implementation of the history plugin).

teccheck, 2025-05-20:
The answer by @luibass92 looks good to me and there hasn't been a response since, so I'll close this as complete

closed 2025-05-20 by teccheck