open62541/open62541 #3528: Monitored Items are not deleted after network failure

Maerdl, 2020-03-30:
## Checklist

Please provide the following information:

 - [x] open62541 Version (): v1.0.1 https://github.com/open62541/open62541/releases/tag/v1.0.1
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x ] Operating system: custom embedded os
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Hey there,

i run into an issue and not sure if i'm right. I have a server running, and it does well. 
I have UA_Expert as client attached to the server. The UA_Expert registers some monitored items in the server. If i disconnect the client in a regular way(close the connection) the monitored items are deleted in the server.
If i interrupt the connection in a hard way(remove the network cable of the client) the send call in function ```connection_write``` of network_tcp fails.
```
n = UA_send(connection->sockfd,
                     (const char*)buf->data + nWritten,
                     bytes_to_send, flags);
            if(n < 0 && UA_ERRNO != UA_INTERRUPTED && UA_ERRNO != UA_AGAIN) {
                connection->close(connection);
                UA_ByteString_deleteMembers(buf);
                return UA_STATUSCODE_BADCONNECTIONCLOSED;
            }
```
```connection->close``` is called.
I have set the server config to accept only 40 monitored items.
If I repeat the hard connection interrupt a few times, depends on how many monitored items i have registered, the server later does not accept more monitored items, because ```UA_Subscription_deleteMembers``` is not called.
I can see if i do a normal close of the connection ```UA_Subscription_deleteMembers``` is called.

So the question is. Who is responsible to delete the monitored items ? The client or the Server if the session will be closed ?

Thanks 
Maerdl

mlang-de, 2020-03-30:
Hello,

> Who is responsible to delete the monitored items ? The client or the Server if the session will be closed ?
in your error case:
either by the session lifetime time out or the subscription lifetime time out, the realted monitoreditems shall be deleted. 
Normally:
Done by/during the DeleteSubscription or CloseSession service.
If the connection will be closed only the securechannel is closed and detached from the session.
The session itself still exists in the server.
After a short connection interruption the UaExpert tries to reuse the old session by calling the ActivateSession.
If your problem is not solved please add a wireshark record.

br

Maerdl, 2020-03-30:
Thanks for your anwer @mlang-de 

While you gave me the answer i read that too.
Maybe i have a wrong understanding of opc ua, in that case please correct me but i thought subscriptions can be reused as long as the session is not deleted and the lifetime is not over.

Can you tell me where i can have a look at to see if it is resued ?

Thanks again!
Maerdl

Maerdl, 2020-03-31:
Hi all,

i diged deeper and found out that each time the connection was interrupted and connect again a new session was created.
See the logs:
```
[1970-01-01 00:00:06.858 (UTC+0000)] info/channel Connection 5 | SecureChannel 1 | Session 412e09b7-0b4c-90f1-1e01-70b7a80a1081 created
Interrupt connection
[1970-01-01 00:01:38.185 (UTC+0000)] info/network       Connection 5 | New connection over TCP
[1970-01-01 00:01:38.245 (UTC+0000)] info/channel       Creating a new SecureChannel
[1970-01-01 00:01:38.246 (UTC+0000)] warn/securitypolicy        Security policy None is used to create SecureChannel. Accepting all certificates
[1970-01-01 00:01:38.248 (UTC+0000)] info/channel       Connection 5 | SecureChannel 2 | Opened SecureChannel
[1970-01-01 00:01:38.281 (UTC+0000)] info/session       Connection 0 | SecureChannel 0 | Session ns=1;g=412e09b7-0b4c-90f1-1e01-70b7a80a1081 | ActivateSession: Could not generate a server nonce
[1970-01-01 00:01:38.298 (UTC+0000)] info/channel       Connection 5 | SecureChannel 2 | Session 1a77e10e-77b3-c17b-9a79-2742fdff1f01 created
```
In ```UA_Session_generateNonce``` both, ```channel``` is NULL which results in
```return UA_STATUSCODE_BADINTERNALERROR;```

So the session can not be reused and it is created a new one. 
I think that is not correct.
Somebody any hints ?

Regards
Maerdl

mlang-de, 2020-03-31:
Maybe #3470 can help you?

Maerdl, 2020-03-31:
Hi @mlang-de ,

that exactly solves my issue.
See log:
```
1970-01-01 00:00:05.936 (UTC+0000)] info/channel       Connection 5 | SecureChannel 1 | Session 6dd1721f-237b-e6a6-1531-d37d2a92a94a created
Interrupt connection
[1970-01-01 00:01:22.108 (UTC+0000)] warn/securitypolicy        Security policy None is used to create SecureChannel. Accepting all certificates
[1970-01-01 00:01:22.110 (UTC+0000)] info/channel       Connection 5 | SecureChannel 2 | Opened SecureChannel
[1970-01-01 00:01:22.128 (UTC+0000)] info/session       Connection 0 | SecureChannel 0 | Session ns=1;g=6dd1721f-237b-e6a6-1531-d37d2a92a94a | ActivateSession: Attach to new channel
[1970-01-01 00:01:22.130 (UTC+0000)] info/session       Connection 5 | SecureChannel 2 | Session ns=1;g=6dd1721f-237b-e6a6-1531-d37d2a92a94a | ActivateSession: Session activated
```
Session and subscription are reused. Perfect, thanks !

Is it planned to merge this into master or a release ?

Thanks again
Mardl

mlang-de, 2020-03-31:
IMHO yes, but there is still some work on the PR to do.
@jpfr Will the PR #3470 back ported into V1.0.1?

closed 2021-05-06 by NoelGraf