open62541/open62541 #3530: PubSub unicast

Fischer-Thomas, 2020-03-31:
I just tried to run a server with PubSub unicast and got the message that PubSub creation failed (no multicast address). A few days ago it still worked even though there was the same warning.

https://github.com/open62541/open62541/blob/e2cea8bf405a1a63b1d11024f3b95d8d63ecbca9/plugins/ua_pubsub_udp.c#L114-L140

Is there a reason that unicast is not supported anymore or is it just untested?

jpfr, 2020-03-31:
This came in as part of PR #3505.
Should be easy to fix. We will still issue a warning though.

closed 2020-03-31 by jpfr

Fischer-Thomas, 2020-04-02:
Hello Julius,

thanks for the fast response.
I just pulled the current master and tried it again.
Sadly it doesn't work. It still tries to join a multicast group even though it recognizes it as unicast.

Here is my output:

[image]

PR #3531 (2020-03-31): fix(pubsub): Allow publishing on a unicast address
Only create a warning.  Fixes #3530.
commit 1b404328 (2020-03-31): fix(pubsub): Allow publishing on a unicast address

Only create a warning.  Fixes #3530.