open62541/open62541 #3532: Add vcpkg installation instructions

PhoebeHui, 2020-04-01:
Hi,

open62541 is available as a port in vcpkg, a C++ library manager that simplifies installation for open62541 and other project dependencies. Documenting the install process here will help users get started by providing a single set of commands to build open62541, ready to be included in their projects.

We also test whether our library ports build in various configurations (dynamic, static) on various platforms (OSX, Linux, Windows: x86, x64) to keep a wide coverage for users.

I'm a maintainer for vcpkg, and [here is what the port script looks like](https://github.com/microsoft/vcpkg/blob/master/ports/open62541/portfile.cmake) We try to keep the library maintained as close as possible to the original library.

I try to add vcpkg installation instructions but I couldn't find the related docs in github except for on https://open62541.org/doc/current/installing.html page.

Could you please help add the following contents to this port?

Installing and building open62541 via vcpkg

You can download and install open62541 using the [vcpkg](https://github.com/Microsoft/vcpkg) dependency manager:

      git clone https://github.com/Microsoft/vcpkg.git
      cd vcpkg
      ./bootstrap-vcpkg.sh
      ./vcpkg integrate install
      vcpkg install open62541

The open62541 port in vcpkg is kept up to date by Microsoft team members and community contributors. If the version is out of date, please [create an issue or pull request](https://github.com/Microsoft/vcpkg) on the vcpkg repository.

jpfr, 2020-04-02:
Sure, we can provide that.
Is it possible to have different versions in vcpkg?
With version-pinning or similar...

PhoebeHui, 2020-04-03:
@jpfr, thanks for your help!

It installed open62541 with latest release revision v1.0.1 now. The contributor or team member will update the version if it's out of date.
vcpkg also supports build open62541 from head in Upstream with '--head' option, however this is not test, there's no guarantee that works.