open62541/open62541 #3535: PubSub: Publish/Subscribe callback design (messageReceiveTimeout)

MartinaFelberFEM, 2020-04-02:
Hi!

I have had a look at the PubSub publish/subscribe callback implementation and was wondering how an application (particularly the Subscriber) can get any information about the status of sent and received network messages.

I have tested the PubSub implementation in a custom realtime environment with overwritten PubSub callback handling (CMAKE flag UA_ENABLE_PUBSUB_CUSTOM_PUBLISH_HANDLING). My use case is similar to your example: https://github.com/open62541/open62541/blob/master/examples/pubsub_realtime/TSN_ETF_loopback.c

At the subscriber thread routine the subscriber calls the callback to receive network data. 
https://github.com/open62541/open62541/blob/285637fc519e2aa92142cc62f280d0226da412d1/examples/pubsub_realtime/TSN_ETF_loopback.c#L647-L648

The callback definition does not provide a statuscode or any diagnostic parameter:
https://github.com/open62541/open62541/blob/285637fc519e2aa92142cc62f280d0226da412d1/include/open62541/server.h#L97

So, how can the Subscriber (and Publisher) know if anything went wrong?
I know that there's a PubSubDiagnosticsType specified by the Opc Foundation, which would provide some details [https://github.com/open62541/open62541/issues/3354](url), but in this use case 

- I need to immediately (realtime requirements) react to errors

-  and I need to know if subscribed DataSets have been received within configured MessageReceiveTimeout (see PubSub specification 6.2.8.6)
[image]

The messageReceiveTimeout can be configured for every DataSet. 

I thought about changing the Callback function defintion (adding status code and a diagnostic parameter or something) to enable the application to notice some errors (encryption/decryption, network message could not be parsed etc.) but I have a problem with the messageReceiveTimeout.

Because the subscribe callback handles a ReaderGroup. A ReaderGroup can contain multiple DataSets with different messageReceiveTimeouts. So I am not sure how to handle this.

https://github.com/open62541/open62541/blob/285637fc519e2aa92142cc62f280d0226da412d1/src/pubsub/ua_pubsub_reader.c#L280-L303

Many thanks for your help

opcua-tsn-team-kalycito, 2020-04-03:
Hello @MartinaFelberFEM ,
Subscriber is under development. We are currently working on the embedded target improvement using RT variables on the Subscriber side. We shall get back to this item after 2 weeks. How quick do you need this implementation?

MartinaFelberFEM, 2020-04-03:
Hi!
This feature is my top priority at the moment.
If you ask like that I would say as soon as possible :)
In general I think it's absolutely necessary, especially for realtime applications.

NoelGraf, 2021-05-06:
@MartinaFelberFEM  can the issue be closed? PR is already integrated.

MartinaFelberFEM, 2021-05-06:
yes, sorry!

closed 2021-05-06 by MartinaFelberFEM