open62541/open62541 #3536: Client not using selected endpoint

Arno-v, 2020-04-02:
## Description

## Background Information / Reproduction Steps

Used CMake options:

Don't remember them all, but at least with UA_ENABLE_ENCRYPTION. 
Used mbedtls-2.16.5

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag):1.0.1
 - [x] Other OPC UA SDKs used (client or server): None
 - [x] Operating system:Windows 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

I have created a server setup with two endpoints:

- With security policy "none"
- With security policy "Basic256Sha256" and SignAndEncrypt.

I have setup the access control on the server to refuse sessions without "Basic256Sha256". 
User token is anonymous.
This works fine from uaexpert. It gets the endpoints, and when I select the endpoint with SignAndEncrypt I can browse and subscribe.

When I use then open62541 client however, I can't connect. It gets the correct endpoint in the config, but it doesn't seem to use it. On the server side it tries to open a session with policy "None" which is refused.

I think it is related to this piece of code in ua_client_connect.c line 1006:
```C    if(!client->channel.securityPolicy) {
        UA_SecurityPolicy *sp =
            getSecurityPolicy(client, UA_STRING("http://opcfoundation.org/UA/SecurityPolicy#None"));
        if(!sp) {
            retval = UA_STATUSCODE_BADINTERNALERROR;
            goto cleanup;
        }
        UA_ByteString remoteCertificate = UA_BYTESTRING_NULL;
        retval = UA_SecureChannel_setSecurityPolicy(&client->channel, sp,
                                                    &remoteCertificate);
        if(retval != UA_STATUSCODE_GOOD)
            goto cleanup;
    }
```
This piece of code gets called during getendpoints, and after that the channel security is never reset.

Is there any way to force the client to use the endpoint with the "Basic256Sha256" policy?

Arno-v, 2020-04-03:
I finally got it running, but I had to patch the code (see attachment).
The other things I had to do:

- Write an own version of UA_ClientConfig_setDefaultEncryption witch also includes a list of issuers which is than passed to UA_CertificateVerification_Trustlist. Else the client will reject the server certificate.
- Set the securityMode of the client config to UA_MESSAGESECURITYMODE_SIGNANDENCRYPT.
- Set the securityPoliciUri of the client config to "http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256"
- Call UA_Client_connect. This will give an error return, but it will set the correct endpoint in the config.
- Call UA_Client_disconnect. This will cleanup the channel and the connection.
- Call UA_Client_connect again. This succeeds and sets up a secure connection.

I am not really happy with this, and I think there is a better solution, e.g. by integrating this in UA_Client_connect_iterate.

[ua_client_connect.c.patch.zip](https://github.com/open62541/open62541/files/4426300/ua_client_connect.c.patch.zip)

Arno-v, 2020-04-15:
Ok, I messed up. I was working in a wrong version.

closed 2020-04-15 by Arno-v

I-Yoo, 2020-04-22:
Did you test the method UA_Client_connect_username ？I used client.c in the folder "\open62541\examples\client.c" to do the experiment. But I noticed that its return code is 0x80020000. I hope you can help me solve this problem.

Arno-v, 2020-04-22:
@I-Yoo No. I was trying to use an encrypted connection which did work properly. Turned out I was using the master branch instead of a released version :-( My mistake. I did not use the user/password authentication.

wuhao1121, 2020-05-24:
> @I-Yoo No. I was trying to use an encrypted connection which did work properly. Turned out I was using the master branch instead of a released version :-( My mistake. I did not use the user/password authentication.

Hello, I met the same problem as you. How did you solve it? Thank you