open62541/open62541 #3544: Overriding log_stdout.h with a custom configuration file

saukijan, 2020-04-07:
## Description

Hello everyone, 

I have been working with open62541 SDK as a server and been looking into implementing a custom configuration for our OPC UA server implementation and part of that custom configuration would be a a custom Logging plugin.

I looked at your [Logging Plugin API](https://open62541.org/doc/1.0/plugin_log.html?highlight=logging) and implemented the following methods in a similar fashion as this pseudo code: 

```c
/* ============================ CustomLogger.h ============================ */
#include <open62541/plugin/log.h>

extern const UA_Logger Custom_Logger_;

void Custom_Logger_log(void *_, UA_LogLevel level, UA_LogCategory category,
                      const char *msg, va_list args);
void Custom_Logger_clear(void *logContext);

/* ============================ CustomLogger.c ============================ */
void Custom_Logger_log(void *_, UA_LogLevel level, UA_LogCategory category,
                      const char *msg, va_list args){
// custom logging api, that I can not show
}
void Custom_Logger_clear(void *logContext) {
// nothing to clear
}
// Forward declaration for ease of use 
const UA_Logger Custom_Logger_clear= {Custom_Logger_log, Custom_Logger_clear, NULL};
```

And do a simple start with my 'main.c' 
```c
#include "CustomLogger.h"
#include <open62541/server.h>

#include <signal.h>
#include <stdlib.h>

static volatile UA_Boolean running = true;
static void stopHandler(int sig) {
    UA_LOG_INFO(Custom_Logger_, UA_LOGCATEGORY_USERLAND, "received ctrl-c");
    running = false;
}

int main(void) {
 signal(SIGINT, stopHandler);
 signal(SIGTERM, stopHandler);

 UA_ServerConfig * configuration_ = (UA_ServerConfig *)malloc(sizeof(UA_ServerConfig)); 
 UA_ServerConfig_setDefault(configuration_ );
 configuration_->logger = Custom_Logger_;

 UA_Server* server = UA_Server_newWithConfig(configuration_);
 UA_StatusCode retval = UA_Server_run(server, &running);
 UA_Server_delete(server);

 return retval == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}
```

## The Problem

In Principal it works, however, not every single part of the SDK uses the new logging, a quick search in the github repository reveals that the following files use the default [log_stdout.h](https://github.com/open62541/open62541/blob/master/plugins/include/open62541/plugin/log_stdout.h) instead of a more generic [log.h](https://github.com/open62541/open62541/blob/master/include/open62541/plugin/log.h):

* [plugins/ua_network_pubsub_mqtt.c](https://github.com/open62541/open62541/blob/58bd161557111847d068650eff5ad670a9aa0395/plugins/ua_network_pubsub_mqtt.c)
* [plugins/mqtt/ua_mqtt_adapter.c](https://github.com/open62541/open62541/blob/e7d3e385e5bcec16a7967764490690879adde3f1/plugins/mqtt/ua_mqtt_adapter.c)
* [arch/network_ws.c](https://github.com/open62541/open62541/blob/af9767240c10f86b29261d8cdfe3abfd26a63345/arch/network_ws.c)
* [plugins/ua_pki_default.c](https://github.com/open62541/open62541/blob/f71912485ec60a42eebbfd96877bef5268c699e6/plugins/ua_pki_default.c)
* [plugins/ua_pubsub_ethernet.c](https://github.com/open62541/open62541/blob/6c0c79c2bd2d0faa7d243a3098531ca728a7df9c/plugins/ua_pubsub_ethernet.c)
* [plugins/securityPolicies/openssl/ua_pki_openssl.c](https://github.com/open62541/open62541/blob/19b3dc0945213764cdf460faabadbda38906c5fb/plugins/securityPolicies/openssl/ua_pki_openssl.c)
* [plugins/ua_pubsub_ethernet_etf.c](https://github.com/open62541/open62541/blob/c09da0b76dbe81657a428ae132b93ccfe940b4b6/plugins/ua_pubsub_ethernet_etf.c)
* [plugins/ua_config_default.c](https://github.com/open62541/open62541/blob/f71912485ec60a42eebbfd96877bef5268c699e6/plugins/ua_config_default.c)
* [plugins/ua_pubsub_udp.c](https://github.com/open62541/open62541/blob/be65f1202ca472ac4a40b8235f19c3461ece05d1/plugins/ua_pubsub_udp.c)
* [plugins/ua_pubsub_ethernet_xdp.c](https://github.com/open62541/open62541/blob/fbb4e0fe4d39fde449a6ab7cf7e4a7198a8e0b5f/plugins/ua_pubsub_ethernet_xdp.c)
* [arch/network_tcp.c](https://github.com/open62541/open62541/blob/cde2a4366246a15c0e25936fd179738663684ec3/arch/network_tcp.c)

## Possible Sollution

I can understand that [plugins/ua_config_default.c](https://github.com/open62541/open62541/blob/f71912485ec60a42eebbfd96877bef5268c699e6/plugins/ua_config_default.c) uses the default logger, however I suggest that other plugins and modules use the aforementioned [log.h](https://github.com/open62541/open62541/blob/master/include/open62541/plugin/log.h) header and get the pointer to the proper logging implementation from the `UA_ServerConfig *`, or explicitly required as an argument in the function footprint. Especially as the logging mechanism is implemented as follows:

```c
void UA_LOG_INFO(const UA_Logger *logger, UA_LogCategory category, const char *msg, ...);
```

An example of this call in [ua_pki_default.c](https://github.com/open62541/open62541/blob/f71912485ec60a42eebbfd96877bef5268c699e6/plugins/ua_pki_default.c#L162) could look like: 

```c
/* open62541/plugins/ua_pki_default.c */

:161 static UA_StatusCode 
:162 reloadCertificates(CertInfo *ci, const UA_Logger *logger){
...
:181 UA_LOG_INFO(logger, UA_LOGCATEGORY_SERVER,
:182                        "Failed to load certificate from %s", f);
...
}
```

Another option could be to force all Plugins to return a valid `UA_StatusCode`  and forbid the use of `UA_Logger *` in functions which do not have `UA_Logger *` or `UA_ServerConfig *` as their argument and handle the logging only within the functions that have these arguments.

This would allow the user to change the logging mechanism without having to patch parts of the SDK or even re-implement the desired functionality.

## Background Information / Reproduction Steps

Not relevant, but I will provide it regardless. 

Used CMake options:

```bash
cmake -DUA_LOGLEVEL="Trace" -DUA_ENABLE_SUBSCRIPTIONS=OFF-DUA_ENABLE_SUBSCRIPTIONS_EVENTS=OFF -DUA_ENABLE_METHODCALLS=ON -DUA_ENABLE_NODEMANAGEMENT=ON -DUA_ENABLE_AMALGAMATION=OFF  -DUA_ENABLE_MULTITHREADING=OFF -DUA_ENABLE_IMMUTABLE_NODES=OFF -DUA_ENABLE_DISCOVERY=ON -DUA_ENABLE_DISCOVERY_MULTICAST=OFF -DUA_ENABLE_DISCOVERY_SEMAPHORE=ON -DUA_NAMESPACE_ZERO="Reduced" -DUA_ENABLE_TYPENAMES=ON -DUA_ENABLE_STATUSCODE_DESCRIPTIONS=ON  ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): v1.0.1
 - [ ] Other OPC UA SDKs used (client or server): Server
 - [ ] Operating system: Debian 8
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-04-09:
Good proposal! I fully agree.

Let's treat this as a meta-issue where we track the status.
So we can do the switchover in multiple independent pull requests.

- [ ] plugins/ua_network_pubsub_mqtt.c
- [ ] plugins/mqtt/ua_mqtt_adapter.c
- [ ] arch/network_ws.c
- [ ] plugins/ua_pki_default.c
- [ ] plugins/ua_pubsub_ethernet.c
- [ ] plugins/securityPolicies/openssl/ua_pki_openssl.c
- [ ] plugins/ua_pubsub_ethernet_etf.c
- [ ] plugins/ua_config_default.c
- [ ] plugins/ua_pubsub_udp.c
- [ ] plugins/ua_pubsub_ethernet_xdp.c
- [ ] arch/network_tcp.c

Just leave a short comment if you are working on any of these plugins.
To avoid double-work.

saukijan, 2020-04-09:
How about we open multiple Issues for each pull request? 

We could label them Work In Progress if it is being worked and assign the developers, and once it's done remove the Work In Progress Label. Furthermore if no one is assigned to that Issue then some one else can take ownership and safely work on it.

jpfr, 2020-04-09:
Let's not overcomplicate.
A PR is also an issue on Github anyway.

saukijan, 2020-04-21:
That's fine with me @jpfr, did you start working on any of these? 

If not I will open a PR for some of the plugins, not sure how long it will take since I just started digging deeper into the code.

jpfr, 2020-04-22:
I have not worked on this.
Contributions are welcome!