open62541/open62541 #3545: What's up with `BADWRITENOTSUPPORTED` for `ATTRIBUTEID_BROWSENAME` :worried:

juangburgos, 2020-04-10:
##  Description

I noticed the most recent master simply returns `UA_STATUSCODE_BADWRITENOTSUPPORTED` when using the **public API** method `UA_Server_writeBrowseName`. What's up with that? 

Does this means now we can only define the browse name upon node instantation? 

If so, that is really going to mess up with my API at [`QUaServer`](https://github.com/juangburgos/QUaServer) :skull:. Isn't there any way to keep supporting `UA_Server_writeBrowseName` by just explaining that it should not be used often?

## Background Information / Reproduction Steps

Just try to use `UA_Server_writeBrowseName`.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): latest master
 - [x] Operating system: any
 - [x] Critical issue

jpfr, 2020-04-10:
There were performance problems with BrowseName lookup.
The solution is to add a sorted tree based on the name hash to all referencing nodes.

I’m surprised that changing the browse name is a common pattern.
We can add that back. But it we then have to touch all references to the node.

juangburgos, 2020-04-10:
I understand. Not a common pattern indeed, but the `QUaServer` API was designed to require no arguments when instantiating a node in order to make it as easy as possible:

```c++
QUaServer server;
// get objects folder
QUaFolderObject * objsFolder = server.objectsFolder();
// instantiate an object
QUaBaseObject * objBase  = objsFolder->addBaseObject();
objBase->setBrowseName("my_object");
```

That `setBrowseName` mosy likely be called only once indeed. The problem I face is that I will have to change my API everywhere, including the [serialization API](https://github.com/juangburgos/QUaServer#serialization) I added recently.

I agree the *BrowseName* should only be defined once, and must be **unique** wrt to the parent node. And browse services been used a lot, I appreciate the effort to make it more performant. 

If I have to change everything then I have to change everything, and also my users will have to change their code. We have to move forward, the world keeps turning. It would just be nice to have some transition when breaking the public API in such a way. Like keeping it functional (though expensive) and mark it as *deprecated* and print some warning to console or something for while, before removing it completely.

I guess I can live with the change and spend some time changing stuff now that we all have a lot of extra *corona-time*, but just a thought for the future.

Thanks for your effort guys.

closed 2020-04-10 by juangburgos