open62541/open62541 #3546: conan packages for open

matkonnerth, 2020-04-11:
not an issue, but maybe interesting for you:
I've setup a build for building binary conan packages (gcc 6,7,8 and clang 7,8,9, shared, for Linux) of the open. Here's the repo of the [recipe](https://github.com/matkonnerth/conan-open62541), here a repo with an [example](https://github.com/matkonnerth/conan-open62541-example) on how to consume the packages.

What do you think of the idea to bring the open as binary package to the official conan repo?: https://github.com/conan-io/conan-center-index

saukijan, 2020-04-14:
Hey @markusgraube I have been working on a conan recipe as well and was about to open an issue about it, but wanted to wait a bit before I worked all of the kinks out.

My recipe is a bit more configurable then yours, but I ran into a couple of problems when running the Unit tests from CMake Test runner as well as some of more advance features like Historiziation, Multithread, etc...

I also categorized the CMake options that exist within the root [CMakeLists.txt](https://github.com/open62541/open62541/blob/v.1.0.1/CMakeLists.txt). 

| CMake Option                              |    Value Type     |  Implemented  | Default Value  |  Notes                                                                           |
| ----------------------------------------- | ----------------- | ------------- | -------------- |--------------------------------------------------------------------------------- |
| CMAKE_BUILD_TYPE                          |      STRING       |      Yes      | RelWithDebInfo |  "RelWithDebInfo", "Release", "Debug", "MinSizeRel"                              |
| UA_ENABLE_AMALGAMATION                    |      ON/OFF       |      Yes      |     OFF        |                                                                                  |
| UA_ARCHITECTURE                           |      STRING       |               |                |   posix/win32                                                                    |
| UA_ENABLE_WEBSOCKET_SERVER                |      ON/OF        |               |                |   adds websocket networking                                                      |
| BUILD_SHARED_LIBS                         |      ON/OFF       |               |                |                                                                                  |
| UA_LOGLEVEL                               |      STRING       |      Yes      |     Trace      |   "Fatal", "Error", "Warrning", "Info", "Debug", "Trace"                         |
| UA_MULTITHREADING                         |      STRING       |               |                |   0-99: No Multithreading, 100-199: Thread-Safe API, >=200: Internal Threads     |
| UA_ENABLE_HISTORIZING                     |      ON/OFF       |               |                |                                                                                  |
| UA_ENABLE_EXPERIMENTAL_HISTORIZING        |      ON/OFF       |               |                |                                                                                  |
| UA_ENABLE_METHODCALLS                     |      ON/OFF       |      Yes      |     ON         |                                                                                  |
| UA_ENABLE_NODEMANAGEMENT                  |      ON/OFF       |      Yes      |     ON         |                                                                                  |
| UA_ENABLE_SUBSCRIPTIONS                   |      ON/OFF       |      Yes      |     ON         |                                                                                  |
| UA_ENABLE_SUBSCRIPTIONS_EVENTS            |      ON/OFF       |      Yes      |     OFF        |                                                                                  |
| UA_ENABLE_DISCOVERY                       |      ON/OFF       |      Yes      |     ON         |                                                                                  |
| UA_ENABLE_DISCOVERY_MULTICAST             |      ON/OFF       |      Yes      |     OFF        |                                                                                  |
| UA_ENABLE_QUERY                           |      ON/OFF       |               |                |                                                                                  |
| UA_ENABLE_COVERAGE                        |      ON/OFF       |               |                |   gcov                                                                           |    
| UA_ENABLE_ENCRYPTION                      |      ON/OFF       |               |                |                                                                                  |
| UA_ENABLE_MICRO_EMB_DEV_PROFILE           |      ON/OFF       |               |                |                                                                                  |
| UA_NAMESPACE_ZERO                         |      STRING       |      Yes      |    Reduced     |   "Minimal", "Reduced", "Full"                                                   |
| UA_BUILD_FUZZING                          |      ON/OFF       |               |                |   advanced                                                                       |
| UA_BUILD_OSS_FUZZ                         |      ON/OFF       |               |                |   advanced                                                                       |
| UA_BUILD_FUZZING_CORPUS                   |      ON/OFF       |               |                |   advanced                                                                       |
| UA_ENABLE_JSON_ENCODING                   |      ON/OFF       |               |                |                                                                                  |
| UA_ENABLE_MULTITHREADING                  |      ON/OFF       |      Yes      |     OFF        |   CMake Advanced Option                                                          |
[... cut ...] em

My idea with this recipe is that conan central index runs both the unit and integration test against a know release package, as declared in [Conan contribution guide](https://github.com/conan-io/conan-center-index/wiki)

This recipe is still a work in progress but It works for all of the standard options declared in the [documentation](https://open62541.org/doc/1.0/) for debian 9 GCC 8, when Unit Tests are excluded. 

I will try to publish an open repository today, if I can, but no promises.

matkonnerth, 2020-04-14:
Wow, thanks for the detailled feedback. I think, we should aim for a "sane, standard configuration" with few options, otherwise the number of buillds will explode (e.g. 10 options, independent of each other -> 1024 builds only for debug).

Regarding the unit tests, here a note from https://github.com/conan-io/conan-center-index/wiki.
"⚠️ Note: This CI service is not a testing service, it is a binary building service for package releases. Unit tests shouldn't be built nor ran in recipes by default. Before submitting a PR it is mandatory to run at least a local package creation."

Another interesting point will be, when open need dependencies like mbedtls or openssl, but these packages are already available on conan-center.

Let me know when you published your repo, we can work together on this.

saukijan, 2020-04-14:
### Testing
Regarding testing, that makes things much easier, however an integration test is till needed so Ill keep that intact. 

### Configurations
Regarding configurations, conan central index would only contain the standard config as specified in [documentation](https://open62541.org/doc/1.0/building.html), however the user should always be able to build the package locally with all of the options he needs.  What we could discuss is the options the user would like to set. 

I went over the recipe again and in total it seems to boil down to 28 options, 20 of which are supported right now.  

Furthermore, I narrowed down the failing cmake options,  `UA_ENABLE_MICRO_EMB_DEV_PROFILE` and `UA_ENABLE_ENCRYPTION` fail to configure (cmake command fails) and `UA_COMPILE_AS_CXX` fails to compile. 

### Third party dependencies
Regarding dependency management, doesn't open62541 repository contain all the dependencies it needs? Or do you mean to rework the current dependency structure to use conan repositories? 

That could be done, thought it requires that each developer installs conan, plus some light modifications to CMakeLists.txt.

saukijan, 2020-04-14:
I publish the repository [here](https://github.com/Hahn-Schickard/open62541-conan-recipe)

saukijan, 2020-04-16:
## Fixed problems

So I fixed the configuration problems that I found:

* `UA_ENABLE_ENCRYPTION`  fails to configure - Encryption options requires mbdetls to be installed on the system, so I added a requirement for mbedtls v2.16.3. This version is licensed under GNU General Public License and Apache, I don't know why there are separate builds under different licenses or what the difference is, so I chose GPL but if that causes problems we can switch to Apache. 

*  `UA_ENABLE_MICRO_EMB_DEV_PROFILE`  fails to configure - that was a bit of an annoyance for me, because I was expecting the [source code artifacts](https://github.com/open62541/open62541/releases) to contain all of the sub-modules it needed already initialized, however, that was not the case, so I added a step to download the appropriates sub-modules and move them to their proper place. I am wondering if this is a bug with CI setup or is that the way it was intended? If it's a bug and it is fixed by the next release, my recipe will work, if it's a feature, then I will need to use @markusgraube approach with explicit git cloning and sub-module initialization. 

## Problems that need fixing
The failed compilation with `UA_COMPILE_AS_CXX ` seems to be a bug, I will open an issue for it.

## Clang options
@markusgraube could you test the recipe under your setup? I am wondering if the clang options that you specified are really necessary (I myself built the package with GCC 8.3, on Debian Buster, and I am not that familiar with clang as a compiler, mostly used it for linting and formatting)

You can build it locally with: 
```bash
conan create . [version]@
```
This will create `[HOME_DIR]/.conan/data/open61541` directory and you can then use the package in your ´connanfile.txt´ by adding `open61541@v1.0.0` or `open61541@v1.0.1`, dependending on which version you built

## Unsupported options
I omitted some of the experimental features like expermental_historization and json, since I havent tried using them myself. Does anyone know how stable they are? 

Also, can anyone explain me what CMake options: `UA_BUILD_FUZZING`, `UA_BUILD_OSS_FUZZ` and `UA_BUILD_FUZZING_CORPUS` do?

matkonnerth, 2020-04-16:
wow, it's working in RelWithDebug and Release on Debian 10 with clang.

I have the following issues:
Then I wanted to build it with debug with the following command
conan create . 1.0.1@ -s build_type=Debug -pr=clang8 (pr is the clang profile on my machine).
But the build type was overriden by
#cmake.definitions["CMAKE_BUILD_TYPE"] = self.options.build_type

Maybe you can leave that out of the options, you can specify it with the conan package settings:
settings = "os", "compiler", "build_type", "arch"

So I commented this out -> ok -> Build Type debug can now be selected, open is built in debug mode -> ok, building the test is also ok, but then when starting the test the following error occurs:
./example: symbol lookup error: /home/matzy/.conan/data/open62541/1.0.1/_/_/package/7188b78e775f26dcac879db86f15791cb47e1a38/lib/libopen62541.so.0: undefined symbol: __asan_option_detect_stack_use_after_return

This is because open links against asan in debug build with clang:
https://github.com/open62541/open62541/blob/34c183155d74ee736040bbe4135262adc7d489ae/CMakeLists.txt#L581

matkonnerth, 2020-04-16:
the asan link issue above is also the reason why I've added the clang options in debug build in my recipe. I don't know, if there's a better solution.

saukijan, 2020-04-16:
Hi @matkonnerth, thanks for the info and I am glad Release versions worked for you, that was my main concern! 

### Regarding build types
You are correct on this one, I seem to have forgotten that conan has its inbuilt build_type support. I will remove it as an option!

### Regarding testing
The options you provided in your [recipe](https://github.com/matkonnerth/conan-open62541/blob/master/conanfile.py) are all related to memory leak tests, and in my recipe I completely ignore anything to do with unit and memory tests, so no unit tests or valgrind/Dr. Memory/ASan targets should be built. I might even explicitly state that it should be off, but at the moment that is done by CMakeLists.txt itself. 

However it is still interesting that cmake fails to detect Debug configuration and add these options itself, it might be the way that the type is checked...

 In principal the following should work: 
https://github.com/open62541/open62541/blob/34c183155d74ee736040bbe4135262adc7d489ae/CMakeLists.txt#L8

However I fail to see the advantage in using it. I usually check it with [CMAKE_BUILD_TYPE](https://cmake.org/cmake/help/latest/variable/CMAKE_BUILD_TYPE.html) and it is being used later in the same CmakeLists.txt: 
https://github.com/open62541/open62541/blob/34c183155d74ee736040bbe4135262adc7d489ae/CMakeLists.txt#L596

Regardless of what is happening, this opens up an important question - **should the finished package build unit tests and allow the user to run them?**

matkonnerth, 2020-04-16:
Hi, 
https://github.com/open62541/open62541/blob/34c183155d74ee736040bbe4135262adc7d489ae/CMakeLists.txt#L581

the flags to use asan are added when build type == Debug and Unix and **NOT** UA_ENABLE_UNIT_TESTS_MEMCHECK

So when I'm using the relWithDebug version of your recipe and pulling the package in in my application, everything works fine. When I pull in the debug version (because I want something debug in my application), conan pulls in the debug version and I get the link error. When I link against asan, everythin is fine.

So I think, when I use a release version of a library via conan, I shouldn't need the asan stuff. Solution could be to let it as it is and tell conan not to pull in the debug version of open62541. But I haven't tried that out. We can setup a skype/teams call if you want to discuss this.

saukijan, 2020-04-17:
Hey @matkonnerth, sure we can set up a team chat next week and discuss this as well as the other problems I mentioned before. I'll send you an email when I have things cleared up on my side.

Still it is interesting to me, why there is a **NOT** specifier there, when all of the other memory tests were disabled. As I understand Address Sanitizer is mainly a memory analysis tool that instruments the code, hence slowing it down at least twice as much as if there were no instrumentation.

saukijan, 2020-05-19:
I have updated conan recipe with Clang AddSan Flags and register for Conan Center Index Early Access Program, once accepted I'll upload the recipe and link it here for review

saukijan, 2020-05-20:
Hello everyone, I moved the recipe [here](https://github.com/Hahn-Schickard/conan-center-index/tree/master/recipes/open62541) and opened a PR to [conan-center-index](https://github.com/conan-io/conan-center-index/pull/1687). 

Once the review process passes the pacakge should be reachable simply by adding `open62541/v1.0.1` in your *conanfile.txt*

@matkonnerth, @jpfr could you add the following labels for this issue so it is a bit more visible?

- [ ] Status: Help wanted
- [ ] Status: Review Needed
- [ ] Type: Enhancement

matkonnerth, 2020-05-21:
@saukijan : I will test the recipe locally the upcoming days and give you feedback. Thank you for your efforts!

matkonnerth, 2020-05-31:
@saukijan : I've tested the 1.0.1 with gcc and clang on debian, worked for me. Thank you, I will close this issue.

closed 2020-05-31 by matkonnerth