open62541/open62541 #3551: create the address space at run time by reading the xml file

Rondine89, 2020-04-14:
Hi!
I'd like to create the address space at run time. Since the nodes in my server depends on the customer requests, my idea is to replicate the nodeset_compiler.py functions in the main of my code, in order to read the xml file and cerate the nodes of the address space. 
Is it possible?

ccvca, 2020-04-14:
There hase been some work already done here https://github.com/open62541/open62541/pull/2937

matkonnerth, 2020-04-14:
> There hase been some work already done here #2937

#2937 implements a nodestore. I took this approach, because adding nodes at runtime over add nodes service was really slow back then. But now I would use the standard interfaces, as demonstrated here: https://matkonnerth.github.io/ModernOPC-Docu/tutorials/nodeset.html

So if you are interested, we could rework the #2937.

Rondine89, 2020-04-15:
Thank you very much!
I'll take a look at the #2937 and I'll let you know.
Thank you again

Rondine89, 2020-04-15:
I've looked at the #2937 and at the https://github.com/matkonnerth/open62541/tree/xmlimportWithTypes/plugins/xmlimport files and I think I could use these files, but there somenthing I am missing: where is the function to import an xml file from a specific path?

I've also looked at the https://matkonnerth.github.io/ModernOPC-Docu/tutorials/nodeset.html and I really really like it, because it seems simplest then the other solution, but I need to have it in a C languagge version, is it possible?

matkonnerth, 2020-04-15:
Here is the xmlimport example:
https://github.com/matkonnerth/open62541/blob/023fa8c04812fc0393932e212a95f7761215b0e4/examples/server_import.c#L31

Here are the sources of the C++ tutorial:
https://github.com/matkonnerth/ModernOPC/tree/master/tutorials/nodesets

Rondine89, 2020-04-15:
Ok, thank you.
In https://github.com/matkonnerth/open62541/blob/023fa8c04812fc0393932e212a95f7761215b0e4/examples/server_import.c#L31 is it possible to replace the argv[cnt] in the f.file = argv[cnt] with the exact path of the file?

matkonnerth, 2020-04-15:
sure!

Rondine89, 2020-04-16:
Hi again!
I get several error in the https://github.com/matkonnerth/open62541/tree/xmlimportWithTypes/plugins/xmlimport nodeset.c when I compile and are all related to the entries targetIdsSize and targetIds of the typedef struct UA_NodeReferenceKind, because they doesn't exist in the struct. I can add them, I just need to know the type of these data,  but I don't know if it could lead to other errors.

matkonnerth, 2020-07-18:
Hi @Rondine89 , is the xmlimport working for your use case?

NoelGraf, 2021-05-06:
@Rondine89 Should the issue remain open? Are you still working on that?

teccheck, 2025-05-20:
We have nodesetloader now, it can loads nodesets at runtime