open62541/open62541 #3552: free(): invalid pointer with UA_Variant_setArrayCopy()

ohnowhathaveidone, 2020-04-14:
## Description
I am tying open62541 into a plugin for CoppeliaSim (https://github.com/CoppeliaRobotics). When writing Arrays to the server, I get a free(): invalid pointer or double free error (plus crash). This happens when I free the variant with ```UA_Variant_delete()``` or ```UA_Variant_clear()``` and subsequently the plugin frees the memory used to pass data on to open62541.  
I do not run into the issue with ```UA_Variant_setScalarCopy()``` for single values. I may be missing something here, but I expecteded them to work in the same way. 

Not calling ```UA_Variant_delete()``` or ```..._clear()``` leads to a memory leak of 64 bytes for the four doubles written. Valgrind output:  
```
==5520== 64 bytes in 2 blocks are indirectly lost in loss record 257 of 461
==5520==    at 0x4C31B25: calloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==5520==    by 0x37E7BD14: UA_Array_copy (in /home/rctrl/Downloads/soft/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/libsimExtOpen62541Wrapper.so)
==5520==    by 0x37E7A49D: UA_Variant_setArrayCopy (in /home/rctrl/Downloads/soft/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/libsimExtOpen62541Wrapper.so)
==5520==    by 0x37DDF297: int genericWriteNumericArray<double, float>(UA_Client*, UA_NodeId, std::vector<float, std::allocator<float> >, std::vector<int, std::allocator<int> >, UA_DataType const*) (plugin.cpp:249)
==5520==    by 0x37DD96B3: writeDoubleArrayByStringID(SScriptCallBack*, char const*, writeDoubleArrayByStringID_in*, writeDoubleArrayByStringID_out*) (plugin.cpp:670)
==5520==    by 0x37E5A6D8: writeDoubleArrayByStringID_callback(SScriptCallBack*) (stubs.cpp:9663)
==5520==    by 0x9BC98AF: _genericFunctionHandler_new(void*, CLuaCustomFunction*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) (in /home/rctrl/Downloads/soft/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/libcoppeliaSim.so)
==5520==    by 0x9BCAA3C: _simGenericFunctionHandler(void*) (in /home/rctrl/Downloads/soft/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/libcoppeliaSim.so)
==5520==    by 0x40351D3: luaD_precall (in /home/rctrl/Downloads/soft/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/liblua5.1.so)
==5520==    by 0x403E90E: luaV_execute (in /home/rctrl/Downloads/soft/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/liblua5.1.so)
==5520==    by 0x403561D: luaD_call (in /home/rctrl/Downloads/soft/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/liblua5.1.so)
==5520==    by 0x4034D76: luaD_rawrunprotected (in /home/rctrl/Downloads/soft/CoppeliaSim_Edu_V4_0_0_Ubuntu18_04/liblua5.1.so)
```

## Background Information / Reproduction Steps
This is the most minimal example I can make to reproduce the error without CoppeliaSim's LibPlugin. The example runs against tutorial_server_variable from the examples folder and writes "double.matrix".  

```c++
#include <iostream>
#include <vector> 

extern "C" {
    #include <open62541/client_config_default.h>
    #include <open62541/client_highlevel.h>
    #include <open62541/client_subscriptions.h>
    #include <open62541/plugin/log_stdout.h>
}

template <typename uaType, typename cppType>
int genericWriteNumericArray (UA_Client *client, UA_NodeId nodeId, 
        std::vector<cppType> inData, std::vector<int> dimensions, 
        const UA_DataType *outDataType){

    //cast inData into appropriate type
    //inData can be int or float
    std::vector<uaType> tmpVec(inData.begin(), 
                               inData.end()); 
    
    std::vector<UA_UInt32> arrayDims(dimensions.begin(), 
                                     dimensions.end());
    
    //make variant
    UA_Variant *value = UA_Variant_new();
    UA_Variant_init(value);
    UA_Variant_setArrayCopy(value, 
                        static_cast<const void *>(tmpVec.data()), 
                        static_cast<size_t>(tmpVec.size()), 
                        outDataType);
    
    value->arrayDimensionsSize = static_cast<size_t>(arrayDims.size());
    value->arrayDimensions = reinterpret_cast<UA_UInt32 *>(arrayDims.data());
    
    //write
    UA_StatusCode retval = UA_Client_writeValueAttribute(client, nodeId, value);
    
    //delete/clear variant - result is the same
    //UA_Variant_clear(value);
    UA_Variant_delete(value);
    return 0;
}

int main (int argc, char *argv[]) {
    std::vector<float> inData;
    std::vector<int> dimensions {2, 2};

    //populate inData 
    for (int i=0; i<argc; i++){
        std::cout<<argv[i]<<std::endl;
        inData.push_back(atof(argv[i]));
    }

    UA_Client *client = UA_Client_new();
    UA_ClientConfig_setDefault(UA_Client_getConfig(client));
    UA_StatusCode success = UA_Client_connect(client, "opc.tcp://localhost:4840");
    
    std::string nodeName = "double.matrix";
    UA_NodeId nodeId = UA_NODEID_STRING(1, (char*) nodeName.c_str());

    int retval = genericWriteNumericArray<UA_Double, float>(client, 
                                                            nodeId, 
                                                            inData, 
                                                            dimensions, 
                                                            &UA_TYPES[UA_TYPES_DOUBLE]);
    std::cout<<retval<<std::endl;

    //disconnect
    UA_Client_disconnect(client);
    UA_Client_delete(client);

    delete(&inData); //-> causes double free error
    return 0;
}
```

Compilation: 
```gcc -c open62541.c -fPIC -o open62541.o ```
```g++ path/to/open62541.o variantDeleteTest.cpp -g -o variantDeleteTest```
```./variantDeleteTest 1 2 3 4```

Used CMake options:

No changes to CMake options were made.  

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.1.0
 - [x] Other OPC UA SDKs used (client or server): No
 - [x] Operating system: Ubuntu 18.04 kernel: 4.15.0-96-generic gcc 7.5.0
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue: No

ccvca, 2020-04-14:
> ```c++
> int main (int argc, char *argv[]) {
>     std::vector<float> inData;
>     [...]
>     delete(&inData); //-> causes double free error
>     return 0;
> }
> ```

As the vector is created on the stack, there is no need for delete. The destructor will be called at the end of the main Function.

ohnowhathaveidone, 2020-04-14:
Yes - in my example, free/delete does not make sense. However, the plugin that I'm wrapping around the code allocates inData on the heap and frees it after a function call. And then my code crashes. 

My question is more related to if it is normal behavior that UA_Variant_delete/clear also frees the memory allocated to the original input, despite having created the variant wit setArray**Copy**. Again, setScalar**Copy** does not cause this problem. With setScalar and setArray (not copying), the double free happens, which makes sense. 

Best, 

Nikita

jpfr, 2020-04-14:
I suspect this is the problem:
```c
 std::vector<UA_UInt32> arrayDims(dimensions.begin(), 
                                     dimensions.end());
```

You set the arraydimensions pointers in the variant.
They are then cleaned up (freed) in the end.
But the C++ vector also wants to free the array.

You could either make a dedicated copy of the arraydimensions.
Or you don't call UA_Variant_delete at the end.

Here's a solution approach that should work. But I didn't try to compile. Take it just as a hint.

```c++
template <typename uaType, typename cppType>
int genericWriteNumericArray (UA_Client *client, UA_NodeId nodeId, 
        std::vector<cppType> inData, std::vector<int> dimensions, 
        const UA_DataType *outDataType){
   
    UA_Variant value;
    UA_Variant_init(&value);

    // !! Don't make a copy here !!
    UA_Variant_setArray(&value, 
                        static_cast<const void *>inData->data(),
                        static_cast<size_t>inData->size(), 
                        outDataType);
    
    value.arrayDimensionsSize = static_cast<size_t>(dimensions->size());
    value.arrayDimensions = reinterpret_cast<UA_UInt32 *>(dimensions->data());
    
    //write
    UA_StatusCode retval = UA_Client_writeValueAttribute(client, nodeId, &value);
    
    // !! don't clean up here !!
    //UA_Variant_delete(value);
    return retval;
}
```

ohnowhathaveidone, 2020-04-15:
Thanks for the reply!
Using ```setArray``` instead of ```setArrayCopy``` and not cleaning up seems to mostly resolve the issue. There still seems to be a minor memory leak, but I can live with this for now. It doesn't scale with the amount of data I'm sending. 

I'll talk to the devs over at CoppeliaSim too. 

Cheers!

closed 2020-04-15 by ohnowhathaveidone