open62541/open62541 #3556: Failed to Connect with QOPCUa Plug In

sanicr0, 2020-04-15:
## Description

I'm using the open62541 Plug-in for QOPCUa to create a Client and connect it to a Server.
With OPC Expert I can connect to this Serve without any issue but when I debug my QT C++ Code I get following Error:
qt.opcua.plugins.open62541: Open62541: Failed to connect

Also I print out the ClientError and Connection Error Enums from QT which are both Unknown Error.

## Background Information / Reproduction Steps

Basically what I'm doing in my code is:

- create a client (which works!)

- connect all signals and slots (also works)

- request endpoints from the localhost + port (works, I'm also getting the right endpoint back)

- connect to endpoint (auth-type is anonymous)

and with connect to endpoint it goes into connecting state but then it throws the unknown errors and disconnect. so it never goes into connected state. and then it throws the 
qt.opcua.plugins.open62541: Open62541: Failed to connect
error.

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 1.0
 - [ ] Other OPC UA SDKs used (client or server): OPC Expert
 - [ ] Operating system: Windows 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-04-15:
Did you integrate encryption?
Sometimes encryption is necessary as the password is encrypted even on an unencrypted SecureChannel.

Can you bring this up on the qtOPCUA github?

sanicr0, 2020-04-15:
i don't need a password, the connection type from the server is: http://opcfoundation.org/UA/SecurityPolicy#None

jpfr, 2020-04-15:
No. Even with an unencrypted SecureChannel, the server might set a "UserTokenPolicy" where just the password is encrypted.
The selected UserTokenPolicy should appear in the logs.

jpfr, 2020-07-07:
Reopen if required.

closed 2020-07-07 by jpfr