open62541/open62541 #3558: Failled to compile with -DUA_COMPILE_AS_CXX=ON

saukijan, 2020-04-16:
## Description

I am building a movable open62541 package based on Conan Dependency Management System  (more info in [Issue #3546](https://github.com/open62541/open62541/issues/3546#issuecomment-614644937)) and when trying out `UA_COMPILE_AS_CXX` Cmake Option I ran into a compilation failure regarding go to's and jumping `error: jump to label ‘error’ [-fpermissive]` it seems that it doesn't want to jump to defined [error label](https://github.com/open62541/open62541/blob/v.1.0.1/src/ua_securechannel.c#L580) due to  crosses initialization. 

## Background Information / Reproduction Steps

Compile with:
```bash
cmake -G "Unix Makefiles" -DCMAKE_BUILD_TYPE="RelWithDebInfo" -DCONAN_CMAKE_CXX_STANDARD="17" -DCONAN_CMAKE_CXX_EXTENSIONS="OFF" -DCONAN_STD_CXX_FLAG="-std=c++17" -DCONAN_IN_LOCAL_CACHE="ON" -DCONAN_COMPILER="gcc" -DCONAN_COMPILER_VERSION="8" -DCONAN_CXX_FLAGS="-m64" -DCONAN_SHARED_LINKER_FLAGS="-m64" -DCONAN_C_FLAGS="-m64" -DCONAN_LIBCXX="libstdc++11" -DBUILD_SHARED_LIBS="True" -DCMAKE_VERBOSE_MAKEFILE="ON" -DUA_ENABLE_SUBSCRIPTIONS="True" -DUA_ENABLE_SUBSCRIPTIONS_EVENTS="False" -DUA_ENABLE_METHODCALLS="True" -DUA_ENABLE_NODEMANAGEMENT="True" -DUA_ENABLE_AMALGAMATION="False" -DUA_ENABLE_MULTITHREADING="False" -DUA_ENABLE_IMMUTABLE_NODES="False" -DUA_ENABLE_WEBSOCKET_SERVER="False" -DUA_ENABLE_HISTORIZING="False" -DUA_ENABLE_DISCOVERY="True" -DUA_ENABLE_DISCOVERY_MULTICAST="False" -DUA_ENABLE_DISCOVERY_SEMAPHORE="True" -DUA_ENABLE_QUERY="False" -DUA_ENABLE_PUBSUB="False" -DUA_ENABLE_DA="False" -DUA_ENABLE_NODESET_COMPILER_DESCRIPTIONS="True" -DUA_NAMESPACE_ZERO="Reduced" -DUA_ENABLE_MICRO_EMB_DEV_PROFILE="False" -DUA_ENABLE_TYPENAMES="True" -DUA_ENABLE_STATUSCODE_DESCRIPTIONS="True" -DUA_ENABLE_HARDENING="False" -DUA_COMPILE_AS_CXX="True" ..
```

Clip of the error message :

```bash
Building CXX object source_subfolder/CMakeFiles/open62541-object.dir/src/server/ua_session.c.o
cd /home/dg1024/.conan/data/open62541/1.0.1/dg1024/testing/build/eb4faa9ef8da1a04da1af59454840e4b8c52328f/source_subfolder && /usr/bin/c++  -DUA_ARCHITECTURE_POSIX -DUA_DYNAMIC_LINKING_EXPORT -D_GLIBCXX_USE_CXX11_ABI=1 -D__STDC_CONSTANT_MACROS -I/home/dg1024/.conan/data/open62541/1.0.1/dg1024/testing/build/eb4faa9ef8da1a04da1af59454840e4b8c52328f/source_subfolder/src -I/home/dg1024/.conan/data/open62541/1.0.1/dg1024/testing/build/eb4faa9ef8da1a04da1af59454840e4b8c52328f/source_subfolder/arch -I/home/dg1024/.conan/data/open62541/1.0.1/dg1024/testing/build/eb4faa9ef8da1a04da1af59454840e4b8c52328f/source_subfolder/include -I/home/dg1024/.conan/data/open62541/1.0.1/dg1024/testing/build/eb4faa9ef8da1a04da1af59454840e4b8c52328f/source_subfolder/plugins/include -I/home/dg1024/.conan/data/open62541/1.0.1/dg1024/testing/build/eb4faa9ef8da1a04da1af59454840e4b8c52328f/source_subfolder/deps -I/home/dg1024/.conan/data/open62541/1.0.1/dg1024/testing/build/eb4faa9ef8da1a04da1af59454840e4b8c52328f/source_subfolder/src/pubsub -I/home/dg1024/.conan/data/open62541/1.0.1/dg1024/testing/build/eb4faa9ef8da1a04da1af59454840e4b8c52328f/source_subfolder/src_generated -I/home/dg1024/.conan/data/open62541/1.0.1/dg1024/testing/build/eb4faa9ef8da1a04da1af59454840e4b8c52328f/source_subfolder -I/home/dg1024/.conan/data/open62541/1.0.1/dg1024/testing/build/eb4faa9ef8da1a04da1af59454840e4b8c52328f/source_subfolder/src/client  -m64 -O2 -g -DNDEBUG   -Wshadow -Wconversion -fvisibility=hidden -fPIC -std=c++17 -o CMakeFiles/open62541-object.dir/src/server/ua_session.c.o -c /home/dg1024/.conan/data/open62541/1.0.1/dg1024/testing/build/eb4faa9ef8da1a04da1af59454840e4b8c52328f/source_subfolder/src/server/ua_session.c
/home/dg1024/.conan/data/open62541/1.0.1/dg1024/testing/build/eb4faa9ef8da1a04da1af59454840e4b8c52328f/source_subfolder/src/ua_securechannel.c: In function ‘UA_StatusCode sendSymmetricChunk(UA_MessageContext*)’:
/home/dg1024/.conan/data/open62541/1.0.1/dg1024/testing/build/eb4faa9ef8da1a04da1af59454840e4b8c52328f/source_subfolder/src/ua_securechannel.c:821:1: error: jump to label ‘error’ [-fpermissive]
 error:
 ^~~~~
/home/dg1024/.conan/data/open62541/1.0.1/dg1024/testing/build/eb4faa9ef8da1a04da1af59454840e4b8c52328f/source_subfolder/src/ua_securechannel.c:782:14: note:   from here
         goto error;
              ^~~~~
/home/dg1024/.conan/data/open62541/1.0.1/dg1024/testing/build/eb4faa9ef8da1a04da1af59454840e4b8c52328f/source_subfolder/src/ua_securechannel.c:792:12: note:   crosses initialization of ‘size_t total_length’
     size_t total_length = pre_sig_length;
            ^~~~~~~~~~~~
/home/dg1024/.conan/data/open62541/1.0.1/dg1024/testing/build/eb4faa9ef8da1a04da1af59454840e4b8c52328f/source_subfolder/src/ua_securechannel.c:790:12: note:   crosses initialization of ‘size_t pre_sig_length’
     size_t pre_sig_length = (uintptr_t)(messageContext->buf_pos) -
            ^~~~~~~~~~~~~~
make[2]: *** [source_subfolder/CMakeFiles/open62541-object.dir/build.make:246: source_subfolder/CMakeFiles/open62541-object.dir/src/ua_securechannel.c.o] Error 1
```

Full CMake output with the error [CppCompatibleFailure.txt](https://github.com/open62541/open62541/files/4487369/CppCompatibleFailure.txt)

Used CMake options:

```bash
cmake -G "Unix Makefiles" -DCMAKE_BUILD_TYPE="RelWithDebInfo" -DCONAN_CMAKE_CXX_STANDARD="17" -DCONAN_CMAKE_CXX_EXTENSIONS="OFF" -DCONAN_STD_CXX_FLAG="-std=c++17" -DCONAN_IN_LOCAL_CACHE="ON" -DCONAN_COMPILER="gcc" -DCONAN_COMPILER_VERSION="8" -DCONAN_CXX_FLAGS="-m64" -DCONAN_SHARED_LINKER_FLAGS="-m64" -DCONAN_C_FLAGS="-m64" -DCONAN_LIBCXX="libstdc++11" -DBUILD_SHARED_LIBS="True" -DCMAKE_VERBOSE_MAKEFILE="ON" -DUA_ENABLE_SUBSCRIPTIONS="True" -DUA_ENABLE_SUBSCRIPTIONS_EVENTS="False" -DUA_ENABLE_METHODCALLS="True" -DUA_ENABLE_NODEMANAGEMENT="True" -DUA_ENABLE_AMALGAMATION="False" -DUA_ENABLE_MULTITHREADING="False" -DUA_ENABLE_IMMUTABLE_NODES="False" -DUA_ENABLE_WEBSOCKET_SERVER="False" -DUA_ENABLE_HISTORIZING="False" -DUA_ENABLE_DISCOVERY="True" -DUA_ENABLE_DISCOVERY_MULTICAST="False" -DUA_ENABLE_DISCOVERY_SEMAPHORE="True" -DUA_ENABLE_QUERY="False" -DUA_ENABLE_PUBSUB="False" -DUA_ENABLE_DA="False" -DUA_ENABLE_NODESET_COMPILER_DESCRIPTIONS="True" -DUA_NAMESPACE_ZERO="Reduced" -DUA_ENABLE_MICRO_EMB_DEV_PROFILE="False" -DUA_ENABLE_TYPENAMES="True" -DUA_ENABLE_STATUSCODE_DESCRIPTIONS="True" -DUA_ENABLE_HARDENING="False" -DUA_COMPILE_AS_CXX="True"
```

## Recommended solution
After a bit of researching this issue, it seems that [code like this](https://github.com/open62541/open62541/blob/v.1.0.1/src/ua_securechannel.c#L790) needs to be surrounded by {} brackets (based on [StackOverflow](https://stackoverflow.com/questions/31513798/error-jump-to-label-foo-crosses-initialization-of-bar)), for example([changes this line](https://github.com/open62541/open62541/blob/v.1.0.1/src/ua_securechannel.c#L790)): 

```cpp
{
 size_t total_length = (uintptr_t)(messageContext->buf_pos) -
        (uintptr_t)messageContext->messageBuffer.data;
}
```

or the declaration must be separated from instantiation, like this: 
```cpp
size_t total_length; 
total_length = (uintptr_t)(messageContext->buf_pos) -
        (uintptr_t)messageContext->messageBuffer.data;
```

I personally don't like goto's since it leads to spaghetti code, thus I would recommend writing a function that takes in `UA_Connection *const` and `UA_ByteString *` as arguments and calls the `connection->releaseSendBuffer()`, something like this: 

```cpp
bool releaseConnectionSendBuffer(UA_Connection *const connection, UA_ByteString *buffer){
return connection->releaseSendBuffer(connection, buffer);
}
```

This removes all of the goto's and makes sure that ` connection->releaseSendBuffer()` is always called with similar semantics, since right now there are many goto's, that do the same thing, but with different footprints, for example, check [Line #580](https://github.com/open62541/open62541/blob/v.1.0.1/src/ua_securechannel.c#L580) and [Line #822](https://github.com/open62541/open62541/blob/v.1.0.1/src/ua_securechannel.c#L822)

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): v1.0.1
 - [ ] Other OPC UA SDKs used (client or server): server
 - [ ] Operating system: Debian Buster
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-04-17:
That's a really sound analysis of the situation

> Thus I would recommend writing a function that takes in UA_Connection *const and UA_ByteString * as arguments and calls the connection->releaseSendBuffer()

Sounds good.
Since you already have a Github account, could you send us a Pull Request with the above change?
Should be pretty straightforward.

saukijan, 2020-04-20:
Sure, Ill make a pull request when I am done testing this fix

o6-nbeier, 2026-10-20:
Thank you for taking the time to open this issue and describe it so clearly.

The problem has since been fixed on the maintained branches (see #3763 and #7828), so we are closing it as resolved.

closed 2026-10-20 by o6-nbeier