open62541/open62541 #3559: Change Node namespace

viktor-morin, 2020-04-17:
How do you change the Node's namespace?
Been using following example code:

	#include <signal.h>
	#include <open62541/server.h>
	#include <open62541/server_config_default.h>

	UA_Boolean running = true;
	void signalHandler(int sig) {
		running = false;
	}

	int main(int argc, char** argv)
	{
		signal(SIGINT, signalHandler); /* catch ctrl-c */

		/* Create a server listening on port 4840 */
		UA_Server *server = UA_Server_new();
		UA_ServerConfig_setDefault(UA_Server_getConfig(server));

		/* Add a variable node */
		/* 1) Define the node attributes */
		UA_VariableAttributes attr = UA_VariableAttributes_default;
		attr.displayName = UA_LOCALIZEDTEXT("en-US", "the answer");
		UA_Int32 myInteger = 42;
		UA_Variant_setScalar(&attr.value, &myInteger, &UA_TYPES[UA_TYPES_INT32]);

		/* 2) Define where the node shall be added with which browsename */
		UA_NodeId newNodeId = UA_NODEID_STRING(1, "the.answer");
		UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
		UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
		UA_NodeId variableType = UA_NODEID_NULL; /* take the default variable type */
		UA_QualifiedName browseName = UA_QUALIFIEDNAME(1, "the answer");

		/* 3) Add the node */
		UA_Server_addVariableNode(server, newNodeId, parentNodeId, parentReferenceNodeId,
								  browseName, variableType, attr, NULL, NULL);

		/* Run the server loop */
		UA_StatusCode status = UA_Server_run(server, &running);

		UA_Server_delete(server);
		return status;
	}

By changing: 

> UA_NODEID_STRING(1, "the.answer");
to

> UA_NODEID_STRING(2, "the.answer");
make's the Node disappear.
Does someone know how to change the namespace? -> ns=2,id="the.answer"

mlang-de, 2020-04-17:
What is return value of UA_Server_addVariableNode? Did you add a namespace by using UA_Server_addNamespace?

viktor-morin, 2020-04-17:
@mlang-de 
with UA_NODEID_STRING(1, "the.answer"), the response from UA_Server_addVariableNode is:

> UA_STATUSCODE_GOOD

and with with UA_NODEID_STRING(2, "the.answer"), the response from UA_Server_addVariableNode is:

> UA_STATUSCODE_BADNODEIDINVALID

I did not use UA_Server_addNamespace, is that needed first? Any example available?

mlang-de, 2020-04-17:
IHMO yes it is required.
Namespace 0 and 1 will be created as default.
Namespace 0 includes (a subset of) the base nodeset.
Namespace 1 is the servers namespace (see OPC UA spec.)

Now you try to add in Namespace 2 which not exists.

See function AddNode_raw() there will be the namespace checked. If you enable tracing you will see some print out like "AddNodes: Namespace invalid".

viktor-morin, 2020-04-17:
Thanks for the help!
Adding following line fixed the issue:

> UA_Server_addNamespace(server, "2")

closed 2020-04-17 by viktor-morin

rony91jp, 2020-12-11:
Hello,

I am also having the same issue. In my case, where should i change from below?

UA_StatusCode
readCurrentCapacitive_Electromagnetic_Flowmeter_Lower_Jet(UA_Server *server,
                                                          const UA_NodeId *sessionId, void *sessionContext,
                                                          const UA_NodeId *nodeId, void *nodeContext,
                                                          UA_Boolean sourceTimeStamp, const UA_NumericRange *range,
                                                          UA_DataValue *dataValue) {
                                                          
                                                          uint16_t value = adc1_get_voltage(ADC1_CHANNEL_0);
                                                                                                                
                                                          UA_Variant_setScalarCopy(&dataValue->value, &value,
                                                                                   &UA_TYPES[UA_TYPES_INT16]);
                                                          dataValue->hasValue = true;
                                                          return UA_STATUSCODE_GOOD;
}

void
addCurrentCapacitive_Electromagnetic_Flowmeter_Lower_JetDataSourceVariable(UA_Server *server) {
    UA_VariableAttributes attr = UA_VariableAttributes_default;
    attr.displayName = UA_LOCALIZEDTEXT("en-US", "Capacitive_Electromagnetic_Flowmeter_Lower_Jet");
    attr.dataType = UA_TYPES[UA_TYPES_INT16].typeId;
    attr.accessLevel = UA_ACCESSLEVELMASK_READ;

    UA_NodeId currentNodeId = UA_NODEID_STRING(1, "CEF_Lower_Jet");
    UA_QualifiedName currentName = UA_QUALIFIEDNAME(0, "Capacitive_Electromagnetic_Flowmeter_Lower_Jet");
    UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
    UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
    UA_NodeId variableTypeNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE);

    UA_DataSource timeDataSource;
    timeDataSource.read = readCurrentCapacitive_Electromagnetic_Flowmeter_Lower_Jet;
    UA_Server_addDataSourceVariableNode(server, currentNodeId, parentNodeId,parentReferenceNodeId, currentName, variableTypeNodeId, attr,timeDataSource, NULL, NULL);
}

Thank you.

viktor-morin, 2020-12-11:
> Hello,
>
> I am also having the same issue. In my case, where should i change from below?
>
> UA_StatusCode
> readCurrentCapacitive_Electromagnetic_Flowmeter_Lower_Jet(UA_Server *server,
> const UA_NodeId [...]

Dosn't seems like it's the same issue since you are using namespace 1 which is defined by default:

> UA_NodeId currentNodeId = UA_NODEID_STRING(1, "CEF_Lower_Jet");