open62541/open62541 #3560: Aggregating servers

MartinaFelberFEM, 2020-04-20:
Hello!

I would like to know if aggregating servers is supported.

Use case:
A client asks the public server A for specific information. 
Server A does not have the requested nodeIds. 
The public server A passes the requests on to hidden server B. Hidden server B sends a response to server A and server A simply forwards the response to server B.
Server A does not need to hold any information (nodes) itself, just forward the requests from client and responses from server B.

Is there any callback mechanism in the SDK to handle unexisting nodeIds? Where you could forward the request to another server B?

I know that server A can instantiate a client, browse/read all of server B's address space or create monitored items for the information it needs.
But in this case I think server A has to duplicate all nodes of server B. It would have to create all nodes it's interested in to present it to the client. And then you could use the method or variable value callbacks to ask server B.

But I don't want to create/duplicate all the nodes of server B. I would just need to forward the requests.
But I haven't found a callback or something to tell the SDK that unexisting nodeIds or all nodeIds of a specific namespace etc. can be requested from another server.

Many thanks for your support!

MartinaFelberFEM, 2020-04-21:
I would like to add a more specific use case:
Let's say you have a high level OPC UA server A which provides an information model.
For time critical data transmission you use an open62541 PubSub server B which runs in a realtime execution environment and provides more performance. The PubSub server B  is hidden. 

An external client only sees the high level server A. 
How can you synchronize server A and B?

- The external client configures server A with the PubSub configuration methods provided by the PublishSubscribe object.
- Server A has to redirect the PubSub configuration to server B.
- The address space of server B is 'empty'. It does not have the information model. But it could ask server A for information about the configured nodeIds in the DataSets and use "fast path" feature for PubSub, I think ...
- Server A needs to synchronize PubSub configuration/diagnostic nodes from server B, to present it for the external client

How can you synchronize all the PubSub nodes (configuration/diagnostics) between server A and B?
Particularly if server A uses PubSub (e.g. MQTT) too and has it's own PubSub configuration/diagnostics?
I think the answer to this question is dependent on the used OpcUa SDK too.
But let's say both servers would use open62541 SDK.

Is there a standard approach to solve this?

Or shall the PubSub server B be public too? And the client has to browse server A to gather the information model data and configure server B independently?

Thanks for your help

ccvca, 2020-04-21:
As far as I know, there is no such callback, that is called, when no node is found.
The implementation may be by possible by implementing a nodestore, which tries to read from the default nodestore and if this is not possible, read from an external Server:
https://github.com/open62541/open62541/blob/578d521931777181d1a881c8335ca5fafcd66eb2/include/open62541/plugin/nodestore.h#L458-L501

Maybe the namespace index can be used to distinguish between the servers and node stores.

OPC UA in generat has Expanded NodeIds, that can point to another server, but in this case the other server must be public available:
https://github.com/open62541/open62541/blob/e2cea8bf405a1a63b1d11024f3b95d8d63ecbca9/include/open62541/types.h#L453-L461

MartinaFelberFEM, 2020-04-21:
Many thanks for your answer! It's very much appreciated!

The nodestore suggestion sounds good. 
Particularly if I could constrain the use case, so that Server A never uses PubSub itself. Maybe then I could safely redirect all client node requests to the Server/PublishSubscribe object to the PubSub server B. Without thinking about it too much.

It could get more complicated if both servers use PubSub, though ...

I will give the nodeStore suggestion a try if the redirection works.

MartinaFelberFEM, 2020-04-23:
Implementing a custom nodestore to read from a remote server works in general. I have tested it with the getNode interface method and retrieved some variables from a remote server. Thanks for your suggestion.

But I am not sure if this is more or less cumbersome than browsing the Server B's address space, copying the nodes and create monitored items etc.
Then I would not depend on the nodestore API ...

I think I can't distinguish with the namespace index of the requests because I would like to aggregate PubSub nodes, which are in namespace 0 in both servers.

I think aggregating would be easier if I would only aggregate "normal" information model nodes and not the PublishSubscribe node of another server. Or nodes of namespace 0 in general, which are used by the server SDKs.
I think I would have to merge the configuration and diagnostics nodes if both servers use PubSub. This could get messy,

juangburgos, 2020-04-23:
From what I understand of the specification, is that the aggregating server should only audit the a client request and then proxy it to the underlying target server. I don´t know of copying the nodes and cteaing monitored items would be considered an strictly aggregating server.

<https://reference.opcfoundation.org/v104/Core/docs/Part2/4.14.3/>

I believe the specification also considers special types to expose the underlying server's namespace:

<https://reference.opcfoundation.org/v104/Core/docs/Part5/6.3.13/>

I haven't had a deep look on how the specification defines explicitly how an aggregating server should behave, but I am interested in the topic.

MartinaFelberFEM, 2020-04-23:
Thanks for your links!
I haven't considered auditing at all ...
I should take more time reading the specification.

Part5 describes that a server can show the server objects of it's aggregated servers.

> Nevertheless, aggregating Servers may want to expose the Server Objects of the OPC UA Servers they are aggregating in their AddressSpace. 
https://reference.opcfoundation.org/v104/Core/docs/Part5/A.2/
Maybe that's a solution ...

I have found this example from OpcFoundation, but I haven't had a deeper look on it yet. 
https://github.com/OPCFoundation/UA-.NETStandard/tree/master/SampleApplications/Workshop/Aggregation
But I don't think that it uses Audit-Events.

ccvca, 2020-04-23:
In my opinion, there are different ways of aggregating.
1. a server combines several servers that are not accessible to the client. Here the aggregating server acts as a proxy.
2. a server offers one or more servers aggregated to act as a cache for the values in order to serve more clients than the original server could handle. In this case the aggregating server creates a copy of the values/nodes to deliver this copy to several clients. For example, a server then has only one connection, even when 100 clients read the values from the aggregating server.

PS: Just to note: There is also redundancy possible, which might use similar concepts:
https://reference.opcfoundation.org/v104/Core/docs/Part4/6.6.2/