open62541/open62541 #3563: How to use UA_Client_connect_username correctly?What should I pay attention to before using?

I-Yoo, 2020-04-21:
Operating system:win10 ; 
Open62541 Version: master ; 
I used client.c in the folder "\open62541-1.0\examples\client.c" to do the experiment. But I noticed that its return code is 0x80020000. I hope someone can help me solve this problem.

I-Yoo, 2020-04-21:
Kepware is used as the server, the security policy is None and Basic256, the output information is as follows:
`[2020-04-21 11:01:00.715 (UTC+0800)] info/client        SecurityPolicy not specified -> use default #None
[2020-04-21 11:01:00.715 (UTC+0800)] warn/securitypolicy        Security policy None is used to create SecureChannel. Accepting all certificates
[2020-04-21 11:01:00.735 (UTC+0800)] info/client        TCP connection established
[2020-04-21 11:01:00.737 (UTC+0800)] info/client        Opened SecureChannel with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
2 endpoints found
URL of endpoint 0 is opc.tcp://127.0.0.1:49320
URL of endpoint 1 is opc.tcp://127.0.0.1:49320
[2020-04-21 11:01:00.739 (UTC+0800)] info/client        Connecting to endpoint opc.tcp://127.0.0.1:49320
[2020-04-21 11:01:00.739 (UTC+0800)] info/client        SecurityPolicy not specified -> use default #None
[2020-04-21 11:01:00.739 (UTC+0800)] warn/securitypolicy        Security policy None is used to create SecureChannel. Accepting all certificates
[2020-04-21 11:01:00.740 (UTC+0800)] info/client        TCP connection established
[2020-04-21 11:01:00.740 (UTC+0800)] info/client        Opened SecureChannel with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2020-04-21 11:01:00.740 (UTC+0800)] info/client        Endpoint and UserTokenPolicy unconfigured, perform GetEndpoints
[2020-04-21 11:01:00.741 (UTC+0800)] info/client        Found 2 endpoints
[2020-04-21 11:01:00.741 (UTC+0800)] info/client        Endpoint 0 has 2 user token policies
[2020-04-21 11:01:00.741 (UTC+0800)] info/client        Rejecting UserTokenPolicy 0 in endpoint 0: security policy 'http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15' not available
[2020-04-21 11:01:00.741 (UTC+0800)] info/client        Rejecting UserTokenPolicy 1 (anonymous) in endpoint 0: configuration doesn't match
[2020-04-21 11:01:00.741 (UTC+0800)] info/client        Rejecting endpoint 1: security policy not available
[2020-04-21 11:01:00.741 (UTC+0800)] error/client       No suitable UserTokenPolicy found for the possible endpoints
UA_Client_connect_username: 80020000:`

I-Yoo, 2020-04-21:
Kepware is used as the server, the security policy is None and Basic256, the output information is as follows:
`**[2020-04-21 11:01:00.715 (UTC+0800)] info/client        SecurityPolicy not specified -> use default #None
[2020-04-21 11:01:00.715 (UTC+0800)] warn/securitypolicy        Security policy None is used to create SecureChannel. Accepting all certificates
[2020-04-21 11:01:00.735 (UTC+0800)] info/client        TCP connection established
[2020-04-21 11:01:00.737 (UTC+0800)] info/client        Opened SecureChannel with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
2 endpoints found
URL of endpoint 0 is opc.tcp://127.0.0.1:49320
URL of endpoint 1 is opc.tcp://127.0.0.1:49320
[2020-04-21 11:01:00.739 (UTC+0800)] info/client        Connecting to endpoint opc.tcp://127.0.0.1:49320
[2020-04-21 11:01:00.739 (UTC+0800)] info/client        SecurityPolicy not specified -> use default #None
[2020-04-21 11:01:00.739 (UTC+0800)] warn/securitypolicy        Security policy None is used to create SecureChannel. Accepting all certificates
[2020-04-21 11:01:00.740 (UTC+0800)] info/client        TCP connection established
[2020-04-21 11:01:00.740 (UTC+0800)] info/client        Opened SecureChannel with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2020-04-21 11:01:00.740 (UTC+0800)] info/client        Endpoint and UserTokenPolicy unconfigured, perform GetEndpoints
[2020-04-21 11:01:00.741 (UTC+0800)] info/client        Found 2 endpoints
[2020-04-21 11:01:00.741 (UTC+0800)] info/client        Endpoint 0 has 2 user token policies
[2020-04-21 11:01:00.741 (UTC+0800)] info/client        Rejecting UserTokenPolicy 0 in endpoint 0: security policy 'http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15' not available
[2020-04-21 11:01:00.741 (UTC+0800)] info/client        Rejecting UserTokenPolicy 1 (anonymous) in endpoint 0: configuration doesn't match
[2020-04-21 11:01:00.741 (UTC+0800)] info/client        Rejecting endpoint 1: security policy not available
[2020-04-21 11:01:00.741 (UTC+0800)] error/client       No suitable UserTokenPolicy found for the possible endpoints
UA_Client_connect_username: 80020000:**`

I-Yoo, 2020-04-21:
Kepware is used as the server, the security policy is None and Basic256, the output information is as follows:
[2020-04-21 11:01:00.715 (UTC+0800)] info/client        SecurityPolicy not specified -> use default #None
[2020-04-21 11:01:00.715 (UTC+0800)] warn/securitypolicy        Security policy None is used to create SecureChannel. Accepting all certificates
[2020-04-21 11:01:00.735 (UTC+0800)] info/client        TCP connection established
[2020-04-21 11:01:00.737 (UTC+0800)] info/client        Opened SecureChannel with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
2 endpoints found
URL of endpoint 0 is opc.tcp://127.0.0.1:49320
URL of endpoint 1 is opc.tcp://127.0.0.1:49320
[2020-04-21 11:01:00.739 (UTC+0800)] info/client        Connecting to endpoint opc.tcp://127.0.0.1:49320
[2020-04-21 11:01:00.739 (UTC+0800)] info/client        SecurityPolicy not specified -> use default #None
[2020-04-21 11:01:00.739 (UTC+0800)] warn/securitypolicy        Security policy None is used to create SecureChannel. Accepting all certificates
[2020-04-21 11:01:00.740 (UTC+0800)] info/client        TCP connection established
[2020-04-21 11:01:00.740 (UTC+0800)] info/client        Opened SecureChannel with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2020-04-21 11:01:00.740 (UTC+0800)] info/client        Endpoint and UserTokenPolicy unconfigured, perform GetEndpoints
[2020-04-21 11:01:00.741 (UTC+0800)] info/client        Found 2 endpoints
[2020-04-21 11:01:00.741 (UTC+0800)] info/client        Endpoint 0 has 2 user token policies
[2020-04-21 11:01:00.741 (UTC+0800)] info/client        Rejecting UserTokenPolicy 0 in endpoint 0: security policy 'http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15' not available
[2020-04-21 11:01:00.741 (UTC+0800)] info/client        Rejecting UserTokenPolicy 1 (anonymous) in endpoint 0: configuration doesn't match
[2020-04-21 11:01:00.741 (UTC+0800)] info/client        Rejecting endpoint 1: security policy not available
[2020-04-21 11:01:00.741 (UTC+0800)] error/client       No suitable UserTokenPolicy found for the possible endpoints
UA_Client_connect_username: 80020000:

I-Yoo, 2020-04-21:
Configuration of Prosys OPC UA Simulation Server:

[image]
Output log information:

> [2020-04-21 14:41:26.862 (UTC+0800)] warn/securitypolicy        Security policy None is used to create SecureChannel. Accepting all certificates
[2020-04-21 14:41:26.887 (UTC+0800)] info/client        TCP connection established
[2020-04-21 14:41:26.891 (UTC+0800)] info/client        Opened SecureChannel with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
7 endpoints found
URL of endpoint 0 is opc.tcp://LAPTOP-NBNTGFHQ:48010
URL of endpoint 1 is opc.tcp://LAPTOP-NBNTGFHQ:48010
URL of endpoint 2 is opc.tcp://LAPTOP-NBNTGFHQ:48010
URL of endpoint 3 is opc.tcp://LAPTOP-NBNTGFHQ:48010
URL of endpoint 4 is opc.tcp://LAPTOP-NBNTGFHQ:48010
URL of endpoint 5 is opc.tcp://LAPTOP-NBNTGFHQ:48010
URL of endpoint 6 is opc.tcp://LAPTOP-NBNTGFHQ:48010
[2020-04-21 14:41:26.903 (UTC+0800)] info/client        Connecting to endpoint opc.tcp://127.0.0.1:48010
[2020-04-21 14:41:26.903 (UTC+0800)] info/client        SecurityPolicy not specified -> use default #None
[2020-04-21 14:41:26.903 (UTC+0800)] warn/securitypolicy        Security policy None is used to create SecureChannel. Accepting all certificates
[2020-04-21 14:41:26.904 (UTC+0800)] info/client        TCP connection established
[2020-04-21 14:41:26.907 (UTC+0800)] info/client        Opened SecureChannel with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2020-04-21 14:41:26.907 (UTC+0800)] info/client        Endpoint and UserTokenPolicy unconfigured, perform GetEndpoints
[2020-04-21 14:41:26.914 (UTC+0800)] info/client        Found 7 endpoints
[2020-04-21 14:41:26.915 (UTC+0800)] info/client        Endpoint 0 has 3 user token policies
[2020-04-21 14:41:26.915 (UTC+0800)] info/client        Rejecting UserTokenPolicy 0 in endpoint 0: security policy 'http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15' not available
[2020-04-21 14:41:26.916 (UTC+0800)] info/client        Rejecting UserTokenPolicy 1 in endpoint 0: security policy 'http://opcfoundation.org/UA/SecurityPolicy#Basic256' not available
[2020-04-21 14:41:26.916 (UTC+0800)] info/client        Rejecting UserTokenPolicy 2 (anonymous) in endpoint 0: configuration doesn't match
[2020-04-21 14:41:26.916 (UTC+0800)] info/client        Rejecting endpoint 1: security policy not available
[2020-04-21 14:41:26.916 (UTC+0800)] info/client        Rejecting endpoint 2: security policy not available
[2020-04-21 14:41:26.917 (UTC+0800)] info/client        Rejecting endpoint 3: security policy not available
[2020-04-21 14:41:26.917 (UTC+0800)] info/client        Rejecting endpoint 4: security policy not available
[2020-04-21 14:41:26.918 (UTC+0800)] info/client        Rejecting endpoint 5: security policy not available
[2020-04-21 14:41:26.918 (UTC+0800)] info/client        Rejecting endpoint 6: security policy not available
[2020-04-21 14:41:26.918 (UTC+0800)] error/client       No suitable UserTokenPolicy found for the possible endpoints
UA_Client_connect_username: 0x80020000

How to solve this problem?

jpfr, 2020-04-22:
Enable encryption support for open62541.

closed 2020-04-26 by I-Yoo

olivluca, 2020-12-22:
I built the 1.1.3 version (pack/1.0.zip) with encryption enabled, I  tried first with openssl then with mbedtls, same result.
The server is a siemens rfc185-c. In anonymous mode I can connect with no problems:

```
Connection 15 | SecureChannel 1080351538 | Opened SecureChannel with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
Rejecting UserTokenPolicy 0 (anonymous) in endpoint 0: configuration doesn't match
Rejecting UserTokenPolicy 1 in endpoint 0: security policy 'http://opcfoundation.org/UA/SecurityPolicy#Basic256' not available
No suitable UserTokenPolicy found for the possible endpoints

```

olivluca, 2020-12-22:
Same result with master