open62541/open62541 #3566: Server on STM-Microcontroller may not free memory correctly

svro86, 2020-04-23:
## Description
I am running a 1.0-version server on an STM32F207 microcontroller. This works quite fine with one issue: If I disconnect and reconnect a client (UAExpert) a couple of times I will get a bad memory error after some time and I will also get this error when I wait a couple of hours (between 4hours and 10hours). If I just let the client be connected the whole time I have no problems and the whole system runs witout errors.

Because of this behavior I do believe that the server may not recognise all client disconnections or may leave a fragment of every client, even after a full disconnection, in RAM which of course gets a big problem for the small memory of microcontrollers. 

I didn´t find were a client is registered by the server in the sourcefiles therefore I couldn´t look deeper into the problem. Maybe someone can point where to look or has solved the problem already?

## Background Information / Reproduction Steps
I build the open62541 according to the documentation and the helping steps from issue #2511 for usinge open62541 with freeRTOS an LwIP.
As ide I used atollic true studio and my compiler option is Optimize Most (-O3) to build my project for the stm. Hard- and middleware are configured with MXCube

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.0
 - [x] Other OPC UA SDKs used (client or server): none in Code but UA Expert as client
 - [x] Operating system: freeRTOS

jpfr, 2020-04-23:
> Because of this behavior I do believe that the server may not recognise all client disconnections or may leave a fragment of every client, even after a full disconnection, in RAM which of course gets a big problem for the small memory of microcontrollers.

That depends on whether the client closes the session.
Sessions are persistent and can be recovered from a new connection.

Maybe you have to limit the number of sessions that can be open at the same time.

svro86, 2020-04-24:
@jpfr thanks for your quick answer.

Your thoughts about sessions could explain the behavior. Do you, by any chance, know if the UaExpert clientsoftware does send a session close on disconnect? I only get the disconnect-messages in the log of the UaExpert.

But as this is not the UaExpert forum I have a follow up question to open2541 as well:
When I connect to my OPCUA Server I see, that there is a session timeout of 1200000 and I think this is the default value. Am I right, that this timeout cancels the session if a client does not respond at least ones within the timeout window? If so I might solve my probem by reducing the timeout time.

is there any chance you can you point me to the file or function were I can set the properties of sessions? 

Thanks in advance

ccvca, 2020-04-24:
Are you looking for this?
https://github.com/open62541/open62541/blob/578d521931777181d1a881c8335ca5fafcd66eb2/plugins/ua_config_default.c#L196-L199

svro86, 2020-04-27:
Thanks, thats what I needed to know.
I am going to solve my problem by reducing the max number of sessions and maybe reduce the time out as well.
Thanks again.

ccvca, 2020-04-27:
@svro86 If this has solved your problem, please close the issue :)

closed 2020-04-27 by svro86