open62541/open62541 #3571: Whether logins using certificates are supported

I-Yoo, 2020-04-27:
Three login methods: anonymous, username/password, and certificate. The first two have been implemented. Does open62541 support the third?

I-Yoo, 2020-04-27:
Certificate login is not held. It's sad.

jpfr, 2020-04-27:
It is supported but you need to go part of the way yourself.

UA_Client_connect_username simply sets the UserIdentityToken in the client config.
You can do that manually (to set the certificate) and call the regular UA_Client_connect.
https://github.com/open62541/open62541/blob/master/src/client/ua_client_connect.c#L1089

On the server, side you can extend the plugin to support certificates.
https://github.com/open62541/open62541/blob/master/plugins/ua_accesscontrol_default.c#L31
The plugins are licensed under the CC0. So you can integrated with your commercial code.

I-Yoo, 2020-04-28:
Thank you. I'll try it

PatrickAulbach, 2020-05-25:
@I-Yoo Do you have any success with certificate login?

NoelGraf, 2021-05-06:
@I-Yoo Is something still open? Can the issue be closed?

closed 2021-06-04 by NoelGraf