open62541/open62541 #3573: which cmake option should I turn on?

xuexiqiangzhe, 2020-04-28:
I am trying to add a opcua-device to Edgex foundry.
following are some header files relevant to open62541 in my source file.
```
#include "open62541/plugin/log_stdout.h"
#include "open62541/plugin/pubsub_udp.h"
#include "open62541/server.h"
#include "open62541/server_config_default.h"
#include "open62541/types_generated.h"
#include "open62541/ua_pubsub.h"
#include "open62541/ua_pubsub_networkmessage.h"
#ifdef UA_ENABLE_PUBSUB_ETH_UADP
#include "open62541/plugin/pubsub_ethernet.h"
#endif
```
In the folder of open62541,I have a source file in the example folder,also including these header files above,I have done 
```
 mkdir build
 cd build
 cmake .. -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_EXAMPLE=ON
 make
```
and compile successfully
similarly,following is what I have done in a new open62541-master folder.
```
mkdir build
cd build
cmake .. -DBUILD_SHARED_LIBS=ON -DUA_ENABLE_PUBSUB=ON
make
sudo make install
```
But compile my source file fail,there are some necessary header files  not installed into /usr/local/include.
Am I right to do this？
Which cmake option should I turn on?

jpfr, 2020-04-29:
Can you post more detailed error logs?
I do not have the same problem here.

xuexiqiangzhe, 2020-04-29:
book@IMC:~/Documents/device-opcua-c $ ./scripts/build.sh 
and following is the part of the log.
```
+ make
+ tee debug.log
[ 50%] Linking C executable device-opcua-c
CMakeFiles/device-opcua-c.dir/main.c.o: In function `UA_ByteString_clear':
/usr/local/include/open62541/types_generated_handling.h:475: undefined reference to `UA_clear'
CMakeFiles/device-opcua-c.dir/main.c.o: In function `UA_Variant_clear':
/usr/local/include/open62541/types_generated_handling.h:754: undefined reference to `UA_clear'
CMakeFiles/device-opcua-c.dir/main.c.o: In function `UA_ServerConfig_setMinimal':
/usr/local/include/open62541/server_config_default.h:61: undefined reference to `UA_ServerConfig_setMinimalCustomBuffer'
CMakeFiles/device-opcua-c.dir/main.c.o: In function `addPubSubConnection':
/home/book/Documents/device-opcua-c/src/c/main.c:360: undefined reference to `UA_Server_addPubSubConnection'
/home/book/Documents/device-opcua-c/src/c/main.c:364: undefined reference to `UA_PubSubConnection_regist'
CMakeFiles/device-opcua-c.dir/main.c.o: In function `subscriptionfetch':
/home/book/Documents/device-opcua-c/src/c/main.c:395: undefined reference to `UA_NetworkMessage_decodeBinary'
/home/book/Documents/device-opcua-c/src/c/main.c:444: undefined reference to `UA_NetworkMessage_deleteMembers'
CMakeFiles/device-opcua-c.dir/main.c.o: In function `run':
/home/book/Documents/device-opcua-c/src/c/main.c:454: undefined reference to `UA_Server_getConfig'
/home/book/Documents/device-opcua-c/src/c/main.c:468: undefined reference to `UA_PubSubTransportLayerUDPMP'
/home/book/Documents/device-opcua-c/src/c/main.c:482: undefined reference to `UA_PubSubConnection_findConnectionbyId'
collect2: error: ld returned 1 exit status
c/CMakeFiles/device-opcua-c.dir/build.make:96: recipe for target 'c/device-opcua-c' failed
make[2]: *** [c/device-opcua-c] Error 1
CMakeFiles/Makefile2:85: recipe for target 'c/CMakeFiles/device-opcua-c.dir/all' failed
make[1]: *** [c/CMakeFiles/device-opcua-c.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

```

saukijan, 2020-04-30:
Hi @xuexiqiangzhe,

I can not reproduce the problem on my side as well (Debian 9.0, on master commit 578d521 ) with the following options: 

`cmake .. -DBUILD_SHARED_LIBS=ON -DUA_ENABLE_PUBSUB=ON`

Can you provide the commit ID (eg. master 0c10b3e ) that you are trying to build as well as your OS info (Distro, version, etc)? 

Furthermore a minimal viable example would help out a lot, since all of the errors are coming from your `main.c`, which could lead to be that the definitions you use are deprecated and no longer supported. Do you remember the commit ID of the last Master build you used? 

Finally I would suggest using v1.0.1 tag ( `git checkout v1.0.1`) unless you really need the latest from the master branch.

xuexiqiangzhe, 2020-04-30:
Commit ID？Sorry I don't know this concept,I just enter https://github.com/open62541/open62541 , choose branch of master,Download ZIP,and unzip it.
I redo all the work above just now,with the open62541-master downloaded just now.Results are the same as above.

I have compiled a source file in the open62541/example/pubsub folder,including these header files above,also including those undefined reference functions like UA_Server_addPubSubConnection.
In open62541-master
```
 mkdir build
 cd build
 cmake .. -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_EXAMPLE=ON
 make
```
and compile successfully
So I think those undefined reference functions like UA_Server_addPubSubConnection,are not deprecated.

saukijan, 2020-04-30:
> Commit ID？Sorry I don't know this concept,I just enter https://github.com/open62541/open62541 , choose branch of master,Download ZIP,and unzip it.
I redo all the work above just now,with the open62541-master downloaded just now.Results are the same as above.

Well if you did that today, you should be on 0c10b3e

I would recommend that you used a git clone command and initialize all of the sub-modules like this: 

```bash
git clone https://github.com/open62541/open62541.git
git submodule update --init --recursive
```

This will make sure that all of the required dependencies are installed and then will allow you to check your current git commit id by running the following in the terminal: 

```bash
git rev-parse HEAD
```

It will give you a long number, something like: `09952c4bc9ff700f27aa31f2373744871246888c` you can copy paste it in and we will know what commit you are using. 

> So I think those undefined reference functions like UA_Server_addPubSubConnection,are not deprecated. 

What I mean by deprecated, is that inside your `main.c` file, which sits at `/home/book/Documents/device-opcua-c/src/c/main.c`, you are using old function footprints which no longer fit (thought looking at the error messages again, it doesn't look like it), it could be that linking to your main file is wrong or something else entirely. 

We simply dont have enough information to understand were is the problem coming from, and we can not replicate it since we dont know how you are using the library...

xuexiqiangzhe, 2020-04-30:
long number: 0c10b3ea408741abcb339bf53e9975969ae09c01

So I describe my whole process just now,to show you how I am using the library.
1.install header files to /usr/local/include and libopen62541 to /usr/local/lib
```
git clone https://github.com/open62541/open62541.git
cd open62541
git submodule update --init --recursive
git rev-parse HEAD
mkdir build
cd build
cmake .. -DBUILD_SHARED_LIBS=ON -DUA_ENABLE_PUBSUB=ON
make
sudo make install
```
2.complie main.c
~/Documents/device-opcua-c $ ./scripts/build.sh
error log,seems different from above:
```
+ make
+ tee debug.log
Scanning dependencies of target device-opcua-c
[ 50%] Building C object c/CMakeFiles/device-opcua-c.dir/main.c.o
/home/book/Documents/device-opcua-c/src/c/main.c:18:33: fatal error: open62541/ua_pubsub.h: No such file or directory
compilation terminated.
c/CMakeFiles/device-opcua-c.dir/build.make:62: recipe for target 'c/CMakeFiles/device-opcua-c.dir/main.c.o' failed
make[2]: *** [c/CMakeFiles/device-opcua-c.dir/main.c.o] Error 1
CMakeFiles/Makefile2:85: recipe for target 'c/CMakeFiles/device-opcua-c.dir/all' failed
make[1]: *** [c/CMakeFiles/device-opcua-c.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

```

following is part of Cmakelists.txt
```
file (GLOB C_FILES *.c)

FILE(STRINGS "../../VERSION" VERSION_NUMBER)

find_library(EDGEX_CSDK_LIB NAMES csdk PATHS ENV CSDK_DIR PATH_SUFFIXES lib)
find_path(EDGEX_CSDK_INCLUDE NAMES edgex/devsdk.h PATHS ENV CSDK_DIR PATH_SUFFIXES include)
find_library(OPEN62541_RC2_LIB NAMES open62541)
add_executable(device-opcua-c ${C_FILES})

TARGET_COMPILE_DEFINITIONS(device-opcua-c PUBLIC VERSION="${VERSION_NUMBER}")

target_include_directories(device-opcua-c PRIVATE ${EDGEX_CSDK_INCLUDE} .)
target_link_libraries(device-opcua-c PRIVATE ${EDGEX_CSDK_LIB} ${OPEN62541_RC2_LIB})
```

xuexiqiangzhe, 2020-04-30:
I have tried dynamic library, just now
1.modify Cmakelist.txt
```
set (PROJECT_SOURCE_DIR ../../)

file (GLOB C_FILES *.c)

FILE(STRINGS "../../VERSION" VERSION_NUMBER)

include_directories(${PROJECT_SOURCE_DIR}/include)
link_directories(${PROJECT_SOURCE_DIR}/lib)
add_executable(device-opcua-c ${C_FILES})

TARGET_COMPILE_DEFINITIONS(device-opcua-c PUBLIC VERSION="${VERSION_NUMBER}")

target_link_libraries(device-opcua-c -lopen62541 -lcsdk)
```
2.cd open62541-master
```
mkdir build
cd build
cmake .. -DBUILD_SHARED_LIBS=ON -DUA_ENABLE_PUBSUB=ON
make
```
3.copy the whole open62541-master to ${PROJECT_SOURCE_DIR}/include,renaming open62541
4.copy the three lib files including libopen62541.so in open62541/build/bin to ${PROJECT_SOURCE_DIR}/lib
5.set the PROJECT_SOURCE_DIR
6.~/Documents/device-opcua-c $ ./scripts/build.sh
error logs,the numbers of undefined reference functions reduced by:
```
+ make
+ tee debug.log
[ 50%] Linking C executable device-opcua-c
CMakeFiles/device-opcua-c.dir/main.c.o: In function `UA_ServerConfig_setMinimal':
/usr/local/include/open62541/server_config_default.h:61: undefined reference to `UA_ServerConfig_setMinimalCustomBuffer'
CMakeFiles/device-opcua-c.dir/main.c.o: In function `addPubSubConnection':
/home/book/Documents/device-opcua-c/src/c/main.c:364: undefined reference to `UA_PubSubConnection_regist'
CMakeFiles/device-opcua-c.dir/main.c.o: In function `subscriptionfetch':
/home/book/Documents/device-opcua-c/src/c/main.c:395: undefined reference to `UA_NetworkMessage_decodeBinary'
/home/book/Documents/device-opcua-c/src/c/main.c:444: undefined reference to `UA_NetworkMessage_deleteMembers'
CMakeFiles/device-opcua-c.dir/main.c.o: In function `run':
/home/book/Documents/device-opcua-c/src/c/main.c:482: undefined reference to `UA_PubSubConnection_findConnectionbyId'
collect2: error: ld returned 1 exit status
c/CMakeFiles/device-opcua-c.dir/build.make:96: recipe for target 'c/device-opcua-c' failed
make[2]: *** [c/device-opcua-c] Error 1
CMakeFiles/Makefile2:85: recipe for target 'c/CMakeFiles/device-opcua-c.dir/all' failed
make[1]: *** [c/CMakeFiles/device-opcua-c.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

saukijan, 2020-04-30:
Alright, it seems that in your main.c file you are including `ua_pubsub.h` the wrong way. (I am not 100% certain, but it looks like at line 18 of file **main.c** it is written `#include <open62541/ua_pubsub.h>` instead of `#include <ua_pubsub.h>` 

Are you using an IDE of some sort? If so try right clicking on the the include defined at line 18 of file main.c and Selecting **Go To Definition** (If you are using Visual Studio Code you can put the cursor on that line and press F12) if you included it wrong, it will give you an error message, otherwise it will open the [ua_pubsub.h](https://github.com/open62541/open62541/blob/master/src/pubsub/ua_pubsub.h) file

xuexiqiangzhe, 2020-04-30:
I have written #include <open62541/ua_pubsub.h> not  #include <ua_pubsub.h> 
I compile under linux ubuntu.

saukijan, 2020-04-30:
Your latest CMakeLists.txt file looks weird, here is the simplest configuration file you can use: 

```CMake
cmake_minimum_required(VERSION 3.6)

project(device-opcua-c C)

find_package(open62541 REQUIRED)

include_directories(${PROJECT_SOURCE_DIR}/include)

file (GLOB C_FILES *.c)

add_executable(main ${C_FILES })
target_link_libraries(main open62541::open62541)

target_include_directories(main INTERFACE
                           $<BUILD_INTERFACE:${PROJECT_SOURCE_DIR}/include>
                           $<BUILD_INTERFACE:${CMAKE_CURRENT_BINARY_DIR}>
                           $<INSTALL_INTERFACE:${CMAKE_INSTALL_INCLUDEDIR}>
)
```

saukijan, 2020-04-30:
> 
> 
> I have written #include <open62541/ua_pubsub.h> not #include <ua_pubsub.h>
> I compile under linux ubuntu.

If I am not mistaken it should be `#include <ua_pubsub.h>`, however I need to check it, since I do not use pubsub often

saukijan, 2020-04-30:
I just tried it with `#include <ua_pubsub.h>` and `#include <open62541/ua_pubsub.h>` neither of the files exist in the installed package, I think this header is an internal header, and thus not included, you probably need to use the public header `#include "open62541/plugin/pubsub.h"` 

Also checking with:

```c
#include "open62541/plugin/log_stdout.h"
#include "open62541/plugin/pubsub_udp.h"
#include "open62541/server.h"
#include "open62541/server_config_default.h"
#include "open62541/types_generated.h"
#include "open62541/ua_pubsub.h"
#include "open62541/ua_pubsub_networkmessage.h"
``` 

fails to find both `#include "open62541/ua_pubsub.h"` and `#include "open62541/ua_pubsub_networkmessage.h"` headers 

Do you need something specific from [ua_pubsub.h](https://github.com/open62541/open62541/blob/master/src/pubsub/ua_pubsub.h) or is [pubsub.h](https://github.com/open62541/open62541/blob/master/include/open62541/plugin/pubsub.h) enough for you?

xuexiqiangzhe, 2020-04-30:
yes,for example,struct UA_PubSubConnection,I need it.

xuexiqiangzhe, 2020-04-30:
> I have tried dynamic library, just now
> 1.modify Cmakelist.txt
> 
> ```
> set (PROJECT_SOURCE_DIR ../../)
> 
> file (GLOB C_FILES *.c)
> 
> FILE(STRINGS "../../VERSION" VERSION_NUMBER)
> 
> include_directories(${PROJECT_SOURCE_DIR}/include)
> link_directories(${PROJECT_SOURCE_DIR}/lib)
> add_executable(device-opcua-c ${C_FILES})
> 
> TARGET_COMPILE_DEFINITIONS(device-opcua-c PUBLIC VERSION="${VERSION_NUMBER}")
> 
> target_link_libraries(device-opcua-c -lopen62541 -lcsdk)
> ```
> 
> 2.cd open62541-master
> 
> ```
> mkdir build
> cd build
> cmake .. -DBUILD_SHARED_LIBS=ON -DUA_ENABLE_PUBSUB=ON
> make
> ```
> 
> 3.copy the whole open62541-master to ${PROJECT_SOURCE_DIR}/include,renaming open62541
> 4.copy the three lib files including libopen62541.so in open62541/build/bin to ${PROJECT_SOURCE_DIR}/lib
> 5.set the PROJECT_SOURCE_DIR
> 6.~/Documents/device-opcua-c $ ./scripts/build.sh
> error logs,the numbers of undefined reference functions reduced by:
> 
> ```
> [100%] Linking C executable device-opcua-c
> CMakeFiles/device-opcua-c.dir/main.c.o: In function `addPubSubConnection':
> /home/book/Documents/device-opcua-c/src/c/main.c:366: undefined reference to `UA_PubSubConnection_regist'
> CMakeFiles/device-opcua-c.dir/main.c.o: In function `subscriptionfetch':
> /home/book/Documents/device-opcua-c/src/c/main.c:397: undefined reference to `UA_NetworkMessage_decodeBinary'
> /home/book/Documents/device-opcua-c/src/c/main.c:446: undefined reference to `UA_NetworkMessage_deleteMembers'
> CMakeFiles/device-opcua-c.dir/main.c.o: In function `run':
> /home/book/Documents/device-opcua-c/src/c/main.c:484: undefined reference to `UA_PubSubConnection_findConnectionbyId'
> collect2: error: ld returned 1 exit status
> c/CMakeFiles/device-opcua-c.dir/build.make:94: recipe for target 'c/device-opcua-c' failed
> make[2]: *** [c/device-opcua-c] Error 1
> CMakeFiles/Makefile2:85: recipe for target 'c/CMakeFiles/device-opcua-c.dir/all' failed
> make[1]: *** [c/CMakeFiles/device-opcua-c.dir/all] Error 2
> Makefile:83: recipe for target 'all' failed
> make: *** [all] Error 2
> ```

saukijan, 2020-04-30:
> 
> 
> yes,for example,struct UA_PubSubConnection,I need ua_pubsub.h.

I assume you want to establish a `UA_PubSubConnection` on your device right? 

From the documentation it seems that you dont need to implement that yourself (thus this functionality is hidden behind internal headers, thought I might be wrong, someone from pub/sub maintainers should clarify this).  

Anyways try using the following tutorial to establish your Pub/sub connection

https://open62541.org/doc/1.0/tutorial_pubsub_publish.html

xuexiqiangzhe, 2020-04-30:
yes,
and I have understand tutorial_pubsub_publish.c Completely several days ago,and modified it to publish my own Dataset successfully.
main.c of device-opcua-c is to going to be the subscriber.
last time I have replaced the content of tutorial_pubsub_subscribe.c with the content of main.c of the part of Open62541.and compile and run successfully,subscribe Dataset successfully.

xuexiqiangzhe, 2020-04-30:
/home/book/Documents/device-opcua-c/src/c/main.c

xuexiqiangzhe, 2020-04-30:
[main.docx](https://github.com/open62541/open62541/files/4558541/main.docx)
 this is my main.c
I have  modified the code of tutorial_pubsub_publish.c to publish my own Dataset successfully.
main.c is for a device-service of edgex,Integrating a function to decode the UDP messages published by  tutorial_pubsub_publish.c

saukijan, 2020-04-30:
I think I see what your problem is. 

Since PubSub is still in development it is not part of the default configuration and you have to implement that part of the configuration yourself, and to do that you need to have the source code of open62541 so running `make install` for open62541 library will not help you. 

What you need to do is to add open62541 as a sub-module for you project (for example, third_parties/open62541), you can achieve this by running the follwoing in your **device-opcua-c** project root 

`git sub-module add https://github.com/open62541/open62541.git third_parties/open62541`

 and then initializing them

 `git submodule update --init --recursive`

You did something similar by coping the open62541 directory inside your project, but that was not enough, since you need to add open62541 as a sub_project inside your CMakeLists.txt and run both your project options and open62541 library options during your build. Which can be painful to maintain.

@jpfr is there a way to set up PubSub Server with open62431 as a library dependency without using internal [ua_pubsub.h](https://github.com/open62541/open62541/blob/master/src/pubsub/ua_pubsub.h) header?

xuexiqiangzhe, 2020-04-30:
oh，PubSub is still in development it is not part of the default configuration，but published API is finished. 
I can use UA_NetworkMessage_decodeBinary() provided by open62541,to decode UDP messages.Achieve the effect that receiving the data from middleware,originating from pubulisher.

jpfr, 2020-05-15:
You can use this header: https://github.com/open62541/open62541/blob/master/include/open62541/server_pubsub.h

No need to use internal API for normal use.

xuexiqiangzhe, 2020-05-18:
> o need to use internal API for normal
it is still the same
`[100%] Linking C executable device-opcua-c
CMakeFiles/device-opcua-c.dir/main.c.o: In function `addPubSubConnection':
/home/book/Documents/device-opcua-c/src/c/main.c:367: undefined reference to `UA_PubSubConnection_regist'
CMakeFiles/device-opcua-c.dir/main.c.o: In function `subscriptionfetch':
/home/book/Documents/device-opcua-c/src/c/main.c:398: undefined reference to `UA_NetworkMessage_decodeBinary'
/home/book/Documents/device-opcua-c/src/c/main.c:447: undefined reference to `UA_NetworkMessage_deleteMembers'
CMakeFiles/device-opcua-c.dir/main.c.o: In function `run':
/home/book/Documents/device-opcua-c/src/c/main.c:485: undefined reference to `UA_PubSubConnection_findConnectionbyId'
collect2: error: ld returned 1 exit status
c/CMakeFiles/device-opcua-c.dir/build.make:94: recipe for target 'c/device-opcua-c' failed
make[2]: *** [c/device-opcua-c] Error 1
CMakeFiles/Makefile2:85: recipe for target 'c/CMakeFiles/device-opcua-c.dir/all' failed
make[1]: *** [c/CMakeFiles/device-opcua-c.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
`

xuexiqiangzhe, 2020-05-18:
Use Source Insight, I find that.
Function UA_PubSubConnection_regist and UA_PubSubConnection_findConnectionbyId are defined in ua_pubsub_writer.c
Function UA_NetworkMessage_decodeBinary and UA_NetworkMessage_deleteMembers are defined in ua_pubsub_networkmessage.c

jpfr, 2020-05-27:
@xuexiqiangzhe , you have to enable PubSub in cmake so that these files are compiled.

Pro, 2020-06-09:
@xuexiqiangzhe is this issue fixed for you? If yes, please close the issue.

xuexiqiangzhe, 2020-06-09:
> @xuexiqiangzhe is this issue fixed for you? If yes, please close the issue.

not yet

xuexiqiangzhe, 2020-06-18:
I've solved it just now.
device-opcua need some header files and source files about open62541 and some header files about edgex.the logic of open62541 is hard to shake.So I choose add the header files of edgex into open62541,and modified the cmakelists of open62541.
this is the cmakelists under example directory.
```
cmake_minimum_required(VERSION 3.0...3.12)
project(open62541-examples)
if(${CMAKE_VERSION} VERSION_LESS 3.12)
    cmake_policy(VERSION ${CMAKE_MAJOR_VERSION}.${CMAKE_MINOR_VERSION})
endif()
# This examples folder can also be built standalone.
# First install open62541 using `make install` then
# copy this folder to any other location and call CMake directly:
#
# cp -r open62541/examples $HOME/open62541_examples
# cd $HOME/open62541_examples
# mkdir build && cd build
# cmake -DUA_NAMESPACE_ZERO=FULL ..
# make -j

if(CMAKE_PROJECT_NAME STREQUAL PROJECT_NAME)
    # Examples are built standalone. Find installed open62541

    if(UA_NAMESPACE_ZERO STREQUAL "FULL")
        find_package(open62541 REQUIRED COMPONENTS FullNamespace)
    else()
        find_package(open62541 REQUIRED)
    endif()

    if(NOT UA_TOOLS_DIR)
        set(UA_TOOLS_DIR ${open62541_TOOLS_DIR})
    endif()

    if(NOT UA_NODESET_DIR)
        set(UA_NODESET_DIR ${open62541_NODESET_DIR})
    endif()

    function(assign_source_group)
        # define empty function. We don't need it in standalone
    endfunction(assign_source_group)

    include_directories(${PROJECT_BINARY_DIR}/src_generated)
endif()

# Required for common.h header file used in examples
#include_directories(${CMAKE_CURRENT_LIST_DIR})
set(CSDK_DIR ${PROJECT_SOURCE_DIR}/csdk-1.1.1/include)
include_directories(${CMAKE_CURRENT_LIST_DIR} ${CSDK_DIR})

find_library(EDGEX_CSDK_LIB NAMES csdk PATHS ${PROJECT_SOURCE_DIR}/csdk-1.1.1/lib NO_DEFAULT_PATH)

if(UA_ENABLE_AMALGAMATION)
    add_definitions(-DUA_ENABLE_AMALGAMATION)
endif()

#############################
# Compiled binaries folders #
#############################

set(CMAKE_RUNTIME_OUTPUT_DIRECTORY ${CMAKE_BINARY_DIR}/bin/examples)#指的build
set(CMAKE_RUNTIME_OUTPUT_DIRECTORY_DEBUG ${CMAKE_BINARY_DIR}/bin/examples)
set(CMAKE_RUNTIME_OUTPUT_DIRECTORY_RELEASE ${CMAKE_BINARY_DIR}/bin/examples)
set(CMAKE_RUNTIME_OUTPUT_DIRECTORY_RELWITHDEBINFO ${CMAKE_BINARY_DIR}/bin/examples)
set(CMAKE_RUNTIME_OUTPUT_DIRECTORY_MINSIZEREL ${CMAKE_BINARY_DIR}/bin/examples)

macro(add_example EXAMPLE_NAME EXAMPLE_SOURCE)
  add_executable(${EXAMPLE_NAME} ${STATIC_OBJECTS} ${EXAMPLE_SOURCE} ${ARGN} ${CSDK_DIR} ${PROJECT_SOURCE_DIR}/common.h)#指的当前cmakelist目录
  target_link_libraries(${EXAMPLE_NAME} ${EDGEX_CSDK_LIB} open62541::open62541 )
  assign_source_group(${EXAMPLE_SOURCE})
  set_target_properties(${EXAMPLE_NAME} PROPERTIES FOLDER "open62541/examples")
  set_target_properties(${EXAMPLE_NAME} PROPERTIES VS_DEBUGGER_WORKING_DIRECTORY "${CMAKE_BINARY_DIR}/bin")
  if(UA_FORCE_CPP)
      set_source_files_properties(${EXAMPLE_SOURCE} PROPERTIES LANGUAGE CXX)
  endif()
endmacro()

####################
#  Example PubSub  #
####################

if(UA_ENABLE_PUBSUB)
   
    add_example(tutorial_pubsub_publish pubsub/tutorial_pubsub_publish.c)
	
    if (BUILD_SHARED_LIBS)
        message(WARNING "PubSub subscriber API is not finished. Build option BUILD_SHARED_LIBS is not supported. Skipping PubSub subscriber examples.")
    else (NOT BUILD_SHARED_LIBS)
	   
        if(UA_ENABLE_AMALGAMATION)
            message(WARNING "PubSub subscriber tutorial (preview) can not be used with AMALGAMATION. Skipping tutorial_pubsub_subscribe.")
        else(NOT UA_ENABLE_AMALGAMATION)
            add_example(tutorial_pubsub_subscribe pubsub/tutorial_pubsub_subscribe.c)
            add_example(main pubsub/main.c)
        endif()
    endif()
    if(UA_ENABLE_PUBSUB_MQTT)
        if(NOT WIN32)
            include_directories(${PROJECT_SOURCE_DIR}/../plugins)
           add_example(tutorial_pubsub_mqtt_publish pubsub/tutorial_pubsub_mqtt_publish.c)
        endif()
    endif()
endif()
```

closed 2020-06-18 by xuexiqiangzhe