open62541/open62541 #3574: migration from v0.3 to v1.1

prater27, 2020-04-28:
Hi,
first of all thank to the contributors for this great work!

I have a C++ wrapper for open62541 v0.3 and I would like to adapt it to use it with v1.1. 
In the Wrapper, I have a smart pointer such as:
```C++
typedef std::unique_ptr< ::UA_ServerConfig, decltype( &::UA_ServerConfig_delete ) > UAServerConfig;
```
where I use UA_ServerConfig_delete as custom destructor. The "equivalent" function in v1.1 seems to be UA_ServerConfig_clean.

In UA_ServerConfig_clean(UA_ServerConfig *config)  the line "UA_free(config);" does not appear, as it does in UA_ServerConfig_delete(UA_ServerConfig *config). Hence, if I used UA_ServerConfig_clean as custom destructor, would this lead to a memory leak?

I also wanted to let you know that the function **UA_ServerConfig_delete still being used at the very end of the documentation**: https://open62541.org/doc/current/nodeset_compiler.html - last chunk of code.

Thank you very much in advance.

Best regards

jpfr, 2020-04-29:
The docs are improve in #3577.

jpfr, 2020-04-29:
In v1.1, the config is *moved* into the server and automatically removed with UA_Server_delete.
This could lead to duplicate _free with your smart pointer.

closed 2020-05-01 by prater27