open62541/open62541 #3575: Server: Events that aren't subtype of ConditionType trigger error log that claims they are

Tradias, 2020-04-28:
## Description

Unexpected error log and potentially other unwanted consequences when triggering an event that is a subtype of TransitionEventType (or similar).

## Background Information / Reproduction Steps

From my understanding all event types that are not a subtype of ConditionType should not trigger the error log:

```
[2020-04-28 18:07:32.575 (UTC+0200)] error/userland	Alarms and Conditions are not supported yet!
```

I have attached an example code to reproduce the issue. Note that the code is identical to https://github.com/open62541/open62541/blob/84f499b9320f6993213010368480e3ccf307e81a/examples/tutorial_server_events.c except line 35 which was changed from:

`UA_NODEID_NUMERIC(0, UA_NS0ID_BASEEVENTTYPE),`

to

`UA_NODEID_NUMERIC(0, UA_NS0ID_TRANSITIONEVENTTYPE),`

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=FULL
        -DUA_ENABLE_SUBSCRIPTIONS=on
        -DUA_ENABLE_SUBSCRIPTIONS_EVENTS=on
```

Potentially related to https://github.com/open62541/open62541/issues/3102

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.0.1
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Windows 10
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached [log.txt](https://github.com/open62541/open62541/files/4547098/log.txt)
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached: [server.c.txt](https://github.com/open62541/open62541/files/4547112/server.c.txt)
 - [ ] Critical issue

sqadasi, 2020-04-29:
Hello @Tradias, the reason you get that error log, is that there is a problem with resolving the filter set by your client. Have you tried with different clients?
If you are using UaExpert, then the log is actually informing you, that the ConditionId filter clause is set in the filter by the Create MonitoredItem Request. However, that clause won't be resolved. You should get a correct Event Notification.

Tradias, 2020-04-29:
Indeed I am using UaExpert and the ConditionId filter is set (and cannot be disabled in my UaExpert). With your explanation the error log makes sense yet without it I had the feeling that I was doing wrong in my server code or that there was something wrong in the open62541 library. Anyways, thank you for the clarification.

closed 2020-04-29 by Tradias