open62541/open62541 #3580: Duplication of mandatory nodes from IEC61850

HMartan, 2020-04-30:
## Description

I used nodeset_compiler to generate .h and .c files from Opc.Ua.IEC61850-7-3.NodeSet2.xml and Opc.Ua.IEC61850-7-4.NodeSet2.xml files [link](https://github.com/OPCFoundation/UA-Nodeset/tree/master/IEC61850). Then I used open62541 library to create OPC UA server from these generated .h and .c files. When I manually add new LLN0 node in my code, the mandatory children nodes are being automatically created as expected. The problem is there are also some duplicated nodes.

- "LLN0" has mandatory node "Beh" that was created automatically only once (as expected).
- "Beh" has mandatory nodes "q" and "stVal" - each was created twice (but should be created only once).

The differences between duplicate "q" nodes were in their randomly generated node Ids and their browse name:
- The first q node: 2, "q"
- The second q node: 3, "q"

The differences between duplicate "stVal" nodes were the same as with "q" node but they additionally differed in their DataType.

When I create the "LLN0" node in UaModeler, everything seems fine (there is only one "q" and one "stVal" node). It looks like open62541 library has a bug that creates these duplicate nodes. Or is it a feature and I just misunderstood something?

[image]

## Background Information / Reproduction Steps

1. Download the attached zip file with minimal working example.
2. Run cmake with included CMakeLists.
3. Run make.
4. Execute the application.
5. Connect to the application with UaExpert or any other client (we tried also client from open62541).
6. Investigate LLN0 -> Beh nodes and its children.

Used CMake options:

(CMakeFile included)
```bash
cmake ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.1-dev
 - [x] Other OPC UA SDKs used (client or server): UaExpert, UaModeler
 - [x] Operating system: Ubuntu Desktop 18
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
[serverWithIec.zip](https://github.com/open62541/open62541/files/4558390/serverWithIec.zip)

 - [ ] Critical issue

jpfr, 2020-05-02:
Looks like an issue on our end.

**Probably** the following happened:

- `Beh` contains a mandatory node `q`
- The NodeSet defines a specific member `q`
- The `Beh` is instantiated first, automatically creating a mandatory `q` member
- The specific member `q` is added afterwards

Probably this requires a fix in the NodeSet compiler to use the AddNodes_begin AddNodes_finish idiom. When the specific `q` member is added between _begin and _finish, then it can be taken into account when the initialization of `Beh` is finished.

HMartan, 2020-05-04:
Thank you for your reply.

It would be nice to have the predefined NodeIds because I would like to export NodeIds from UaModeler to use them in my source code in form of #define clauses.

I looked into the code of the nodeset_compiler to find out if I could repair it myslef but I got lost. I was too complex. I would be happy to help if I can somehow speed up resolving the issue.

Johnsik, 2020-11-30:
Hi Julius!

I'm see that NodeSet compiler are use the addNode_begin addNode_finish idiom.
However Beh ObjectType contains a mandatory node **2:q** but NodeSet defines a specific member **3:q** so server considers that is different nodes!

When I was trying to model the instance of LLN0 in UaModeler using the same Nodeset I've get only one 2:q node.
When I manually changed generated c-file for Opc.Ua.IEC61850-7-4.NodeSet2.xml and set UA_Server_addNode_begin(..., UA_QUALIFIEDNAME(ns[1], "q"), ...) instead of UA_Server_addNode_begin(..., UA_QUALIFIEDNAME(ns[2], "q"), ...) I've get in server only one 2:q node.

So I thing it's maybe wrong that UA_Server_addNode_begin use qualified name to find existing nodes...

Pavel

Twoshrubs, 2021-11-08:
We also have a project where we need to preserve Node Id's.

If it helps, we get the same issue when creating an 'UA_NS0ID_ANALOGITEMTYPE' then creating our own EURange within the addNode_begin and finish.

You end up with two identical EURange's ..one we created and one auto created.

o6-nbeier, 2026-10-22:
Thanks for reporting this.

We could not reproduce it on 1.4 or later, and versions below 1.4 no longer receive fixes, so we are closing this issue.

If you still see the problem on 1.4 or later, please reopen the issue or open a new one. Include the open62541 version, your platform and build configuration, and a minimal way to reproduce it.

closed 2026-10-22 by o6-nbeier