open62541/open62541 #3581: LDS ME  -> ServerOnNetworkCallback returns discovery url with hostname

magiino, 2020-04-30:
Hello,

I'm running two opc ua servers on two different machines in local network (machines obtain IP address by DHCP). Both servers are LDS with other funcionality like DA, so application type is `_config->applicationDescription.applicationType = UA_APPLICATIONTYPE_SERVER;`

The problem is that `ServerOnNetworkCallback()` returns discovery url with hostname -> `opc.tcp://marek-ubuntu-ntbook:4850` and not with ip address `opc.tcp:192.168.0.221:4850` so the 
registering to LDS server fails, because client can't find server by discovery url with hostname.

What is the best approach to fix this?

## Checklist
Please provide the following information:
 - [ x ] open62541 Version (release number or git tag): v1.0.1
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ Ubuntu 18.04.4 LTS ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-05-05:
@Pro Can you have a look?

Pro, 2020-05-06:
@magiino I will try to answer your questions:

1) LDS Hostname vs IP

The OPC UA Specification Part 12 defines, that the mDNS announcement can contain a hostname OR the ip address of the host (see https://reference.opcfoundation.org/v104/GDS/docs/C.3/). Independent of the discovery features, we use by default the hostname for the endpoint URL of the server.
The discovery code just takes the endpoint url and basically copies it to the mDNS announcement.
Therefore, the announced mDNS content is the same as the configured endpoint discovery URLs.
You can also use Wireshark with the `mdns` filter to see the messages.
In correctly set-up networks, a host should be able to resolve a hostname to the corresponding IP address. To manually hardcode it, you can add the hostname to `/etc/hosts`
In specific cases where the is no DNS server or the resolving doesn't work, it's possible to override the hostname with a specific IP address. This can be achieved by using `UA_ServerConfig_setCustomHostname()` and just passing an IP address.

An example is shown here for a microcontroller:
https://github.com/Pro/open62541-esp32/blob/d87a53e99fe3767027335f9dbaf9c37a44035ce6/main/main.c#L128

2) multiple mDNS announces

Our LDS-ME implementation just responds to any mDNS query for the opc ua namespace.
In your case it could be that another application sends a query every 5 minutes, and then there are multiple mDNS announces since there are multiple servers running in the same network announcing themselves to each other.
Use Wireshark with the `mdns` filter to see what's going on.

Typically you need to implement higher-level logic to filter out multiple announces. In addition, you also should implement some logic to detect, if it is an already known server, or if the server was restarted in between and therefore you need to re-browse. In my setups I'm caching the startup time and comparing it to see if it is a new instance.

3) Same client for multiple register calls

In general one client should also only be connected to one server, otherwise you always need to create new connections.
Since the periodic register handles the connection internally, I strongly recommend to use different clients for different servers. Just create a hash map from Endpoint-URL to Client.

4) Multiple capabilities filter

You are right: 
it looks like the corresponding filter code is wrong:
https://github.com/open62541/open62541/blob/f71912485ec60a42eebbfd96877bef5268c699e6/src/server/ua_services_discovery_multicast.c#L157

I need to create a fix for this.

Pro, 2020-05-06:
@magiino can you please test #3588 if it fixes the capability filtering for you?

magiino, 2020-05-07:
Thank you for amazing answer! I can confirm #3588 fix works.

commit cf4f7b78 (2020-05-06): fix(discovery): Correctly filter ServerOnNetwork (ignore case, match multiple)

Fixes #3581
commit 222538b2 (2020-05-07): fix(discovery): Correctly filter ServerOnNetwork (ignore case, match multiple)

Fixes #3581
commit e8c5e576 (2020-05-08): fix(discovery): Correctly filter ServerOnNetwork (ignore case, match multiple)

Fixes #3581
commit f4b764ed (2020-05-08): fix(discovery): Correctly filter ServerOnNetwork (ignore case, match multiple)

Fixes #3581
closed 2020-05-07 by magiino