open62541/open62541 #3589: Reading nodes during subscription callback

kleskjr, 2020-05-06:
## Description

We have a client that is subscribed to a node value change from a server (through UA_Client_MonitoredItems_createDataChange). During the subscription callback, the client is reading a different value from the same server (UA_Client_readValueAttribute). So far we detected 3 ways that this process can fail:

- if the  ReadRequest takes too long, and a new PublishResponse arrives from the server, the client reports BadRequestHeaderInvalid and crashes after ~6 seconds.

- during the callback function a Request to renew the SecureChannel is sent  -> a ReadRequest is sent  -> Requesting to renew the SecureChannel ->Received an unknown SecurityToken   -> disconnect.

- during the callback function Requesting to renew the SecureChannel, SecureChannel renewed -> Sending a request of type PublishRequest ->Received an unknown SecurityToken -> the client remains connected but the subscription is lost.

Looks like that asynchronous callback functions are not appropriate to be used for reading and writing variables from a server. My question is whether there is a more canonical way to read and write values as reaction on a data change on the server side?

thanks!

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_LOGLEVEL=1 -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_SUBSCRIPTIONS=ON -DBUILD_SHARED_LIBS=ON
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 578d521931777181d1a881c8335ca5fafcd66eb2
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2020-05-06:
We just pushed a big update to the client.
Can you try again with the current master?

Anyway, this has priority one.
We need to add unit tests for all the described cases.

kleskjr, 2020-05-07:
Thanks jpfr,

thanks for the reply! I am attaching a few logs describing the different scenarios (did a dozen or so runs).

Case 1: A second subscription callback is called just after the readRequest during the first callback.

```
subscribe for ns=3, i=10203
[2020-05-07 11:32:33.317 (UTC+0200)] debug/client       Sending a request of type CreateMonitoredItemsRequest
[2020-05-07 11:32:33.317 (UTC+0200)] debug/client       Decode a message of type CreateMonitoredItemsResponse
[2020-05-07 11:32:33.317 (UTC+0200)] debug/client       Subscription 1 | Added a MonitoredItem with handle 1
[2020-05-07 11:32:33.317 (UTC+0200)] info/userland      Monitoring 0, id 1
[2020-05-07 11:32:33.317 (UTC+0200)] debug/client       Sending a request of type PublishRequest
[2020-05-07 11:32:33.317 (UTC+0200)] debug/client       Sending a request of type PublishRequest
[2020-05-07 11:32:33.317 (UTC+0200)] debug/client       Sending a request of type PublishRequest
[2020-05-07 11:32:33.317 (UTC+0200)] debug/client       Sending a request of type PublishRequest
[2020-05-07 11:32:33.317 (UTC+0200)] debug/client       Sending a request of type PublishRequest
[2020-05-07 11:32:33.317 (UTC+0200)] debug/client       Sending a request of type PublishRequest
[2020-05-07 11:32:33.317 (UTC+0200)] debug/client       Sending a request of type PublishRequest
[2020-05-07 11:32:33.317 (UTC+0200)] debug/client       Sending a request of type PublishRequest
[2020-05-07 11:32:33.317 (UTC+0200)] debug/client       Sending a request of type PublishRequest
[2020-05-07 11:32:33.317 (UTC+0200)] debug/client       Sending a request of type PublishRequest
(Dates.unix2datetime(time()), opcelement.value) = (2020-05-07T09:32:34.3, 3620.0f0) ## here it enters first callback
[2020-05-07 11:32:34.662 (UTC+0200)] trace/client       Client internal async
[2020-05-07 11:32:34.662 (UTC+0200)] debug/client       Sending a request of type ReadRequest
(Dates.unix2datetime(time()), opcelement.value) = (2020-05-07T09:32:34.662, 3621.0f0) ## here it enters second callback
[2020-05-07 11:32:34.662 (UTC+0200)] trace/client       Client internal async
[2020-05-07 11:32:34.662 (UTC+0200)] debug/client       Sending a request of type ReadRequest
[2020-05-07 11:32:34.662 (UTC+0200)] info/client        Error receiving the response with status code BadRequestHeaderInvalid
[2020-05-07 11:32:34.662 (UTC+0200)] debug/client       Decode a message of type ReadResponse
scounter = OPCElement{Float32}(0x000027da, 0x0003, 3621.0f0, 0.0)
[2020-05-07 11:32:35.694 (UTC+0200)] debug/client       Sending a request of type PublishRequest
[2020-05-07 11:32:35.695 (UTC+0200)] debug/client       Sending a request of type PublishRequest
(Dates.unix2datetime(time()), opcelement.value) = (2020-05-07T09:32:36.57, 3622.0f0)
[2020-05-07 11:32:36.569 (UTC+0200)] trace/client       Client internal async
[2020-05-07 11:32:36.569 (UTC+0200)] debug/client       Sending a request of type ReadRequest
[2020-05-07 11:32:36.570 (UTC+0200)] debug/client       Decode a message of type ReadResponse
scounter = OPCElement{Float32}(0x000027da, 0x0003, 3622.0f0, 0.0)
[2020-05-07 11:32:37.572 (UTC+0200)] debug/client       Sending a request of type PublishRequest
(Dates.unix2datetime(time()), opcelement.value) = (2020-05-07T09:32:38.57, 3623.0f0)
[2020-05-07 11:32:38.569 (UTC+0200)] trace/client       Client internal async
[2020-05-07 11:32:38.570 (UTC+0200)] debug/client       Sending a request of type ReadRequest
[2020-05-07 11:32:38.570 (UTC+0200)] debug/client       Decode a message of type ReadResponse
scounter = OPCElement{Float32}(0x000027da, 0x0003, 3623.0f0, 0.0)
[2020-05-07 11:32:39.573 (UTC+0200)] debug/client       Sending a request of type PublishRequest
[2020-05-07 11:32:39.661 (UTC+0200)] debug/client       Sending a request of type CloseSessionRequest
[2020-05-07 11:32:39.662 (UTC+0200)] debug/client       Decode a message of type CloseSessionResponse
[2020-05-07 11:32:39.663 (UTC+0200)] info/userland      Subscription Id 1 was deleted
state callback.....
[2020-05-07 11:32:39.663 (UTC+0200)] info/userland      The client is disconnected
```

Case 2: The second scenario is rather stochastic but always occurs after a few minute or a few tens of minutes. 

```
[2020-05-07 09:59:29.319 (UTC+0200)] trace/client       Client internal async
(Dates.unix2datetime(time()), opcelement.value) = (2020-05-07T07:59:29.615, 829.0f0)
[2020-05-07 09:59:29.615 (UTC+0200)] debug/client       Sending a request of type ReadRequest
[2020-05-07 09:59:29.615 (UTC+0200)] debug/client       Decode a message of type ReadResponse
scounter = OPCElement{Float32}(0x000027da, 0x0003, 829.0f0, 0.0)
[2020-05-07 09:59:30.618 (UTC+0200)] debug/client       Sending a request of type PublishRequest
[2020-05-07 09:59:30.718 (UTC+0200)] trace/client       Client internal async
[2020-05-07 09:59:31.319 (UTC+0200)] trace/client       Client internal async
(Dates.unix2datetime(time()), opcelement.value) = (2020-05-07T07:59:31.614, 830.0f0)
[2020-05-07 09:59:31.614 (UTC+0200)] debug/channel      Connection 19 | SecureChannel 1350 | Requesting to renew the SecureChannel
[2020-05-07 09:59:31.614 (UTC+0200)] debug/channel      Requesting to open a SecureChannel
[2020-05-07 09:59:31.614 (UTC+0200)] debug/channel      OPN message sent
[2020-05-07 09:59:31.614 (UTC+0200)] debug/client       Sending a request of type ReadRequest
[... cut ...]
[2020-05-07 10:39:24.112 (UTC+0200)] trace/client       Client connect iterate
[2020-05-07 10:39:24.112 (UTC+0200)] trace/client       Client connect iterate
[2020-05-07 10:39:24.112 (UTC+0200)] trace/client       Client connect iterate
[2020-05-07 10:39:24.112 (UTC+0200)] trace/channel      Connection 19 | SecureChannel 0 | Verifying chunk signature
```
Just for comparison, the same client in which value reading is commented out, runs rock solid since a week. Let me know if providing some minimal example would help.

kleskjr, 2020-05-13:
Thanks jpfr,
the logs above are from the old version (578d521).
Pulling the latest commits, I no longer experience the BadRequestHeaderInvalid when reading values and subscription callback overlap. Thanks!

closed 2020-05-13 by kleskjr