open62541/open62541 #3590: Connect choosees wrong security mode

Ulf-Johnsson, 2020-05-07:
[trace.log](https://github.com/open62541/open62541/files/4591524/trace.log)

## Description
It seems that when trying to establish a encrypted connection to KepServer (I have also tried other servers, like UnifiedAutomation) the connection always fail with 'BadSecurityChecksFailed).
When checking diagnostics in KepServer you can see that open62541 is attempting to establish a connection using securityMode: None.
I am using the encryption_client example, and it should only use SignEncrypt as security mode, but it seems None is choosen for whatever reason.

[image]

## Background Information / Reproduction Steps
Setup a KepServer that only allows connections using Basic256Sha256 encryptions and add client certificate to the trusted list.
Run client_encryption example with same certificates.

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master head
 - [x] Other OPC UA SDKs used (client or server): KepServerEX
 - [x] Operating system: Windows 10
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

ccvca, 2020-05-07:
Looks like a duplicate of #3578 
May be resolved by #3592 and #3579
You may try if the code of #3592 and #3579 if this fixes your issue. (#3579 has extended the encryption example)

Ulf-Johnsson, 2020-05-08:
The current master version does not work.
I Applied the suggested fix in #3579 and that seems to fix the issue.
It would be beneficial if this fix was applied to the master as soon as possible.

jpfr, 2020-05-08:
Hello @Ulf-Johnsson ,

the current master, after #3592, should resolve the issue.
Can you check again?

Note that it is normal that a #None SecureChannel is opened initially.
This is only used to retrieve the Endpoints with the server certificate.

That is then used to open an encrypted SecureChannel in a second step.

If you have out-of-band access to the Endpoint information, you can set it in the client config.
Then the initial #None SecureChannel is omitted.

Ulf-Johnsson, 2020-05-11:
Hi, thansks for your reply.

I tested the issue with a build from the master from 2020-05-06, but in that version the problem is
still present.

Applying the fix from #3579 seems to solve the problem.

Ulf-Johnsson, 2020-05-11:
Update.

I just tried building a version with the latest version of master, and now it seems to work.

jpfr, 2020-05-11:
Wonderful.

closed 2020-05-11 by jpfr