open62541/open62541 #3595: adding FileType to the minimal server

EmbSys2019, 2020-05-11:
first of all thanks for the great work.

i'm using Nucleo-F746 which runs OPCUA minimal server and FreeRTOS.
Everything work fine and i want to add the FileType to my server and implement its functions.

i don't want to use the full server, so i did what @schroeder-  in [https://github.com/open62541/open62541/issues/2862](url) wrote.

i added **FileType** at the end of the tools/schema/datatypes_minimal.txt and built everything once again with cmake.

i don't really know how to use the FileType and how should i instantiate it. what i did is so:

```
  UA_NodeId fileID; 
  UA_ObjectAttributes fileAttr = UA_ObjectAttributes_default;
  fileAttr.displayName = UA_LOCALIZEDTEXT("en-US", "File");
  UA_Server_addObjectNode(server, UA_NODEID_NULL,
                          UA_NODEID_NUMERIC(0, UA_NS0ID_FILETYPE),
                          UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
                          UA_QUALIFIEDNAME(1, "File"), UA_NODEID_NUMERIC(0, UA_NS0ID_BASEOBJECTTYPE),
                          fileAttr, NULL, &fileID);
```

how can i know if the FileType is added to the server? ncuz i can't see FileType under Types folder when i run the server.

how should the functions look like? is there any example 

thnx 

## Checklist

 - [*] open62541 Version (release v1.0)
 - [*] Other OPC UA SDKs used (client or server): Server
 - [*] Operating system: FreeRTOS
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

tsillanp, 2020-08-28:
Hi @EmbSys2019 ,
were you able to proceed? Any implementation examples to push to the repo or here or through private message?

NoelGraf, 2021-05-06:
@EmbSys2019 What is the status of the issue?

NoelGraf, 2021-05-20:
I would then close the issue. Please reopen if required!

closed 2021-05-20 by NoelGraf